nextcloud/nextcloudpi #575: test canceling out?

testbird, 2018-07-26:
https://github.com/nextcloud/nextcloudpi/blob/81c53af4709897f7cf024c10c4a066b22fdbbce6/update.sh#L107

Is it intentional to test after force deleting?:

```
# clean build flags armbian
rm -f /.ncp-image

# not for image builds, only live updates
[[ ! -f /.ncp-image ]] && {
```

nachoparker, 2018-07-26:
Yes, it is there to fix a mistake that was introduced in some of the released images in June preventing them from update. Will delete that line when there is a new batch of images.

Thanks for keeping an eye on the code for possible mistakes. In the future it would be ideal to talk about these things in the commit itself, rather than opening a new issue.

closed 2018-07-26 by nachoparker

testbird, 2018-07-26:
Ok, I see.

(I was just going through the code (not the commits), looking how adaptions to install in arbitrary installs like this VM image may work: https://ftp.freedombox.org/pub/freedombox/testing-latest/freedombox-testing-free_latest_all-amd64.vdi.xz)

testbird, 2018-07-26:
(A note towards a more general curl installer: At least some ncp-config modules like nc-init, nc-backup etc. would need to be installed, to provide NC management functions for the host's configuration management UI, like for example the freedombox  "plinth" web interface.)

nachoparker, 2018-07-27:
>  looking how adaptions to install in arbitrary installs 

cool! there is always room for improvement

>  At least some ncp-config modules like nc-init, nc-backup etc. would need to be installed, to provide NC management functions for the host's configuration management UI, like for example the freedombox "plinth" web interface

I see, that shouldn't be too hard. I tried to keep every feature modular and self contained. At the end of the day, it is a matter of choosing a collection of `.sh` scripts and installing them with `install_script`. You can see examples in `install.sh` and `update.sh`

testbird, 2018-07-28:
I seem to find many inconsitencies, e.g. in lamp.sh, why is everything in install() and nothing in configure()?

nachoparker, 2018-07-28:
Well, `install()` only runs during build and first installation of a `.sh` file. `configure()` is what gets triggered from ncp-config / ncp-web. `lamp.sh` only runs during image build, so we can put everything in the `install()` step.

- `install_script x.sh` runs the install step of `x.sh`
- `activate_script x.sh` runs the config step of `x.sh`

Read more [here](https://github.com/nextcloud/nextcloudpi/wiki/ncp-app-developer-guide).

Ask anything!

testbird, 2018-07-28:
Thanks for the link!
When I looked at the wiki yesterday I didn't find it. Probably because the "Home" is not referencing that entry point. (Menu on the right suffers an overflow and is not categorized.)

nachoparker, 2018-07-28:
yes... it's getting messy :(

testbird, 2018-07-28:
Is there some doc about update.sh?

Maybe the fixes would better fit into the feature scripts, possibly in an update() function.
That might make it easier to skip configurations and updates of external settings.

nachoparker, 2018-07-31:
hi, 

mmm there is no documentation about it, but basically it

- updates (overwrites) the scripts and ncp-web
- applies fixes to older images

I like your idea of having an `update()` hook in ncp-apps, and I considered the idea in the past. The main reason why I ended up putting everything together is so I can keep track of the fixes.

This is because I am trying to avoid accumulating fixes. Lines older than a month or a month after the latest release I clean them up, as they are already fixed in the build. Obviously it is not practical to accumulate old fixes forever.

I hope it makes sense and yeah, any ideas to improve are welcome.

testbird, 2018-08-01:
Yeah, regarding the time line the overview makes sense.

One month could be a little too fast to remove, maybe. It seemed to me once I wasn't able to update an installation to the current state properly, after it was offline for 12 weeks or so.

Usually database migrations are also kept at least until major version steps. This and re-running updates with state checking might be easier in the separate files.

Overall, I'd say a little re-ordering and reviewing/explaining comments could make the current code easier to understand.

(If you get a chance could you maybe test https://github.com/testbird/nextcloudpi in your build setup, to see if it works with the relaxed php versioning etc.?)

testbird, 2018-08-02:
Wish you a good relocation!