nextcloud/nextcloudpi #576: Clarity on where to ask questions/report possible bugs.

Adamantish, 2018-07-26:
### DO NOT ASK QUESTIONS, USE THE FORUMS

...But there is a tag available literally called "question". Also most of the official articles have comments from @nachoparker recommending to ask here. e.g. [this one](https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/)

mschilli87, 2018-07-27:
> Also most of the official articles have comments from @nachoparker recommending to ask here. e.g. [this one](
)

[direct link](https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/#comment-2077)

ovpc, 2018-07-27:
He asked
> Please, ask any technical questions or report problems in github

Meaning, anything that might contribute to solving technical problems, with ncp code.
Github is for anyone who wants to use and contribute to that code. So it is for Code and Wiki contributors.

General inquiries and user issues should be posted on [help.nextcloud's appliances section](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

You can also view and join in general discussions on Telegram.org at Nextcloud and NextcloudPi channels

Adamantish, 2018-07-27:
I was actually suggesting that if you want this to stop happening, can you edit the comments on the blog posts with a lot of traffic?

nachoparker, 2018-07-27:
It is a problem that so many people ask here. I try to instruct people to use the forum for questions everywhere but keeps happening. I didn't think before this conversation that it could be because of comments, and I edited the comments pointed out by @mschilli87.

It is hard to believe that I tell people to use the forums in

- nextcloudpi.com
- announcements
- github
- github issue template

, and still 40% of the people ask here. But it happens, so the blog comments might be an explanation. People think that an old comment in an old post is more reliable that the up-to-date official sources, and when they open the GH issue to ask they see _DO NOT ASK QUESTIONS, use the forums_ but still they  ask in GH. Hard to believe, but it happens.

Anyhow, I appreciate the input, and any suggestions. 

Thanks!