nextcloud/nextcloudpi #577: [docker] Only IP allowed as $DOMAIN argument

el-dom, 2018-07-27:
```
NextCloudPi version  v0.57.8
NextCloudPi image    NextCloudPi_06-20-18
distribution         Debian GNU/Linux 9 \n \l
USB devices          none
datadir              /data/app/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.8G/7.1G
rootfs usage         4.8G/7.1G
swapfile             none
Nextcloud check      ok
Nextcloud version    13.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   172.17.0.2
gateway              172.17.0.1
interface            eth0
certificates         none
NAT loopback         no
uptime               5:38
```
I tried to add the hostname of my raspberry pi as a trusted domain entry while using `docker run` as described in the `Usage` section here: https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/

Complete command: `docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-armhf nextcloud`

However, `nextcloud` doesn't show up in the `trusted_domains` section of `config.php`. I'm not sure if i'm looking at the right script, but if [this](https://github.com/nextcloud/nextcloudpi/blob/4412ef754c67cdf03995817146836c8773eaaa84/docker-common/nextcloud/020nextcloud#L56) is what is responsible for adding the argument as a trusted domain, it only allows IPs. I'm not sure if i'm doing it wrong or the instructions are wrong in claiming that either URL or IP are allowed.

nachoparker, 2018-07-28:
You sir are right. Probably should get rid of that check

el-dom, 2018-07-28:
Maybe something without any sanity checks like
```
if [ -z "$2" ]
  then
    echo "No domain argument supplied. Not adding trusted domain."
else 
    sudo -u www-data php "$OCC" config:system:set trusted_domains 6 --value="$2" 
fi
```
?
I thought it might be `$1` but `$2` is what the script is currently using. So i assume this is correct. I'm no bash expert ;)

nachoparker, 2018-09-10:
fixed. Will be available next time I push an image

closed 2018-09-10 by nachoparker