nextcloud/nextcloudpi #582: Move to PHP7.2

nachoparker, 2018-08-01:
Upgrade in progress in branch `php72` ([link](https://github.com/nextcloud/nextcloudpi/tree/php72)). I was waiting for the armhf redis php 7.2 package to be available, which [seems to be the case](https://jenkins.rfc1925.org) already.

@testbird I see in your repo that you are also trying to achieve this, so I thought I could push my local branch even though it is in progress so we can combine it with what you are doing.

testbird, 2018-08-02:
Note that instead of switching to 7.2, I changed the versioned install commands (php7.0) into generic ones (php), and added stop/start commands for php 7.2 (short of detecting the actual installed versions).
(This may not make everything work with php7.2, but allows to install and test with debian buster.)

nachoparker, 2018-08-02:
I also tried that, but IIRC what happened is that it selected some PHP7.1 and some PHP7.0 stuff, so I ended up doing it that way. I don't remember the details though, so if it can be done in a generic way, that's better.

Another option, is to use a variable, like

```
PHPVER=7.2
apt install php${PHPVER}-redis
```
```

testbird, 2018-08-03:
I can confirm `cd /etc/php ; for dir in * ; do echo $dir ; done`
returns 7.0, 7.1 and 7.2, it is still mixed up.

nachoparker, 2018-08-04:
I am liking the idea of having a variable to make sure we don't accidentally install the wrong PHP version

testbird, 2018-08-05:
Where in the code is library.sh copied from /tmp to /usr/local?

nachoparker, 2018-08-05:
[here](https://github.com/nextcloud/nextcloudpi/blob/master/update.sh#L55)

testbird, 2018-08-06:
Oh, then I completely missed where update.sh comes in during install.

nachoparker, 2018-08-06:
```
apt purge postfix
apt install postfix
```

testbird, 2018-08-17:
What to do about this?:
The /.ncp-image file does not work as an indicator whether the the scripts are run in a image build run (instead of a provisioning run), because the file gets deleted from update.sh and update.sh is run during a build (from ncp.sh). Thus the file will never be there after update.sh.

nachoparker, 2018-08-18:
hi @testbird 

That is not a problem. I tried to explain that I added the `rm ./ncp-image` line in update.sh temporarily to fix a situation where some Odroid images were shipped with it accidentaly. The idea is to remove that line. Actually it could be removed safely already.

I will buy a computer soon and hopefully I will **start** to have more time to come back to NCP. I don't know when because after moving and starting a new job there are a lot of things to do before I can have time for this as I used to.

testbird, 2018-08-18:
I had expected your build scripts won't clean up the logs for example, if update.sh removes that file in the middle of building.

nachoparker, 2018-08-18:
you are right. I will get rid of that line before I build the images for the next release. Actually we can get rid of it now... actually I just did ;)