nextcloud/nextcloudpi #584: docker: letsencrypt broken

nachoparker, 2018-08-02:
Seems like there are some reports that LE is broken in the latest docker image.

https://help.nextcloud.com/t/lets-encrypt-script-broken-docker/34478/4

I won't have a computer available for several weeks, so if anybody could look at this it would be great.

nachoparker, 2018-08-04:
thanks @ovpc, we would have to see why this is failing to find the path. If I recall correctly, `/etc/letsencrypt` should be a link to `/data/letsencrypt` or similar.

Maybe you can open a shell in the container with `docker exec -ti nextcloudpi bash` and look around, see if there is anything weird going on.

Please, try this in a clean container with a clean volume, and do not try to run nc-update. You can remove the volume with `docker volume rm ncdata`.

Pant, 2018-08-04:
I have an expired cert as well. Let's encrypt doesnt work. I installed NCP with Curl Installer.

```
cat /var/log/letsencrypt/letsencrypt.log
2018-08-04 23:07:29,770:DEBUG:letsencrypt.cli:Root logging level set at 30
2018-08-04 23:07:29,775:INFO:letsencrypt.cli:Saving debug log to /var/log/letsencrypt/letsencrypt.log
2018-08-04 23:07:29,778:DEBUG:letsencrypt.cli:letsencrypt version: 0.4.1
2018-08-04 23:07:29,778:DEBUG:letsencrypt.cli:Arguments: ['-n', '--no-self-upgrade', '--webroot', '-w', '/var/www/nextcloud', '--hsts', '--agree-tos', '-m', 'pantelis.fedora@gmail.com', '-d', 'nextcloud.panteleimon.me']
2018-08-04 23:07:29,781:DEBUG:letsencrypt.cli:Discovered plugins: PluginsRegistry(PluginEntryPoint#webroot,PluginEntryPoint#null,PluginEntryPoint#manual,PluginEntryPoint#standalone)
2018-08-04 23:07:29,783:DEBUG:letsencrypt.cli:Requested authenticator webroot and installer None
2018-08-04 23:07:29,784:DEBUG:letsencrypt.plugins.webroot:Creating root challenges validation dir at /var/www/nextcloud/.well-known/acme-challenge
2018-08-04 23:07:29,785:DEBUG:letsencrypt.display.ops:Single candidate plugin: * webroot
Description: Webroot Authenticator
Interfaces: IAuthenticator, IPlugin
Entry point: webroot = letsencrypt.plugins.webroot:Authenticator
Initialized: <letsencrypt.plugins.webroot.Authenticator object at 0xb612a7b0>
Prep: True
2018-08-04 23:07:29,789:DEBUG:letsencrypt.cli:Selected authenticator <letsencrypt.plugins.webroot.Authenticator object at 0xb612a7b0> and installer None
2018-08-04 23:07:29,868:DEBUG:letsencrypt.cli:Picked account: <Account(8a62d42ba900365a324603c1e99db5b0)>
2018-08-04 23:07:29,890:DEBUG:root:Sending GET request to https://acme-v01.api.letsencrypt.org/directory. args: (), kwargs: {}
2018-08-04 23:07:29,910:INFO:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (1): acme-v01.api.letsencrypt.org
2018-08-04 23:07:30,349:DEBUG:requests.packages.urllib3.connectionpool:"GET /directory HTTP/1.1" 200 280
2018-08-04 23:07:30,362:DEBUG:root:Received <Response [200]>. Headers: {'Content-Length': '280', 'Expires': 'Sat, 04 Aug 2018 23:07:30 GMT', 'Strict-Transport-Security': 'max-age=604800', 'Server': 'nginx', 'Connection': 'keep-alive', 'Pragma': 'no-cache', 'Cache-Control': 'max-age=0, no-cache, no-store', 'Date': 'Sat, 04 Aug 2018 23:07:30 GMT', 'X-Frame-Options': 'DENY', 'Content-Type': 'application/json', 'Replay-Nonce': 'YT05zkuYgiiLmw6dvJ80vOond8YRS4EqSViTj_RBf30'}. Content: '{\n  "new-authz": "https://acme-v01.api.letsencrypt.org/acme/new-authz",\n  "new-cert": "https://acme-v01.api.letsencrypt.org/acme/new-cert",\n  "new-reg": "https://acme-v01.api.letsencrypt.org/acme/new-reg",\n  "revoke-cert": "https://acme-v01.api.letsencrypt.org/acme/revoke-cert"\n}'
2018-08-04 23:07:30,363:DEBUG:acme.client:Received response <Response [200]> (headers: {'Content-Length': '280', 'Expires': 'Sat, 04 Aug 2018 23:07:30 GMT', 'Strict-Transport-Security': 'max-age=604800', 'Server': 'nginx', 'Connection': 'keep-alive', 'Pragma': 'no-cache', 'Cache-Control': 'max-age=0, no-cache, no-store', 'Date': 'Sat, 04 Aug 2018 23:07:30 GMT', 'X-Frame-Options': 'DENY', 'Content-Type': 'application/json', 'Replay-Nonce': 'YT05zkuYgiiLmw6dvJ80vOond8YRS4EqSViTj_RBf30'}): '{\n  "new-authz": "https://acme-v01.api.letsencrypt.org/acme/new-authz",\n  "new-cert": "https://acme-v01.api.letsencrypt.org/acme/new-cert",\n  "new-reg": "https://acme-v01.api.letsencrypt.org/acme/new-reg",\n  "revoke-cert": "https://acme-v01.api.letsencrypt.org/acme/revoke-cert"\n}'
2018-08-04 23:07:30,383:DEBUG:parsedatetime:parse (top of loop): [30 days][]
2018-08-04 23:07:30,398:DEBUG:parsedatetime:CRE_UNITS matched
2018-08-04 23:07:30,400:DEBUG:parsedatetime:parse (bottom) [][30 days][][]
2018-08-04 23:07:30,400:DEBUG:parsedatetime:weekday False, dateStd False, dateStr False, time False, timeStr False, meridian False
2018-08-04 23:07:30,401:DEBUG:parsedatetime:dayStr False, modifier False, modifier2 False, units True, qunits False
2018-08-04 23:07:30,401:DEBUG:parsedatetime:_evalString(30 days, time.struct_time(tm_year=2018, tm_mon=8, tm_mday=4, tm_hour=23, tm_min=7, tm_sec=30, tm_wday=5, tm_yday=216, tm_isdst=0))
2018-08-04 23:07:30,401:DEBUG:parsedatetime:_buildTime: [30 ][][days]
2018-08-04 23:07:30,402:DEBUG:parsedatetime:units days --> realunit days
2018-08-04 23:07:30,402:DEBUG:parsedatetime:return
2018-08-04 23:07:30,403:DEBUG:letsencrypt.storage:Should renew, less than 30 days before certificate expiry 2018-08-02 19:30:49 UTC.
2018-08-04 23:07:30,403:INFO:letsencrypt.cli:Cert is due for renewal, auto-renewing...
2018-08-04 23:07:30,409:DEBUG:letsencrypt.cli:Exiting abnormally:
Traceback (most recent call last):
  File "/usr/bin/letsencrypt", line 9, in <module>
    load_entry_point('letsencrypt==0.4.1', 'console_scripts', 'letsencrypt')()
  File "/usr/lib/python2.7/dist-packages/letsencrypt/cli.py", line 1986, in main
    return config.func(config, plugins)
  File "/usr/lib/python2.7/dist-packages/letsencrypt/cli.py", line 706, in obtain_cert
    _, action = _auth_from_domains(le_client, config, domains, lineage)
  File "/usr/lib/python2.7/dist-packages/letsencrypt/cli.py", line 453, in _auth_from_domains
    original_server = lineage.configuration["renewalparams"]["server"]
  File "/usr/lib/python2.7/dist-packages/configobj.py", line 554, in __getitem__
    val = dict.__getitem__(self, key)
KeyError: 'server'

```
I am not sure if it is different from docker because I am on custom so something maybe goes wrong on me.

Pant, 2018-08-04:
I found a solution for me...
```
sudo add-apt-repository ppa:certbot/certbot
sudo apt-get update 
sudo apt-get upgrade letsencrypt -y
```
Maybe packages in debian/ubuntu repos aren't totally up to date...

nachoparker, 2018-08-05:
Seems like there is some symlink madness going on. It looks for `../../archive/xxx`, but in the docker container that is in `/data/etc/archive` and doesn't exist, because it is in `/etc/letsencrypt/archive`.

My guess is that this is what's happening. Could you try manually moving around those folders and see if it gets fixed? I mean, moving `/data/etc/live` to `/etc/letsencrypt/live`

thanks @ovpc.

ovpc, 2018-08-06:
The folder /etc/letsencrypt does not exist.
I tried creating it and first simlink then copy them from
/var/lib/docker/volumes/ncdata/_data/etc/live/ (where the certs get created)
to /etc/letsencrypt/
But get same error when re-running letsencrypt.

ovpc, 2018-08-06:
Tried with another subdomain and letsencrypt was succesfully completed, but it somehow broke config and site is no longer reacheable after reloading docker.

```
Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for xxx.xxxxx.com
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Generating key (2048 bits): /etc/letsencrypt/keys/0007_key-certbot.pem
Creating CSR: /etc/letsencrypt/csr/0007_csr-certbot.pem
IMPORTANT NOTES:
 - Congratulations! Your certificate and chain have been saved at
   /etc/letsencrypt/live/xxx.xxxx.com-0001/fullchain.pem. Your cert
   will expire on 2018-11-04. To obtain a new or tweaked version of
   this certificate in the future, simply run certbot again. To
   non-interactively renew *all* of your certificates, run "certbot
   renew"
 - If you like Certbot, please consider supporting our work by:

   Donating to ISRG / Let's Encrypt:   https://letsencrypt.org/donate
   Donating to EFF:                    https://eff.org/donate-le

System config value trusted_domains => 4 set to string xxx.xxxx.com
System config value overwrite.cli.url set to string https://xxx.xxxx.com
enabled letsencrypt
Done. Press any key...
```

The file /etc/letsencrypt/live/xxx.xxxx.com-0001/fullchain.pem  and xxx.xxxx.com-0001/fullchain.pem supposed to be created but are nowhere to be found, so thats probably why site is no longer available. I installed locate  used updatedb to make sure the files are not in some other dir. They seem to vanish and only simlinks remain.

nachoparker, 2018-08-06:
That is good info. I think that confirms that the links are messed up the way I mentioned above. We will have to find another way of persisting LE. Maybe just persist `/etc/letsencrypt`, or maybe add `/etc/letsencrypt/archive` too

Thanks @ovpc!!

tomcferguson, 2018-09-08:
Hi Nacho, any news on this bug? I've just managed to set nextcloudpi up after trying and failing to set up nextcloud+mariadb+letsencrypt manually several times so thanks for making nextcloudpi!

If and when this bug does get fixed would I have to pull a new docker image or would it be able to be sorted through nc-update?

Many thanks!

nachoparker, 2018-09-08:
either way ;)

tomcferguson, 2018-09-08:
Great news! Have you managed to find a solution for the bug yet? Also is there any way to check whether an update is available before running nc-update? Something like "You are running v0.59.0, latest is v0.59.4"

Thanks!

nachoparker, 2018-09-08:
unfortunately I haven't had the time yet but it will be part of next release (NC14) or earlier.

I have been quite inactive lately because of life but I am back to business

tomcferguson, 2018-09-08:
Great to hear! Thanks again for keeping this project going!

nachoparker, 2018-09-10:
fixed in 8936f0e. Users will receive fix through nc-update once I test it

closed 2018-09-10 by nachoparker

tomcferguson, 2018-09-16:
Hi Nacho,

Just updated to ncp0.59.8 and tried to run letsencrypt but still got this:

> An unexpected error occurred:
> IOError: [Errno 2] No such file or directory: u'/etc/letsencrypt/live/MY.DNS.DOMAIN-0001/cert.pem'

Should this have been fixed with the ncp-update or are you still testing?

Many thanks!

Tom

shaan7, 2018-09-17:
@nachoparker I see that you've pushed the fix. Can you update the image `ownyourbits/nextcloudpi-armhf` to contain the fix?

nachoparker, 2018-09-21:
You can get the fix through the regular update method (`sudo ncp-update` or through ncp-web)

philipgo2, 2019-02-17:
I still get the same warning on 1.6.3 (curl-install on Raspberry Pi). Seems like the automatic fix did not work. What should I do?