nextcloud/nextcloudpi #585: Installation failed!

btlinx, 2018-08-03:
Hello,
I tried several times to install nextcloudpi and the installation never complete.
Procedure:
1.write raspbian lite img on a 4gb sd card
2.put sd in the pi and ssh to it.
3.run command "curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash"

it always stops here:
> System config value trusted_domains => 5 set to string nextcloudpi.local
System config value mail_smtpmode set to string php
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
**Error: Could not download app calendar**
pi@raspberrypi:~ $

This line seems to be in nc-init.sh. I looked up and there is a command "sudo -u www-data php /var/www/nextcloud/occ app:install calendar"
when I run this command manually it works and says app calendar installed!
How can I fix this and why doesn't installation complete after that error?
Thank you in advanced.

nachoparker, 2018-08-04:
Hi, 

Thanks for reporting this. Maybe they removed or renamed the app? in any case, I suggest you download the ready to use SD card version of NCP at nextcloudpi.com

Cheers,

btlinx, 2018-08-06:
Hi,

I used the SD image and it's working now. Thank you.

Regards

nachoparker, 2018-09-10:
I am going to rebuild now, so I will fix it if I run into the error

closed 2018-09-10 by nachoparker

jdainen, 2019-12-26:
I was installing this with docker and had the same issue.  Also it hangs at Starting Next Clog for the first time.

jdainen, 2019-12-27:
This causes the Nextcloud to fail.  If I delete the ncdata it still fails.  Is there some reason the calendar app doesn't download?

gbafana, 2020-06-23:
Same for me installing nextcloudpi on docker on my odroidxu4 (HC1). No matter what i do it always displays "Error: Could not download app calendar".
```

Initializing empty volume..
Making /usr/local/etc/ncp-config.d persistent ...
Making /etc/services-enabled.d persistent ...
Making /etc/letsencrypt persistent ...
Making /etc/shadow persistent ...
Making /etc/cron.d persistent ...
Making /etc/cron.daily persistent ...
Making /etc/cron.hourly persistent ...
Making /etc/cron.weekly persistent ...
Making /usr/local/bin persistent ...
Making /etc/apache2 persistent ...
Starting PHP-fpm
Starting Apache
Starting mariaDB
2020-06-23  8:27:07 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 201 ...
Starting Redis
Starting Cron
Starting Postfix
Uninitialized instance, running nc-init...
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 11 set to string <My `IP>`
System config value trusted_domains => 1 set to string 172.17.0.4
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
nextcloudpi enabled
Error: Could not download app calendar
Init done

```

Please resolve this issue