nextcloud/nextcloudpi #588: Missing datadirectory

cottoa00, 2018-08-08:
I've been running NCP without issue for quite some time but this morning it reported I had a missing datadirectory. We had some power issues last night but seems as if the USB drive did not automount on re-start?

I ssh in and see that I have /media but nothing in/under it. Trying to find easiest way to "correct" this situation. Tried the admin UI to run nc-datadir but it errors on missing directory... automount is on... tried complete shutdown and restart hoping would automount successfully but no change.

Should I rerun the installation wizard or tweak a config file somewhere to get "/media/USBdrive/ncdata" back and working? Don't want to whack anything when a simple reset is possible.

Thanks.

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.58.1
NextCloudPi image    NextCloudPlus_04-21-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata (doesn't exist)
rootfs usage         2.0G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   10.0.0.151
gateway              10.0.0.1
interface            eth0
certificates         ecnextcloud.ddns.net
NAT loopback         no
uptime               45min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Your data directory is invalid
Please check that the data directory contains a file ".ocdata" in its root.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:145
Stack trace:
#0 /var/www/nextcloud/console.php(91): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Aug 08 14:40:46.381115 2018] [proxy_fcgi:error] [pid 1026:tid 1708921904] [client 10.0.0.22:47620] AH01067: Failed to read FastCGI header
[Wed Aug 08 14:40:46.381285 2018] [proxy_fcgi:error] [pid 1026:tid 1708921904] (104)Connection reset by peer: [client 10.0.0.22:47620] AH01075: Error dispatching request to :4443: 
[Wed Aug 08 14:40:47.755302 2018] [mpm_event:notice] [pid 959:tid 1992806400] AH00491: caught SIGTERM, shutting down
[Wed Aug 08 14:40:57.884588 2018] [ssl:warn] [pid 518:tid 1992146944] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Aug 08 14:40:58.046025 2018] [ssl:warn] [pid 823:tid 1992146944] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Aug 08 14:40:59.005142 2018] [mpm_event:notice] [pid 823:tid 1992146944] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Aug 08 14:40:59.005373 2018] [core:notice] [pid 823:tid 1992146944] AH00094: Command line: '/usr/sbin/apache2'
[Wed Aug 08 14:44:52.022709 2018] [proxy_fcgi:error] [pid 916:tid 1683072048] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:44:53.168190 2018] [proxy_fcgi:error] [pid 916:tid 1716659248] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  touch(): Utime failed: Permission denied in /var/www/ncp-web/ncp-output.php on line 81\n'
[Wed Aug 08 14:45:07.151946 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:45:09.662545 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:46:59.804506 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:47:19.183475 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:47:28.065549 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:48:34.756946 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:49:21.378680 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 14:56:15.369487 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 15:02:30.401890 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 15:03:05.644156 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'
[Wed Aug 08 15:03:07.734756 2018] [proxy_fcgi:error] [pid 916:tid 1699865648] [client 10.0.0.22:47660] AH01071: Got error 'PHP message: PHP Warning:  fclose(): supplied resource is not a valid stream resource in /var/www/ncp-web/L10N.php on line 166\n'

```
</details>

<details>
<summary>Database logs</summary>

```

2018-08-08 14:40:48 1639088960 [Note] /usr/sbin/mysqld: Shutdown complete

2018-08-08 14:41:19 1989074944 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2018-08-08 14:41:19 1989074944 [Note] InnoDB: The InnoDB memory heap is disabled
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2018-08-08 14:41:19 1989074944 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Compressed tables use zlib 1.2.8
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Using Linux native AIO
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Using generic crc32 instructions
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Completed initialization of buffer pool
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Highest supported file format is Barracuda.
2018-08-08 14:41:19 1989074944 [Note] InnoDB: 128 rollback segment(s) are active.
2018-08-08 14:41:19 1989074944 [Note] InnoDB: Waiting for purge to start
2018-08-08 14:41:20 1989074944 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 216542436
2018-08-08 14:41:20 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-08-08 14:41:20 1989074944 [Note] Plugin 'FEEDBACK' is disabled.
2018-08-08 14:41:20 1989074944 [Note] Server socket created on IP: '127.0.0.1'.
2018-08-08 14:41:20 1989074944 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/USBdrive/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

nachoparker, 2018-08-11:
hi,

try to mount it manually, it will fail with an error if the filesystem got corrupted, and you might be able to fix it with `fsck`

cottoa00, 2018-08-27:
Mounted it manually and failed as you expected... used fsck to repair... then mount worked and system was back up. When shutdown and re-launched it did not automount. 
Manully remounted again... Went into the maintenance panel and used nc-datadir and nc-automount w/o error. Still not automounting... is there something I should do in the maint panel or do I need to tweak some underlying config files?

cottoa00, 2018-08-27:
this is what is in /etc/fstab...

proc            /proc           proc    defaults          0       0
PARTUUID=30b283f0-01  /boot           vfat    defaults          0       2
PARTUUID=30b283f0-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

... and here is results of UUID query
root@nextcloudplus:/home/pi# ls -l /dev/disk/by-uuid/*
lrwxrwxrwx 1 root root 15 Aug 27 21:27 /dev/disk/by-uuid/90a83158-560d-48ee-9de9-40c51d93c287 -> ../../mmcblk0p2
lrwxrwxrwx 1 root root 15 Aug 27 21:27 /dev/disk/by-uuid/A365-6756 -> ../../mmcblk0p1
lrwxrwxrwx 1 root root 10 Aug 27 21:27 /dev/disk/by-uuid/db514d17-e34a-48c4-9455-1ce8ac657c1d -> ../../sda1

nachoparker, 2018-08-28:
it should automount to `/media/xxxx`. You can check `systemctl status nc-automount` to investigate further

cottoa00, 2018-08-28:
ran service check and got the following; (my original mount point was "/media/USBdrive"... am I mismatched with how script is truying to mound drive?

 nc-automount.service - Automount USB drives
   Loaded: loaded (/usr/lib/systemd/system/nc-automount.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-08-27 21:27:31 UTC; 17h ago
  Process: 332 ExecStartPre=/bin/bash -c rmdir /media/* || true (code=exited, status=0/SUCCESS)
 Main PID: 367 (udiskie)
   CGroup: /system.slice/nc-automount.service
           └─367 /usr/bin/python3 /usr/bin/udiskie -NTF

Aug 27 21:27:31 nextcloudplus systemd[1]: Starting Automount USB drives...
Aug 27 21:27:31 nextcloudplus bash[332]: rmdir: failed to remove '/media/USBdrive': Directory not empty
Aug 27 21:27:31 nextcloudplus bash[332]: rmdir: failed to remove '/media/USBdrive1': Not a directory
Aug 27 21:27:31 nextcloudplus systemd[1]: Started Automount USB drives.
Aug 27 21:27:35 nextcloudplus udiskie[367]: mounted /org/freedesktop/UDisks2/block_devices/sda1 on /media/ecCloudDri

nachoparker, 2018-08-29:
Try the following...

```
sudo umount /media/*
sudo rmdir /media/*
```

then, inspect whatever is left in `/media/`, there is probably junk files or something. Clean that up until there is nothing there, and reboot.

cottoa00, 2018-08-29:
Cleared some stuff from /media/ and am back in business... Thanks

nachoparker, 2018-08-30:
awesome!

cottoa00, 2019-02-14:
FYI... those of you using the Talk app... a new iOS version 3.3.0 is now avail on the app store... we removed the old, installed the new and voila, it ringss like WhatsApp... Amen! And thanks to everyone bring us these alternatives! Not sure where this should go...sorry but thought it might be useful.

closed 2019-02-14 by cottoa00