nextcloud/nextcloudpi #589: [ ENHANCEMENT ] VPN Client

shadowstreik, 2018-08-11:
It would be very useful to have an openVPN client and usable through NCP.

Co0p3r21, 2018-12-04:
Do you mean you want to include openVPN NCP? That should be [PiVPN](https://github.com/pivpn/pivpn) if you are using a Pi.

An why do you need a client in your NCP installation? Do you want your server to connect via VPN to another Network. 

Or do you want to connect from another device to the network of your server (Pi)? In that case you can just use [PiVPN](https://github.com/pivpn/pivpn) with `curl -L https://install.pivpn.io | bash`

shadowstreik, 2018-12-04:
"Do you mean you want to include openVPN NCP?"

Yes.

"An why do you need a client in your NCP installation?"

https://www.whatismyip.com/what-is-a-vpn/

"Do you want your server to connect via VPN to another Network."

I would like it to connect to a VPN service such as ProtonVPN(specifically) using an inclusive NCP-driven function, not something I add, manually.

Thank you for looking at this.

--
============================================================
CONFIDENTIALITY NOTE: The information transmitted, including attachments, may or may not be intended only for the person(s) or entity to which it is addressed and may or may not contain confidential and/or privileged material. Any review, re-transmission, dissemination or other use of, or taking of any action in reliance upon this information by persons or entities other than the intended recipient may or may not be prohibited.

If you may have received this in error, please contact the sender via E-Mail, only, and destroy any copies of this information.

IMPORTANT: This E-Mail may or may not be intended for the use of the individual addressee(s) named above and may or may not contain information that is confidential, privileged or unsuitable for overly sensitive persons with low self-esteem, no sense of humor or irrational religious and/or political beliefs. Any dissemination, distribution or copying of this E-Mail may or may not be authorized(either explicitly or implicitly) and may constitute an irritating social faux-pas. Also, any E-Mail sent from the originating author may or may not claim ownership of all E-Mail sent and received on local computer(s) in addition to all non-solicited E-Mails received.

CAUTION: Unless the word 'absquatulation' has been used in its correct context somewhere other than in this warning, it may or may not have any legal nor grammatical use and may be ignored. No animals were harmed in the transmission of this E-Mail, although the kelpie next door may or may not be living on borrowed time, let me tell you. Those of you with an overwhelming fear of the unknown may or may not be gratified to learn that there is no hidden message revealed by reading this warning backwards, so just ignore that alert notice from Microsoft.

ANTIDOTE: However, by pouring a complete circle of salt around yourself and your computer you can ensure that no harm befalls you and your pets. If you have received this E-Mail in error, please add some nutmeg and egg whites, whisk and place in a warm oven for 40 minutes.

CONFIDENTIALITY NOTE: The information transmitted, including attachments, may or may not
be intended only for the person(s) or entity to which it is addressed and may or may not contain
confidential and/or privileged material. Any review, re-transmission, dissemination or other use
of, or taking of any action in reliance upon this information by persons or entities other than the
intended recipient may or may not be prohibited.

If you may have received this in error, please contact the sender via E-Mail, only, and destroy
any copies of this information.

IMPORTANT: This E-Mail may or may not be intended for the use of the individual addressee(s)
named above and may or may not contain information that is confidential, privileged or
unsuitable for overly sensitive persons with low self-esteem, no sense of humor or irrational
religious and/or political beliefs. Any dissemination, distribution or copying of this E-Mail may or
may not be authorized(either explicitly or implicitly) and may constitute an irritating social faux
pas. Also, any E-Mail sent from the originating author may or may not claim ownership of all E
Mail sent and received on local computer(s) in addition to all non-solicited E-Mails received.

CAUTION: Unless the word 'absquatulation' has been used in its correct context somewher
other than in this warning, it may or may not have any legal nor grammatical use and may be
ignored. No animals were harmed in the transmission of this E-Mail, although the kelpie next
door may or may not be living on borrowed time, let me tell you. Those of you with an
overwhelming fear of the unknown may or may not be gratified to learn that there is no hidden
message revealed by reading this warning backwards, so just ignore that alert notice from
Microsoft.

ANTIDOTE: However, by pouring a complete circle of salt around yourself and your computer
you can ensure that no harm befalls you and your pets. If you have received this E-Mail in error,
please add some nutmeg and egg whites, whisk and place in a warm oven for 40 minutes.

Skål! Aho! Remember what's important. "So say we all!"

Co0p3r21, 2018-12-04:
Okay I see you want to use something like ProtonVPN (not openVPN anymore). 
But why does a Nextcloud Pi instance need a VPN connection to another Network. 

I mean it is a Server, therefore it mostly receives data. And you can´t make use of any "hide the ip feature" as well. Because you need an IP to connect to the server. 

The only reason I can see is: 
You want to secure data trafic between your Nextcloud Client (Browser, Phone-Client etc.). Maybe you want to have a look at the build in end-to-end encryption. 

Maybe I need a more detailed usecase.

shadowstreik, 2020-01-02:
Closing issue. I am no longer involved with this project.

closed 2020-01-02 by shadowstreik

petesimonic, 2021-08-29:
> Do you mean you want to include openVPN NCP? That should be [PiVPN](https://github.com/pivpn/pivpn) if you are using a Pi.
>
> An why do you need a client in your NCP installation? Do you want your [...]

Could the issue be returned? I can describe a (typical) use case.

User owns a VPN-server which is accessible from Internet. User don't want to expose NCP to Internet OR can't make it accessible from home. (Security or infrastructure limitations.) VPN client (OpenVPN) enables access to NCP via Internet, but only for users with VPN certs/access. It's best balance of protection and availability for a household scenario, which is target for NCP (as I assume).

**Btw, is it ok to just install Openvpn via APT, and use it as the client?** A dedicated feature with GUI would be perfect, of course.

shadowstreik, 2021-08-29:
This closed was closed on Jan 2, 2020. I am no longer involved in the project. Thank you.

petesimonic, 2021-08-29:
> This issue was closed on Jan 2, 2020. I am no longer involved in the project. Thank you.

That's why I didn't touch your comments. Maybe you should try to unsubscribe. I believe @nachoparker can reply if he would find the topic valid.