nextcloud/nextcloudpi #590: 'Internal Server Error' - PHP down, no access to Nextcloud

richtercamden, 2018-08-12:
I know how I caused it, but I don't know exactly what 'it' is. I made the (dumb) decision to try and upgrade to PHP 7.2, and doing so purged the default install of PHP 7.0. Since I couldn't install PHP 7.2, I decided to reinstall PHP 7.0, but after doing this, my server is completely inaccessible. Before this, it was fine. I also noticed in the report that it marks PHP as 'down'. I assume I broke / missed a library? How do I get it back up? Thanks in advance!

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.58.1
NextCloudPi image    NextCloudPi_06-27-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.0G/30G
rootfs usage         2.0G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    13.0.5.2
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.42
gateway              192.168.1.1
interface            wlan0
certificates         camdenrichter.com
NAT loopback         yes
uptime               11min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.42",
            "5": "nextcloudpi.local"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "13.0.5.2",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
[... 636 lines cut ...]
\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Server.php(691): OC\\\\Server->getMemCacheFactory
()\\n#10 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/pimple\\\/pimple\\\/src\\\/Pimple\\\/Container.php(1
13): OC\\\\Server->OC\\\\{closure}(*** sensitive parameters replaced ***)\\n#11 \\\/var\\\/www\\\/nextcl
oud\\\/lib\\\/private\\\/AppFramework\\\/Utility\\\/SimpleContainer.php(116): Pimple\\\\Container->offse
tGet('OC\\\\\\\\App\\\\\\\\AppManag...')\\n#12 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/ServerCo
ntainer.php(132): OC\\\\AppFramework\\\\Utility\\\\SimpleContainer->query('OC\\\\\\\\App\\\\\\\\AppManag
...')\\n#13 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Utility\\\/SimpleContainer.
php(164): OC\\\\ServerContainer->query('OC\\\\\\\\App\\\\\\\\AppManag...')\\n#14 \\\/var\\\/www\\\/nextc
loud\\\/3rdparty\\\/pimple\\\/pimple\\\/src\\\/Pimple\\\/Container.php(109): OC\\\\AppFramework\\\\Utili
ty\\\\SimpleContainer->OC\\\\AppFramework\\\\Utility\\\\{closure}(*** sensitive parameters replaced ***)
\\n#15 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Utility\\\/SimpleContainer.php(1
16): Pimple\\\\Container->offsetGet('AppManager')\\n#16 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\
/ServerContainer.php(132): OC\\\\AppFramework\\\\Utility\\\\SimpleContainer->query('AppManager')\\n#17 \
\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Server.php(1684): OC\\\\ServerContainer->query('AppManag
er')\\n#18 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(332): OC\\\\Server->getApp
Manager()\\n#19 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(113): OC_App::getEnab
ledApps()\\n#20 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/base.php(662): OC_App::loadApps(Array)\\n#21 \\\/v
ar\\\/www\\\/nextcloud\\\/lib\\\/base.php(1081): OC::init()\\n#22 \\\/var\\\/www\\\/nextcloud\\\/status.
php(36): require_once('\\\/var\\\/www\\\/nextcl...')\\n#23 {main}\",\"File\":\"\\\/var\\\/www\\\/nextclo
ud\\\/lib\\\/private\\\/Memcache\\\/Factory.php\",\"Line\":97,\"Hint\":\"Is the matching PHP module inst
alled and enabled?\"}","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version
":"13.0.5.2"}

```
</details>

nachoparker, 2018-08-12:
Hello,

This is not a bug or feature request, so I suggest you seek for help in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

closed 2018-08-12 by nachoparker

misterlindsey, 2019-02-27:
Richter, did you resolve? Getting an Internal Server Error after running updates. I believe it to be related to PHP 7.2 as well. Erroring at drive.mattlindsey.com and not sure which error log to check for more verbose info. Any insight would be great.

misterlindsey, 2019-02-27:
Updated NextCloudPi in admin console and everything is back up. Not sure what happened.