nextcloud/nextcloudpi #592: Automatically create new user for SSH access

TomTurnschuh, 2018-08-16:
For security and other reasons it would be nice to be able to automatically create a new user only for SSH via ncp-config.

#### Workflow:
The admin can define any new user name with a password. This user is then automatically created with permissions for SSH access.

nachoparker, 2018-08-18:
This was original behavior, but we removed it. See https://github.com/nextcloud/nextcloudpi/issues/388

But it got easily messy with people creating many users accidentally. Once granted SSH access, the user can add more SSH enabled users through command line.

Thanks for the suggestion. Feel free to insist or send a PR if you think of a good way of doing this.

closed 2018-08-18 by nachoparker

TomTurnschuh, 2018-08-23:
Maybe it could make sense to just add a single additional user called 'ncpssh' or similar, without the option to change this name?
This user with root access could then be the same for all platforms; and it isn't the standard root user.

Additional SSH users can then be enabled again through command line, if needed.

nachoparker, 2018-08-25:
mmm don't really see the point. You have the `pi` user in Raspbian, and the `root` user elsewhere. SSH is disabled by default and ncp won't let you enable it with default passwords.

If you want a specific user you can always create it and disable the default user if you want.