nextcloud/nextcloudpi #595: Info fail2ban and trusted domain error

Haraade, 2018-08-18:
Hello.
Seeing some connection attempts against server.
Has therefore modified fail2ban.

/etc/fail2ban/jail.local

[nextcloud2]
enabled = true
port = http,https
filter = nextcloud2
logpath = /data/nextcloud/nextcloud.log
maxretry = 1
findtime = 1
bantime = 604800 ; 1 week

[ufwban]
enabled = true
port = ssh, http, https
filter = ufwban
logpath = /var/log/ufw.log
maxretry = 1
findtime = 1
bantime = 604800
action = ufw

and

/etc/fail2ban/filter.d/nextcloud2.conf

[INCLUDES]
before = common.conf

[Definition]
failregex = ^.*"remoteAddr":"".Trusted domain error.$
ignoreregex =

and

/etc/fail2ban/filter.d/ufwban.conf

[INCLUDES]
before = common.conf

[Definition]
failregex = UFW BLOCK.* SRC=
ignoreregex =

nachoparker, 2018-08-18:
Cool! so can you explain a little bit? it seems like you added a jail for banning based on UFW. Anything other changes?

Thanks for sharing

Haraade, 2018-08-19:
The reason for using ufwban is that I have registered many connection attempts from the same ip address, to port 80. Only one try and ip is baned for one week (or the time you want). No other changes.

ovpc, 2018-08-19:
I had (and still have) lots of failed attemps to ssh, none to http or https, so created following script and put it in roots crontab to run twice daily. I also set maxtry = 3 in jail.local for both ssh and nextcloud.

 ```
#!/bin/bash
##read from log, get banned ip's, take 8th colum, write to txt file
cat /var/log/fail2ban.log | grep Ban | awk '{print $8}' OFS='\t' > /home/oliver/bin/fail2ban2ufw4root.txt
#
sleep 1
##read from txt file, exec insert
while read IPADDR
do 
	/usr/sbin/ufw insert 1 deny from "$IPADDR"
done < /home/oliver/bin/fail2ban2ufw4root.txt

```

nachoparker, 2018-08-22:
Cool. #408 also seems to me like a good solution for securing the HTTP server

@Haraade would you send a PR with this?

Haraade, 2018-08-22:
Hello. 
do not have the opportunity to do it now. But if somebody else can do it then it would have been great. Port options in ufwban are not required because action = ufw blocks ip.

dmaroulidis, 2018-10-24:
Hi, just to be sure @nachoparker do you want a PR that fixes #408 and adds the above ufw-based filter?

ovpc, 2018-10-24:
Not sure everyone would want it enabled by default?!
You would have to make sure the IP address(es) of trusted users (yourself at least) be ignored in fail2ban config file, to make sure you don't get locked out yourself permanently.

nachoparker, 2018-10-28:
https://github.com/nextcloud/nextcloudpi/blob/master/etc/ncp-config.d/fail2ban.sh

https://github.com/nextcloud/nextcloudpi/wiki/ncp-app-developer-guide

nachoparker, 2018-11-02:
Implemented in PR https://github.com/nextcloud/nextcloudpi/pull/704

thanks everyone!

closed 2018-11-02 by nachoparker