nextcloud/nextcloudpi #606: doubtful kernel version

BO41, 2018-08-23:
the [README.md](README.md) of this repo says
>4.9 Linux Kernel ( NEW 03-13-2017 )

I am not sure if this makes sense.
As far as I know the newest kernel version is 4.18.x.

I guess there is a typo but I don't know the real version

mschilli87, 2018-08-23:
@BO41: But 4.9 < 4.18.x as 9 < 18, or am I confused? So isn't it possible that the kernel used is version 4.9 which was 'new' a year and a half ago and in the meantime there where 18 - 9 = 9 new kernel versions, such that the latest now is 4.18.x?

BO41, 2018-08-23:
@mschilli87 no sorry, I got confused.
I was thinking like 4.18, 4.28, ... 4.88, 4.9
I assumed it would be 4.09 and not 4.9.

sorry, looks fine then :)

closed 2018-08-23 by BO41