nextcloud/nextcloudpi #609: RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54

crossr1, 2018-08-29:
@nachoparker - I don't know enough about redis to fix this. So, I need you help. Here are my output files:

[redis.txt](https://github.com/nextcloud/nextcloudpi/files/2333329/redis.txt)
[ncp-report.txt](https://github.com/nextcloud/nextcloudpi/files/2333334/ncp-report.txt)

Thanks,
Richard

Paste the output of `sudo ncp-report`

crossr1, 2018-08-29:
I was finally able to determine that it was a permissions issue on the /var/log/redis.log file. For some reason, it had root:root permissions instead of redis:redis. System is operational again....

closed 2018-08-29 by crossr1

nachoparker, 2018-12-17:
in any case it would be `chmod -R redis:redis /var/log/redis`, but I wonder how this happened, doesn't happen to me

markedphillips, 2018-12-17:
Sorry, but is this chown -R redis:redis /var/log/redis and are there any particular bits for the permissions? chmod?

markedphillips, 2018-12-17:
Wow, that one ownership issue indeed fixed the system, it came up clean.

nachoparker, 2018-12-17:
awesome

still interested to know how this happened, could you share output of `sudo ncp-report`, `systemctl status redis` and `cat /var/log/redis/redis-server.log` please?

cottoa00, 2019-06-14:
I am having same issue with redis server went away... here is requested output from my system;

637:M 13 Jun 14:55:14.725 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
637:M 13 Jun 14:55:14.725 # Server started, Redis version 3.2.6
637:M 13 Jun 14:55:14.877 * DB loaded from disk: 0.151 seconds
637:M 13 Jun 14:55:14.877 * The server is now ready to accept connections at /var/run/redis/redis.sock
637:signal-handler (1560437724) Received SIGTERM scheduling shutdown...
637:signal-handler (1560437724) Received SIGTERM scheduling shutdown...
637:M 13 Jun 14:55:24.607 # User requested shutdown...
637:M 13 Jun 14:55:24.607 * Saving the final RDB snapshot before exiting.
637:M 13 Jun 14:55:24.874 * DB saved on disk
637:M 13 Jun 14:55:24.875 * Removing the pid file.
637:M 13 Jun 14:55:24.875 * Removing the unix socket file.
637:M 13 Jun 14:55:24.875 # Redis is now ready to exit, bye bye...

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncp-diag: line 27: jq: command not found
grep: /etc/mysql/mariadb.conf.d/90-ncp.cnf: No such file or directory
Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v0.66.4
NextCloudPi image    NextCloudPlus_04-21-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      174G/916G
rootfs usage         2.3G/15G
swapfile             /var/swap
dbdir                 (doesn't exist)
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               41min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('a2b270f1d0b40ac...')
#1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/private/App/AppManager.php(380): OC\App\InfoParser->parse('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/private/legacy/app.php(551): OC\App\AppManager->getAppInfo('files', false, NULL)
#4 /var/www/nextcloud/lib/private/AppFramework/App.php(65): OC_App::getAppInfo('files')
[... 88 lines cut ...]
{"reqId":"WWlrp8CoAZcAAAOVs40AAAAH","level":2,"time":"2017-07-15T01:11:03+00:00","remoteAddr":"66.87.81.78","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/dav\/principals\/users\/admin\/","message":"Trusted domain error. \"66.87.81.78\" tried to access using \"ecnextcloud.ddns.net\" as host.","userAgent":"iOS\/9.3.2 (13F69) dataaccessd\/1.0","version":"12.0.0.29"}
{"reqId":"WWnEjsCoAZcAAAOVs44AAAAI","level":2,"time":"2017-07-15T07:30:22+00:00","remoteAddr":"169.229.3.91","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"169.229.3.91\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"12.0.0.29"}
{"reqId":"WWnLTsCoAZcAAAOVs48AAAAJ","level":2,"time":"2017-07-15T07:59:10+00:00","remoteAddr":"184.105.247.252","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.247.252\" tried to access using \"100.14.82.22\" as host.","userAgent":"--","version":"12.0.0.29"}
{"reqId":"WWoRNsCoAZcAAAOWd3sAAEYY","level":2,"time":"2017-07-15T12:57:26+00:00","remoteAddr":"137.226.113.21","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"137.226.113.21\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko\/20100101 Firefox\/50.0","version":"12.0.0.29"}
{"reqId":"WWoRN8CoAZcAAAOWd4MAAEYG","level":2,"time":"2017-07-15T12:57:27+00:00","remoteAddr":"137.226.113.21","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=fbc9c8ac349f665c24b153acfccad9cf-0","message":"Trusted domain error. \"137.226.113.21\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko\/20100101 Firefox\/50.0","version":"12.0.0.29"}

```
</details>

 redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-06-14 17:49:25 UTC; 42min ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)

Jun 14 17:49:25 nextcloudplus systemd[1]: redis-server.service: Start request repeated too quickly.
Jun 14 17:49:25 nextcloudplus systemd[1]: Failed to start Advanced key-value store.
Jun 14 17:49:25 nextcloudplus systemd[1]: redis-server.service: Unit entered failed state.
Jun 14 17:49:25 nextcloudplus systemd[1]: redis-server.service: Failed with result 'exit-code'.
[... cut ...]
Jun 14 18:13:31 nextcloudplus systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
Jun 14 18:13:45 nextcloudplus systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
Jun 14 18:13:46 nextcloudplus systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'
Jun 14 18:13:46 nextcloudplus systemd[1]: [/lib/systemd/system/redis-server.service:14] Unknown lvalue 'RunTimeDirectory' in section 'Service'

cottoa00, 2019-06-14:
This came about after I attempted an upgrade from v0.66.4 -> v1.13.1 that choked... see my other input posted on a different thread at 
https://help.nextcloud.com/t/ncp-unreachable-after-blackout-redis/52107/5

fxstbg, 2019-08-08:
Hello !
My Nextcloupi server is down, and I don't understand what happened (I did a dist-upgrade of my Armbian two weeks ago, and everything seemed to work like a charm... didn't have any error messages when syncing with my Android devices...). Today I connected to NextcloudPi and got a "Redis down" on System info. 
I read the tread (and many others) and changed the owner on /var/log/redis to redis:redis (it was root:root). 
Still got a "Nextcloud Check error" on System info, and cannot log to Nextcloud.

Here are my logs 
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.15.3
distribution         Debian GNU/Linux 10 \n \l
automount            yes
USB devices          sda sdb 
datadir              /media/myCloudDrive/data
data in SD           no
data filesystem      btrfs
data disk usage      4,8G/466G
rootfs usage         5,8G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               10min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('ac931e8221d05c0...')
#1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/private/App/AppManager.php(381): OC\App\InfoParser->parse('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/private/legacy/app.php(551): OC\App\AppManager->getAppInfo('files', false, NULL)
#4 /var/www/nextcloud/lib/private/AppFramework/App.php(66): OC_App::getAppInfo('files')
#5 /var/www/nextcloud/lib/private/legacy/app.php(236): OC\AppFramework\App::buildAppNamespace('files')
#6 /var/www/nextcloud/lib/private/legacy/app.php(119): OC_App::registerAutoloading('files', '/var/www/nextcl...')
#7 /var/www/nextcloud/lib/base.php(640): OC_App::loadApps(Array)
#8 /var/www/nextcloud/lib/base.php(1056): OC::init()
#9 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#10 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#11 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu Aug 08 22:51:47.004748 2019] [authz_host:error] [pid 1624:tid 548231127440] [client 192.168.1.2:65527] AH01753: access check of 'localhost' to /img/toggle.svg failed, reason: unable to get the remote host name
[... 36 lines cut ...]
{"reqId":"XUyMUmyXmBZQiJZwRNL@MwAAABE","level":3,"time":"2019-08-08T20:55:46+00:00","remoteAddr":"192.168.1.1","user":"--","app":"PHP","method":"PROPFIND","url":"\/remote.php\/dav\/principals\/users\/frantzouille\/","message":"Error: Class 'OCA\\DAV\\Connector\\Sabre\\ExceptionLoggerPlugin' not found at \/var\/www\/nextcloud\/remote.php#57","userAgent":"Mac+OS+X\/10.14.5 (18F132) CalendarAgent\/416.5.1","version":"16.0.1.1"}
{"reqId":"XUyMUmyXmBZQiJZwRNL@NAAAAA8","level":3,"time":"2019-08-08T20:55:46+00:00","remoteAddr":"192.168.1.1","user":"--","app":"index","method":"PROPFIND","url":"\/","message":{"Exception":"RedisException","Message":"Redis server went away","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Redis.php","line":54,"function":"get","class":"Redis","type":"->","args":["ac931e8221d05c03c06af9cb8b5cf225\/core.appinfo\/var\/www\/nextcloud\/apps\/files\/appinfo\/info.xml1557931172"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/App\/InfoParser.php","line":57,"function":"get","class":"OC\\Memcache\\Redis","type":"->","args":["\/var\/www\/nextcloud\/apps\/files\/appinfo\/info.xml1557931172"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/App\/AppManager.php","line":381,"function":"parse","class":"OC\\App\\InfoParser","type":"->","args":["\/var\/www\/nextcloud\/apps\/files\/appinfo\/info.xml"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/app.php","line":551,"function":"getAppInfo","class":"OC\\App\\AppManager","type":"->","args":["files",false,null]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/App.php","line":66,"function":"getAppInfo","class":"OC_App","type":"::","args":["files"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/app.php","line":236,"function":"buildAppNamespace","class":"OC\\AppFramework\\App","type":"::","args":["files"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/app.php","line":119,"function":"registerAutoloading","class":"OC_App","type":"::","args":["files","\/var\/www\/nextcloud\/apps\/files"]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":640,"function":"loadApps","class":"OC_App","type":"::","args":[["session"]]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1056,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/index.php","line":40,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Redis.php","Line":54,"CustomMessage":"--"},"userAgent":"Mac+OS+X\/10.14.5 (18F132) CalendarAgent\/416.5.1","version":"16.0.1.1"}
[... cut ...]
{"reqId":"XUyMfyaZrVzVswZ5AqeNagAAAEU","level":3,"time":"2019-08-08T20:56:31+00:00","remoteAddr":"192.168.1.1","user":"--","app":"PHP","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications?format=json","message":"RedisException: Redis server went away at \/var\/www\/nextcloud\/lib\/private\/Memcache\/Redis.php#54","userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.5.1final (build 20181204) (Nextcloud)","version":"16.0.1.1"}

```
</details>

Could anybody help me ? Thanks !

en3r0, 2019-10-30:
Just had this issue a few days ago, so it is still happening for some reason.

pasniak, 2019-11-16:
I upgraded from 1.17.? to 1.18.0 and got the same error:
```
An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('a797db6be932177...')
#1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/private/App/AppManager.php(381): OC\App\InfoParser->parse('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/private/legacy/app.php(552): OC\App\AppManager->getAppInfo('files', false, NULL)
#4 /var/www/nextcloud/lib/private/AppFramework/App.php(66): OC_App::getAppInfo('files')
#5 /var/www/nextcloud/lib/private/legacy/app.php(237): OC\AppFramework\App::buildAppNamespace('files')
#6 /var/www/nextcloud/lib/private/legacy/app.php(120): OC_App::registerAutoloading('files', '/var/www/nextcl...')
#7 /var/www/nextcloud/lib/base.php(640): OC_App::loadApps(Array)
#8 /var/www/nextcloud/lib/base.php(1056): OC::init()
#9 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#10 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#11 {main}
```

IgnisQuisVir, 2019-11-18:
I experienced the same problem today, after upgrading via apt upgrade on armbian.
The fix with the ownership also helped me.
sudo chown -R redis:redis /var/log/redis

pleasemarkdarkly, 2020-06-03:
So this happened to me for the second time and the permissions update isn't helping. Was this ever resolved? The above referenced logs are here - https://transfersh.pleasemarkdarkly.com/tGIk1/20200603.tar.gz

pleasemarkdarkly, 2020-11-06:
Yay, or not so yay, it happened again. here are the [logs](https://transfersh.pleasemarkdarkly.com/PC8R9/20201106-1351-ncp_report-system-redis.txt).

hudecof, 2020-11-14:
I've got fresh neew install and needed also add `www-data` to the `redis` group.

```
root@XXXXXX:~# id www-data
uid=33(www-data) gid=33(www-data) groups=33(www-data),122(redis)
```