nextcloud/nextcloudpi #610: file transfer via samba creates hidden files of the files that has been transferred....

chromoxdor, 2018-08-29:
I started to use samba for local integration of the storage.
After i copied some pictures and performed a file scan i opened the gallery app and found some weird thumbnails. Somehow the filescan added hidden files of the files i copied and the gallery tries to display them. These hidden files are not existent in the folder when i access it via samba.
This is not just limited to picures... it happens with all files.

[image]

NextCloudPi version  v0.58.1
NextCloudPi image    NextCloudPi_06-27-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/McCloud/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      130G/932G
rootfs usage         2.0G/7.2G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    13.0.5.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.188.33
gateway              192.168.188.1
interface            wlan0
certificates         cloud.myftp.org
NAT loopback         no
uptime               20:52

chromoxdor, 2018-08-30:
creating those files seems to be a mac related issue as others have experienced this too:1
[https://apple.stackexchange.com/questions/136248/disable-storage-of-invisible-files-on-my-cfs-or-smb-network-storage](url)
 but since it seems to be impossible to prevent macos from doing so it would be the best to prevent the gallery.app from including hidden files into the previews....

nachoparker, 2018-08-30:
I have seen this before, yes. I suggest you open a request in the gallery repo

closed 2018-08-30 by nachoparker