nextcloud/nextcloudpi #614: [Rock64] 6 devices detected while there is 1 external HDD only

prog-amateur, 2018-09-02:
Hello, I can't format and change Nextcloud Data directory, I can't change the label too, and I can't use my HDD (actually, it is an external RAID 1, I mean 2 drives set up like it was 1 drive only) as data location. 

Message is something like `6 devices detected`, and when I unplug the drive, it shows 5 devices. 
I have already read some similar issues and I know that there is some command to do on the CLI (including `sudo-ncp-report`), however, when I try to connect through the display connected to the Rock64, I can't login as ncp and admin password, because just after accepting the login, it tells me that the account does not exist yet. 

How can I do please ? Thank you very much for your help.

nachoparker, 2018-09-02:
Enable SSH from ncp-web, and you will be able to log in. The user `ncp` is **not** for SSH access, but you can enable a password for `root`.

Once inside the terminal, you can give us the output of

```
```
, so we can try to figure out what is going on.

That would help fixing it. Thanks!

nachoparker, 2018-09-09:
I can't see your hard drive in that list. is it mounted?

prog-amateur, 2018-09-10:
@nachoparker sorry, I did not see your message. You are right, I have given the list with the hard drive UNmounted. Please find the same list with the hard drive (sda) mounted : 
`sda            8:0    0  3.7T  0 disk 
mtdblock0     31:0    0    4M  0 disk 
mtdblock1     31:1    0  256K  0 disk 
mmcblk0      179:0    0 14.5G  0 disk 
mmcblk0p1    179:1    0 14.3G  0 part /
mmcblk0boot0 179:32   0    4M  1 disk 
mmcblk0boot1 179:64   0    4M  1 disk 
mmcblk0rpmb  179:96   0    4M  0 disk `

What do you think ? Thank you very much.

nachoparker, 2018-09-11:
assuming that automount works, you will have your device available at `/media/USBdrive` or `/media/label`. Where are you trying to move it to with `nc-datadir`?

nachoparker, 2018-09-11:
Also, I fixed the detection, but I still have to try it out. If you want, you can try changing this line manually and see if that doesn't detect too many USBs.

Fix -> 6954e12

closed 2018-09-11 by nachoparker

prog-amateur, 2018-09-11:
> assuming that automount works, you will have your device available at /media/USBdrive or /media/label. Where are you trying to move it to with nc-datadir?

I have labelled the drive as `NCP`, so I try to put `/media/NCP` in the field, then I have the following message : 
```
[ nc-datadir ]
Refusing to move to the SD card. Abort
```

> Also, I fixed the detection, but I still have to try it out. If you want, you can try changing this line manually and see if that doesn't detect too many USBs.
> 
> Fix -> 6954e12

I have tryed to copy paste the lines manually but the folder `ncp-config.d` does not exist. Shall I create it (and the script too) manually ? 

Thank you

nachoparker, 2018-09-12:
> If I try to put /media/NCP in the field of nc-datadir through ncp-web gui, then I have the following message 

Yes, that is by design because we don't want people to leave ther NC data scattered on the root of the USB drive: you should select a folder inside of it. 

Admittedly the error message is not clear and also you are not the first one to try, so I implemented support for this.

Ok, since you are being patient with this, maybe you can help me test this. If you type

```
sudo ncp-update devel
```

, you will get two fixes: one for the format issue and another one for being able to move to `/media/NCP`. This will overwrite any manual changes you did to those files above.

If you try to move it now to `/media/NCP` it should move it to `/media/NCP/ncdata`. I haven't been able to test it myself but it should work and it would be very useful if you could confirm this.

> Also after rebooting, the ncp-format.sh appeard in the web GUI which shall not be normal I think

No, that must be something you did when editing files directly. You can safely delete that file from `/usr/local/etc/ncp-config.d/`. Don't worry, every time you do `sudo ncp-update devel` you will have any files that you accidentally remove restored.

Cheers

prog-amateur, 2018-09-12:
Thank you for helping me and maybe other people with Rock64 like me. I have a good and a bad news. 

- Good news first ^^ : `ncp-format.sh` disappeard from the web GUI after deleting it from SSH, then reboot, then launch `sudo ncp-update devel` from SSH, then reboot. 

- Bad news : trying to put `/media/NCP` on the field of web GUI > nc-datadir still gives the following error (even after the update v0.59.2) : 
 ```
 [ nc-datadir ]
 ERROR: the user www-data does not have access permissions over /media/NCP
 ```
So I have made a new folder names NCP : in SSH  `cd /media/NCP`, then `mkdir NCP`, and put in web GUI > nc-datadir `/media/NCP/NCP` but I have the same error message : 
```
[ nc-datadir ]
ERROR: the user www-data does not have access permissions over /media/NCP
```
I am afraid we still have nightmare to solve ^^

nachoparker, 2018-09-12:
No nightmares,  you have a permissions issue, probably because you were messing around doing things manually. Just do

```
sudo chmod ugo+x /media/NCP
```

, and try again.

Also, are you able to use `nc-format-USB` now?

prog-amateur, 2018-09-12:
YESSSSS !!!!!! Nightmare becomes a dream now it works, thank you so much. 

Using `sudo chmod ugo+x /media/NCP` and put `/media/NCP/` in the field of _web GUI > nc-datadir_ gives me : 

```
[ nc-datadir ]
Maintenance mode enabled
moving data dir from /var/www/nextcloud/data to /media/NCP/ncdata...
Nextcloud is in maintenance mode - no apps have been loaded

System config value datadirectory set to string /media/NCP/ncdata
Nextcloud is in maintenance mode - no apps have been loaded

System config value logfile set to string /media/NCP/ncdata/nextcloud.log
Nextcloud is in maintenance mode - no apps have been loaded

Maintenance mode disabled
```
And System info gives now : 
```
USB devices	sda
datadir	/media/NCP/ncdata
data in SD	no
data filesystem	ext2/ext3
data disk usage	148M/3.6T
```

So I have 2 (easy) questions : 

1. About `nc-format-USB`, will it format into **BTRFS** ? Because I would like to keep **ext4** as format. And I see on the wizard that it is proposed to format the USB drive to **ext4**. So ext4 or BTRFS ? 
2. Last but not least, will it be possible to change the iso in order to not have to put manually `sudo chmod ugo+x /media/NCP` and avoid SSH ? Because what I really like in your Raspberry iso version is that it is ready out of the box. It is very useful for beginners like me compared to Rock64 iso where we have to use SSH and some commands, what do you think ?

Thank you very much !

nachoparker, 2018-09-12:
Great!

1. I recommend BTRFS because it is more advanced and some neat features of NCP use it like `nc-snapshots`, but it is not required. ext is also supported, but if you want to use it you have to format it manually to ext4

2. You normally don't need to do anything, it works out of the box, but you have to follow the wizard if you want everything to "just work" automatically. You had to do `chmod` because probably at  some point permissions became weird during your experiments.