nextcloud/nextcloudpi #615: spdyn.de  IPv6

Slyccer, 2018-09-03:
I had problems with spdyn.de....
Found out it had to with IPv6 vs IPv4.

In nextcloudpi/etc/ncp-config.d/spDYN.sh "https://api.ipify.org/" is used to get the ip adress. But this api only provides IPv4.

I guess for the future you should use an API wich provides IPv6 (like "https://myexternalip.com/raw", wich i am using now:)  )

TomTurnschuh, 2018-09-14:
The functionality does work with IPv4. When defining a host at spdyn.de, you can add any IP, even a local one such as 192.168.1.50. When activating, NCP sends the external IP which is then linked to your domain.

However, it may take more than half an hour until the DNS cache records are updated and your cloud is accessible using the new domain. During this time, even Let's Encrypt is not possible to activate.

Still, the use of IPv6 could be discussed.

nachoparker, 2018-09-14:
We need a developer with IPv6 to verify/fix this. Not my case.

I have been thinking for a while to substitute all the DDNS custom methods that we have for `inadyn`. It might have good suppport for IPv6

TomTurnschuh, 2018-09-15:
It's actually an easy task to use IPv6 for spDYN, only the domain providing the IP address has to be changed.
However, does it make sense to definitely switch to IPv6, or are there still home routers around that do only support IPv4?

TomTurnschuh, 2018-09-15:
Maybe it's better to do it in cunjunction with #560.

nachoparker, 2018-09-16:
Can somebody with ipv4 and somebody with ipv6 help test this?

Get `v0.59.13` from `devel`
```
sudo ncp-update devel
```
, then verify that it works so I can include it for all users

Thanks!

nachoparker, 2018-09-18:
Thanks @TomTurnschuh

closed 2018-09-18 by nachoparker

traori, 2018-10-07:
Hello nachoparker,
before long time i ask you to implement the "spdyn". I mean i told you i found the follow site
https://canox.net/2016/06/die-eigene-cloud-mit-dem-raspberry-pi-und-nextcloud/
Now i saw that this guy build something here
https://gitlab.com/CANOXNET/spdynu
Did you think i can give a coorperation ? Maybe he can help something. What you mean or think?
cYu

traori, 2018-10-11:
Whats the reason why we don't use the follow adress
checkip.spdyn.de

on
https://wiki.securepoint.de/SPDyn/meineIP
they write "Der Host liefert die von der HTTP-Verbindung verwendete IPv4- oder IPv6-Adresse zurück."
Is the problem the HTTP (its not HTTPS) ?

cYu

TomTurnschuh, 2018-10-12:
@traori: The only reason that foreign addresses are used to get the IP is that it was done like this before.
We could change that. There once was a discussion on that in #635.