nextcloud/nextcloudpi #616: nc-limits warning in ncp logs

Iolaum, 2018-09-03:
The following warning appears in `/var/log/ncp.log`:

```
[ nc-limits ]
./nc-limits.sh: line 33: [[: 768M: value too great for base (error token is "768M")
Using 4 PHP threads
./nc-limits.sh: line 63: [[: 3gb: value too great for base (error token is "3gb")

```

Note sure what problem it signifies.

Paste the output of `sudo ncp-report`:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.59.0
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      367G/917G
rootfs usage         1.7G/3.6G
swapfile             /media/wd1/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    13.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   192.168.1.20
gateway              192.168.1.1
interface            eth0
certificates         $WEBSITE
NAT loopback         yes
uptime               23:34

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "$WEBSITE:443",
            "1": "192.168.1.20",
            "4": "$WEBSITE",
            "2": "$WEBSITE",
            "3": "$WEBSITE"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/$WEBSITE",
        "dbtype": "mysql",
        "version": "13.0.6.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "mail_smtpmode": "php",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "loglevel": "2",
        "logfile": "\/media\/wd1\/ncdata\/nextcloud.log",
        "log_type": "file",
        "theme": "",
        "updater.release.channel": "stable"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Sep 03 06:25:05.001706 2018] [mpm_event:notice] [pid 918:tid 1992507392] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l   configured -- resuming normal operations
[Mon Sep 03 06:25:05.001797 2018] [core:notice] [pid 918:tid 1992507392] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Move .step file of updater to backup location","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Add move avatar background job","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::info: Repair info: Repair step already executed","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Add preview cleanup background jobs","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Fix potential broken mount points","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::info: Repair info: No mounts updated","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Repair language codes","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Install new core bundle components","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Repair invalid paths in file cache","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Rename folder with user specific keys","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Add log rotate job","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::step: Repair step: Clear frontend caches","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::info: Repair info: Image cache cleared","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"13.0.5.2"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"13.0.6.1"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"13.0.6.1"}
{"reqId":"hxkdrguDabBTl6tP7QiE","level":1,"time":"2018-09-03T06:35:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"--","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"13.0.6.1"}

```
</details>

ArtJames, 2018-09-03:
I had the same problem some updates ago.
Open /etc/redis/redis.conf
At the end of the file, one of the update scripts added double entries and strange values.
Just keep 'maxmemory 768135168' and restart redis or reboot the system.
In my case, the error was gone.

Iolaum, 2018-09-05:
I checked `/etc/redis/redix.conf` and indeed it had two `maxmemory 3gb` entries. I deleted the second one. I assume this is related to the error in nc-limits line 63.

Regarding the error in nc-limits line 33 it points out to .user.ini However after a quick look:

```
# cat /var/www/nextcloud/.user.ini 
upload_max_filesize=2G
post_max_size=2G
memory_limit=768M
mbstring.func_overload=0
always_populate_raw_post_data=-1
default_charset='UTF-8'
output_buffering=0
```
I don't know what to change. The error is likely related to the line `memory_limit=768M`.

ArtJames, 2018-09-05:
Just change the memory_limit from 768M to 768135168
Seems to be a bug in nc-limits.

nachoparker, 2018-09-10:
fixed in devel, thanks guys!

closed 2018-09-10 by nachoparker