nextcloud/nextcloudpi #617: Temporary directory /var/www/nextcloud/data/tmp is not present or writable

feutl, 2018-09-04:
I installed a Debian 9.5 64BIT on a dedicated hardware and ran the script to get NextcloudPi installed.
Everything worked fine so far.

I activate the installation and tried to login using the *ncp* user.
Afterwards I migrated from my existing installation. Imported the database, imported the "old" config.php and merged it with the settings for the existing.
Using config.php I moved the data path to a NFS mount (via fstab).

It went smooth like hell :) Thanks for that.

But I felt the system was slower than the old Ubuntu 16 32BIT where I tinkered a lot. So I searched and found the following error in the Nextcloud protocol

> Temporary directory /var/www/nextcloud/data/tmp is not present or writable

I created the directory *data/tmp* and assigned it to www-data and now it is smooth like hell :)
But why was the directory not there from the beginning and where is this "setting" coming from anyhow.

nachoparker, 2018-09-10:
that's weird, that setting is in nc-nextcloud.sh, and the directory is created there.

how did you import your old instance? if your old instance wasn't NCP then that's the reason.

thanks for reporting!

feutl, 2018-09-10:
As mentioned above, I used the backup of a previous installation which I did manually.

Which means I had the config.php and a DB dump

My files are residing on NFS mount, which I mount using fstab outside the nextcloud (var/ww) directory and the config.php references this.

I still like that damn script you wrote, this was so hazzle free 👍 
Except the this and the number of fpm processes I need to adjust manually, but the rest - so smooth 
THANKS

nachoparker, 2018-09-10:
But did you copy your old NC folder to the new location? if that's the case, then that's why you don't have the `tmp` directory.

Otherwise, please details the step that took you to a situation with no `tmp` directory..

Thanks

feutl, 2018-09-10:
* I installed using the scipted.
* Activated the setup, logged into the admin panel and the nextcloud default user to test if the everything works.
* After that I stopped all services
* used mysql on the CLI to import the old database of my ubuntu / nextcloud installation
* copied the config.php settings from my "old" nextcloud installation to the "new" config.php
* one of the is the data path pointing to /media/nextcloud-data/
* adjusted the fstab to mount a NFS share as /meda/nextcloud-data/
* rebooted the machine
* Logged in to nextcloud with the old users, seeing the old data and being able to start where I stopped with the old machine :D 
* realized that I got lots of errors because of the missing tmp in ../nextcloud/data/
* created the tmp directory in ../nextcloud/data/ and errors are gone

Yes this tmp directory is not located at /media/nextcloud-data/ - the NFS share. Because the error directly pointed to  /var/www/nextcloud/data/tmp
I am not very sad about it, because a tmp on a NFS share is just not fast enough compared to an internal disk ;)

feutl, 2018-09-11:
What I do not understand is, why it raises an error?
Or not pointing to a tmp in the "new" data directory, it specifically tried to use tmp in the old even after I changed the config.php to a new data directory.
Only after creating the tmp in /var/www/nextcloud/data/tmp it stopped complaining, but this is not the "actual" data directory referenced int the config.php - so this does not make sense for me

nachoparker, 2018-09-11:
[Everything has an explanation](https://github.com/nextcloud/nextcloudpi/blob/master/etc/ncp-config.d/nc-nextcloud.sh#L240)

:D

nachoparker, 2018-09-11:
Actually it would be better to use config.php now that I think about it, I am trying not to modify global files where possible (like `/etc/php/.../php.ini`).

So you tried 

```
'tempdirectory' => '/tmp/nextcloudtemp',
```

, and it didn't work? did you do any research on the topic?

feutl, 2018-09-12:
I know, reading code makes life so much easier 😄 need to get used to it 😉 

I did not do any research on the issue. I saw the error in the logs, created the folder and changed the rights and was just happy.
Honestly, in my case, having the tmp on the data directory of config.php would be not good at all because my data directory is a NFS share and having a tmp on a NFS share just is not very performant at all.

The `tempdirectory` option in the config.php is great, but I never had it in the original config.php and even did not know it exists. My old setup did not have that issue, why so ever. So honestly, I have not used or tested it, but I gonna do now :)

Perhaps setting the tempdirectory via NCP would be great, or automatically create it on the data directory mentioned in the config.php. I just do not know the best solution, I am happy that mine is on the local disk ;) and the data directory is on the NFS share I have mounted.

THANKS FOR YOUR HELP

nachoparker, 2018-09-12:
Please, let us know if the config.php option works for you. 

You are on a very specific case with your NFS setup and that is not supported, obviously. In your case it is normal that you have to make some adjustments here and there. I think that for the average user it is good the way it is ;)

feutl, 2018-09-13:
its working
thanks for your support

SMichel72, 2018-12-09:
I have thousands of this, number is permanently growing: 

`Temporary directory {/media/.../ncdata/tmp ".""} is not present or writable`
and
`is_writable() expects parameter 1 to be a valid path, array given at /var/www/nextcloud/lib/private/TempManager.php#260`

It's a new nextcloudpi-installation but data has been moved to an external storage.

Any suggestions?

Thanks

SMichel

ovpc, 2018-12-09:
Did you try adding:

'tempdirectory' => '/tmp/nextcloudtemp',

to your config.php ?

ovpc, 2018-12-09:
Did you try adding 

> 'tempdirectory' => '/tmp/nextcloudtemp',

to your config.php?

SMichel72, 2018-12-09:
Thanks for the answer. And where can I find the config.php in a nextcloudpi installation?

ovpc, 2018-12-09:
/var/www/nextcloud/config/config.php

SMichel72, 2018-12-09:
Only the warning `Temporary directory /tmp/nextcloudtemp is not present or writable` is left.

ovpc, 2018-12-09:
Did you restart webserver?

> sudo systemctl restart apache2

SMichel72, 2018-12-09:
Yes, but the message apears about 3 times per sec.

ovpc, 2018-12-09:
You may have to create it first:

> sudo mkdir /tmp/nextcoudtemp
> sudo chown www-data /tmp/nextcoudtemp

then restart webserver

SMichel72, 2018-12-09:
Yesss! That's it. No warning or error left in the protocol.
Thank you very much!
yours
SMichel

pilepalle, 2019-01-31:
Hi, I have the same problem on my NextcloudPI. I get up to three times a second the error "Temporary directory / var / www / nextcloud / data / tmp is not present or writable" What can I do exactly here. Unfortunately, I do not really know much about Linux. The error is posted after the automatic update to the newest Nextcloud 15.0.2. Can someone help me please. Thank you

nachoparker, 2019-02-02:
@pilepalle 

```
sudo mkdir /var/www/nextcloud/data/tmp
sudo chown -R www-data: /var/www/nextcloud/data/tmp
```

feutl, 2019-02-06:
The solution from @nachoparker works but after an upgrade you run into the same issue, if you moved you data directory manually to a NFS share os another disk etc without using NCP options.

What I did to avoid it was creating a directory somewhere on the disk. I my case it was `/var/www/tmp` and made the necessary changes to the owner.
After that I changed the config.php and added
`'tempdirectory' => '/var/www/tmp/',`
This is now persistent and after upgrades I am not running into the same issues again.

pilepalle, 2019-02-09:
@nachoparker 

thank you very much. That's it. Mistakes are gone.

pilepalle, 2019-02-09:
@feutl 

Thanks for the information. I'll remember if it happens to me too.

JimmyKater, 2019-02-12:
well i discovered the same problem today after a fresh install of NC 14.0.4. (restored old data from backup) and updated to NC15.0.4

and i AM pretty sure that I once did the moving of the data-dir with an early version of ncp-web. so it's not really that I didn't use our wonderful tool it is just that ncp-web forgot about it on it's development ;-)
so wouldn't it be nice to implemet some routine that would avoid such problems in the future? like: where is the data located (nextclound does know where) and then add the missing directory there?

feutl, 2019-02-13:
I would like to have a option to set the tmp path via ncp
The thing is, even most of the development considers a raspi (or other soc) with an external hdd to be used, I use it with a NFS share, which is mounted as data and I do not want to have the tmp on the NFS share but on the local disk of my PC.