nextcloud/nextcloudpi #621: unattended upgrades need modification for armbian

PascalHeynmoeller, 2018-09-06:
I use odroid hc-1.  today from var/log/unattended-upgrades/unattended-upgrades-log
allowed origins are:
o=Raspbian,n=stretch,l=Raspbian', 'origin=Debian,codename=stretch,label=Debian-Security'

something messed up my armbian unattended-upgrades conf

/etc/apt/apt.conf.d/50unattended-upgrades is missing but there is an 20ncp-upgrades from 2018-07-14

PascalHeynmoeller, 2018-09-11:
Ola @nachoparker ,
['o=Raspbian,n=stretch,l=Raspbian', 'origin=Debian,codename=stretch,label=Debian', 'origin=Debian,codename=stretch,label=Debian-Security']

Debian Security is now inside.

nachoparker, 2018-09-11:
Thanks for doing that research!

I think that the Armbian packages are only for kernel and Armbian specific stuff, but I might be wrong. 

I suggest you run it manually

```
sudo unattended-upgrades -d
```

, and look at the log. It will tell you what matches and what not and you will see things being upgraded.

```
tail -f /var/log/unattended-upgrades
```

Haraade, 2018-09-12:
/etc/apt/apt.conf.d/02-armbian-periodic                

// Enable the update/upgrade script (0=disable)
APT::Periodic::Enable "0";

// Do "apt-get update" automatically every n-days (0=disable)
APT::Periodic::Update-Package-Lists "21";

// Do "apt-get upgrade --download-only" every n-days (0=disable)
APT::Periodic::Download-Upgradeable-Packages "0";

// Run the "unattended-upgrade" security upgrade script
// every n-days (0=disabled)
// Requires the package "unattended-upgrades" and will write
// a log in /var/log/unattended-upgrades
APT::Periodic::Unattended-Upgrade "7";

// Do "apt-get autoclean" every n-days (0=disable)
APT::Periodic::AutocleanInterval "120";

nachoparker, 2018-09-24:
LOL it seems that I already adapted unattended upgrades for Armbian some time ago. I just double-checked that it works by running `sudo unattended-upgrades -u`

closed 2018-09-24 by nachoparker