nextcloud/nextcloudpi #627: Keep language setting of the NCP web interface when updating NCP

TomTurnschuh, 2018-09-13:
After updating NCP the language of the NCP web interface gets resetted to "auto". It would be nice if the language setting could be kept on the initially set value.

chiyi4488, 2018-09-13:
What is your language? Because ncp will automatically query the language of your computer system and update it automatically. If not, you should skip to English. Or.... I don't know. This may be a bug in the language code of ncp.

TomTurnschuh, 2018-09-13:
@chiyi4488: My auto language is German, but I usually switch to English. After an NCP update, the language setting is "auto" and thus German again.

chiyi4488, 2018-09-13:
Ok, this may require someone else to help, I don't know how to write programs and fixes. Sorry.

nachoparker, 2018-09-25:
fixed in `v0.60.8`

closed 2018-09-25 by nachoparker