nextcloud/nextcloudpi #628: Modify the language option bar text to white(web)

chiyi4488, 2018-09-13:
as the picture shows

[image]
[image]
[image]
[image]
[image]

nachoparker, 2018-09-15:
that is really weird @chiyi4488. It doesn't happen to me in Firefox Linux. gi what browser are you using?

nachoparker, 2018-09-15:
anybody else seeing this?

chiyi4488, 2018-09-15:
chrome
I am not sure everyone is like this, but at least all my computers are like this.

hurrikan77, 2018-09-18:
I have this issue too.
I use vivaldi which is based on chromium as I know.

But in Firefox the languages ar shown in black.

nachoparker, 2018-09-20:
That's so weird... I mean it's just a HTML dropdown

nachoparker, 2018-10-07:
This must be a browser issue, nothing to fix

Thanks for reporting

closed 2018-10-07 by nachoparker

nachoparker, 2018-10-09:
I can't reproduce it, so if somebody has the issue please send me the fix