nextcloud/nextcloudpi #630: Issue with nc-format-USB

TomTurnschuh, 2018-09-14:
There is an issue with formatting a mounted drive:

1. Install and use NCP and NC, with data on a hard drive
2. Reinstall NCP
3. Enabling automount
4. The hard drive cannot be formated even though it is indeed mounted. The error says:
´ERROR: counted 0 devices. Please, only plug in the USB drive you want to format´

I didn't check yet with a unformated drive.

nachoparker, 2018-09-14:
mmm that was probably introduced by this commit  https://github.com/nextcloud/nextcloudpi/commit/e25f5d18628047c7b33fd912c253b01447d766fb

do you think you can investigate this? what is the output of 

```
lsblk -ln
```

for you?

TomTurnschuh, 2018-09-14:
That's what I thought as well. Thanks for the direct link.

The output of `lsblk -ln` is the following:
```
sda         8:0    0 931.5G  0 disk
sda1        8:1    0 931.5G  0 part /media/dataDrive
mmcblk1   179:0    0  14.6G  0 disk
mmcblk1p1 179:1    0  14.4G  0 part /
zram0     253:0    0    50M  0 disk /var/log
zram1     253:1    0 249.3M  0 disk [SWAP]
zram2     253:2    0 249.3M  0 disk [SWAP]
zram3     253:3    0 249.3M  0 disk [SWAP]
zram4     253:4    0 249.3M  0 disk [SWAP]
```

I'll try to solve it.

nachoparker, 2018-09-14:
fixed https://github.com/nextcloud/nextcloudpi/pull/631

thanks again @TomTurnschuh !!

closed 2018-09-14 by nachoparker

alexcanepa, 2021-01-03:
FYI, the `counted 0 devices` also happens when you plug in a drive that's APFS (Apple's new proprietary system). 

It seems the only workaround (from a Mac) is to format to FAT before plugging into the setup wizard and reformatting to Btrfs (although currently having trouble with this)

nachoparker, 2021-01-19:
@alexcanepa what is the output of `lsblk -f` in that case?

JohnAtl, 2022-09-23:
```
root@nextcloudpi:~# lsblk -ln
sda         8:0    0  3.6T  0 disk
sda1        8:1    0  200M  0 part
sda2        8:2    0  3.6T  0 part
mmcblk0   179:0    0 58.9G  0 disk
mmcblk0p1 179:1    0  256M  0 part /boot
mmcblk0p2 179:2    0 58.7G  0 part /
```

JohnAtl, 2022-09-23:
For others that might have this problem: reformatted the drive as FAT32 with GUID partition scheme, then needed to reboot the Raspberry Pi for nc-format-USB to work, even though the drive was listed as /dev/sda.

PR #631 (2018-09-14): Update nc-format-USB.sh
Resolves #630

Delete `-1` in calculation of drives number.
commit 5bc39373 (2018-09-14): nc-format-USB: drive number was off by one (#631)

Resolves #630
Delete `-1` in calculation of drives
commit a00cd0b5 (2018-09-14): nc-format-USB: drive number was off by one (#631)

Resolves #630
Delete `-1` in calculation of drives