nextcloud/nextcloudpi #635: Discussion: URLs to get IP adresses for DDYN_spDYN.sh and others

TomTurnschuh, 2018-09-16:
#### For discussion:

spDYN has its own URLs to get IP addresses in raw format, however, they don't support encryption but only HTTP. Is it still worth to use them, as the can be assumed to work as long as their DDNS service is online?
Right now, other providers for IPv4 and IPv6 are used.

nachoparker, 2018-09-20:
I don't think I understand what you are trying to say. Care to explain a bit further?

nachoparker, 2018-09-20:
Mmmm, I guess it does not matter in this case, as long as you don't leak private information (like the login credentials to your DDNS account).

I don't understand what you mean by _other providers_. Do you mean other DDNS services?

In any case, if there is an option for HTTPS we should always do that.

TomTurnschuh, 2018-09-21:
> I don't understand what you mean by other providers. Do you mean other DDNS services?

Not necessarily. Just any organisation providing such an URL that responses with the raw IP address.

nachoparker, 2018-09-22:
But we don't need to know our public IP, what we need is to tell the DDNS provider what our public IP is, so they can update their records and make our registered domain name point to our house even when the IP changes.

For that, we have to tell the DDNS provider directly, right? or am I missing something?

TomTurnschuh, 2018-09-22:
That's right. But to be able to tell the DDNS provider our public IP we first need to know it ourselves.

For spDYN there are for example the following two URLs used that report back our public IPv4 and IPv6 addresses, respectively:
https://api.ipify.org/
https://myexternalip.com/raw
The respective IP address is then sent to the DDNS provider.

Drawback is: both services/URLs seem to be provided by single persons.

The DDNS provider spDYN itself can do the same with http://checkip4.spdyn.de/ and http://checkip6.spdyn.de/, however, those URLs are not encrypted.

nachoparker, 2018-09-22:
I get you now. I don't know why we need to know the IP. We can just contact them with our identification hash and they should be able to get it from that connection in one step. That's how no-ip does it at least.

In any case, the IP is public anyway so I see no harm unless there's some other sensitive information being passed around unencrypted.

nachoparker, 2018-09-23:
Ok, in that case you can change it to whatever you like. In this case I don't think it matters too much if it's HTTP or HTTPS because the IP is public anyway.

closed 2018-09-23 by nachoparker