nextcloud/nextcloudpi #636: Issue with letsencrypt.sh and IPv6 address

TomTurnschuh, 2018-09-16:
When using an IPv6 only DDNS configuration, letsencrypt.sh shows the following output:
```
[ letsencrypt ]
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for [subdomain].spdns.org
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. [subdomain].spdns.org (http-01): urn:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching http://[subdomain].spdns.org/.well-known/acme-challenge/WFxbUkPyYZnWfJnj-iJDdRuKAoF8Z9ZohepTHbeLYmI: Error getting validation data
IMPORTANT NOTES:
- The following errors were reported by the server:

Domain: [subdomain].spdns.org
Type: connection
Detail: Fetching
http://[subdomain].spdns.org/.well-known/acme-challenge/WFxbUkPyYZnWfJnj-iJDdRuKAoF8Z9ZohepTHbeLYmI:
Error getting validation data

To fix these errors, please make sure that your domain name was
entered correctly and the DNS A record(s) for that domain
contain(s) the right IP address. Additionally, please check that
your computer has a publicly routable IP address and that no
firewalls are preventing the server from communicating with the
client. If you're using the webroot plugin, you should also verify
that you are serving files from the webroot path you provided.
```
Using an IPv4 address with the same settings works well.

TomTurnschuh, 2018-09-16:
An old version of letsencrypt/certbot without IPv6 support is used. Instructions for installation of a new version (with a response indicating IPv6 support) can be found [here](https://certbot.eff.org/lets-encrypt/debianstretch-apache).
However, I still didn't manage to get it running.

TomTurnschuh, 2018-09-16:
Related to #293.

TomTurnschuh, 2018-09-21:
@Slyccer: Can you verify that the app `letsencrypt` does not work with IPv6 addresses?
According to #615 you are using spDYN with an IPv6 address, right?

nachoparker, 2018-09-27:
@TomTurnschuh  I have another little nit with spDYN... how do you feel about using wget instead of curl? the main reason is to be more consistent and also to save a few MiB in the docker images, that need to be as slim as possible

TomTurnschuh, 2018-09-27:
I don't mind using wget. I just didn't change anything on that part yet.
I'll probably also simplify the code a bit, as it has quite some unnecessary and unused overhead.

nachoparker, 2018-10-27:
awesome thanks! I think we can wait until Debian supports LE with ipv6, unless somebody wants to provide a fix

closed 2018-10-27 by nachoparker