nextcloud/nextcloudpi #638: TUI: Enabled apps don't have a name

TomTurnschuh, 2018-09-18:
In the TUI every activated app/option does not show a name in the overview, as the following screenshot shows.

[image]

When disabling an app, its name appears again.

nachoparker, 2018-09-19:
That happens when you are missing fonts, I think you need to install `locales-all` or similar, but where are you seeing this? raspberry pi?

TomTurnschuh, 2018-09-19:
Thanks, now it looks much better. There was no locale set.
It's on an ODROID device where I installed your ODROIDHC2 image.

BTW: Is there a difference in performance when generating all locales instead of only a single one?

nachoparker, 2018-09-19:
no difference in performance, but locales can take quite a bit of space. If that happened with the image that means we have to set the locale as part of the build

nachoparker, 2018-09-24:
Did you manage to fix it? what did you have to change?