nextcloud/nextcloudpi #641: NextCloud 14 Upgrade

bernd-wechner, 2018-09-21:
Given NextCloud 14 is out now. I posted a question here:

https://help.nextcloud.com/t/whats-the-timeline-of-a-nextcloudpi-distro-of-nextcloud-14/36981

at the beginning of the week. But seems no activity there. To wit, posting it here. 

Given NextCloud 14 is released, is there a timeline on NextCloudPi updating to 14? Just curious.

bernd-wechner, 2018-09-21:
> In the future, please use the forum for questions

Gladly. Can you clarify by linking to "the forum". Is the NextCloud forum I linked to above? If so, only reason I posted here is no answer there and here looks vibrant. Your clarification has been very helpful thanks! (I'm not feeling very experimental with my NextCloud server right now, so happy to wait till it's formally adopted by the NCP project I think).

nachoparker, 2018-09-30:
The upgrade is in testing right now, you can test it or wait a couple days