nextcloud/nextcloudpi #650: No space left in ZRAM

TomTurnschuh, 2018-09-26:
### Issue
Today I got the issue that zram has no space left.
An error occurs for example when running `nc-fix-permissions` (or any other ncp app such as `nc-scan` or `nc-ramlogs`), as the following indicates:
```
Launching nc-fix-permissions
/usr/local/etc/library.sh: line 117: echo: write error: No space left on device
Cannot load Zend OPcache - it was already loaded
tee: /var/log/ncp.log: No space left on device
fixing permissions in /media/dataDrive/ncdata... done
/usr/local/etc/library.sh: line 120: echo: write error: No space left on device
Done. Press any key...
```
The filesystem usage is the following:
```
Filesystem      Size  Used Avail Use% Mounted on
udev            932M     0  932M   0% /dev
tmpfs           200M   22M  179M  11% /run
/dev/mmcblk0p1   15G  3.6G   11G  26% /
tmpfs           999M     0  999M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           999M     0  999M   0% /sys/fs/cgroup
tmpfs           999M  4.0K  999M   1% /tmp
/dev/zram0       49M   48M     0 100% /var/log
/dev/sda1       932G   51G  880G   6% /media/USBdrive
tmpfs           200M     0  200M   0% /run/user/1001

```
I detected this issue after updating NCP from v0.60.3 to v0.60.8, however, I already had an issue with NC CSync access (`CSync failed to access`) before, which could be related.

### Temporary resolution
Deleting the log file `/var/log/ncp.log` did resolve the issue for the moment. However, still more than 70 % of the available zram space are used.

TomTurnschuh, 2018-09-27:
> It also might be useful to see what else is taking up space in `/var/log`

This might be interesting: The other big file is `mail.info` with 43 out of total 48 MB in this folder (at the moment). This will be the next file to clear on our device, as is is growing quickly.

nachoparker, 2018-09-27:
Interesting... we might want to tweak the logrotate for that too. I think it is better that than to take more RAM for logs

TomTurnschuh, 2018-09-27:
> I am almost ready with the new release, do you think you will have time to have a look at it?

I can try – if I manage to install it on my devices. Until now I always used your ready-to-use images to start with.

nachoparker, 2018-09-27:
you can just set it up in your pi, and when you are happy with it, paste here the config file...

... or modify `ncp.sh`/`update.sh` and send a PR, either way ;)

TomTurnschuh, 2018-09-28:
@nachoparker: I just found your testing downloads. :smile:

nachoparker, 2018-09-29:
Yes! we are testing the new release

https://help.nextcloud.com/t/nextcloudpi-php7-2-nc14-0-1-a-call-for-testing/

It was also announced on Telegram

nachoparker, 2018-10-06:
I looked at your logs, there's a lot of

```
Oct  4 17:18:47  postfix/local[25723]: warning: hash:/etc/aliases is unavailable. open database /etc/aliases.db: No such file or directory
Oct  4 17:18:47  postfix/local[25723]: warning: hash:/etc/aliases: lookup of 'www-data' failed
Oct  4 17:18:47  postfix/local[25725]: warning: hash:/etc/aliases is unavailable. open database /etc/aliases.db: No such file or directory
Oct  4 17:18:47  postfix/local[25725]: warning: hash:/etc/aliases: lookup of 'www-data' failed
Oct  4 17:18:47  postfix/local[25722]: warning: hash:/etc/aliases is unavailable. open database /etc/aliases.db: No such file or directory
Oct  4 17:18:47  postfix/local[25722]: warning: hash:/etc/aliases: lookup of 'root' failed
Oct  4 17:18:47  postfix/local[25727]: warning: hash:/etc/aliases is unavailable. open database /etc/aliases.db: No such file or directory
Oct  4 17:18:47  postfix/local[25727]: warning: hash:/etc/aliases: lookup of 'root' failed
```
Can you try running

```
newaliases
```

, see if it goes away?

My logs were also quite big. I think we should watch the logs and implement sane logrotate options for each element in the system. I like that better than increasing the amount of RAM for logs

TomTurnschuh, 2018-10-06:
Those warnings disappeared.
However, there are still many log entries. Their timestamps (every 5 minutes) may point to a cron job. I didn't find any with a 5 minute rhythm, though.

The following extracts show the entries, wherof all appear multiple times in each timestamp. Maybe you see where they come from and if they're necessary?

#### mail.info **and** mail.log

```
Oct  6 20:01:44 nextcloudpi postfix/qmgr[1908]: A9700CCF: from=<>, size=2792, nrcpt=1 (queue active)
Oct  6 20:01:44 nextcloudpi postfix/qmgr[1908]: AE585C0E: from=<www-data@nextcloudpi.localdomain>, size=2865, nrcpt=1 (queue active)
Oct  6 20:01:44 nextcloudpi postfix/local[3707]: warning: dict_nis_init: NIS domain name not set - NIS lookups disabled
Oct  6 20:01:44 nextcloudpi postfix/local[3707]: A9700CCF: to=<www-data@nextcloudpi.localdomain>, relay=local, delay=218544, delays=218544/0.04/0/0.44, dsn=2.0.0, status=sent (delivered to mailbox)
Oct  6 20:01:44 nextcloudpi postfix/qmgr[1908]: A9700CCF: removed
```

#### mail.warn
```
Oct  6 19:56:44 nextcloudpi postfix/local[1911]: warning: dict_nis_init: NIS domain name not set - NIS lookups disabled
```

The complete files I copied again to the cloud folder as before.

TomTurnschuh, 2018-10-06:
> I think we should watch the logs and implement sane logrotate options for each element in the system. I like that better than increasing the amount of RAM for logs

I agree that increasing the amount of RAM for logs is not the best idea. Even with more RAM it would be full some time.

nachoparker, 2018-10-07:
I implemented logrotate limits to keep log size under control in `v0.62.10`

closed 2018-10-07 by nachoparker