nextcloud/nextcloudpi #652: Trailing slash in CLI URL required

TomTurnschuh, 2018-09-28:
In order for `nc_prettyURL` to work, the domain in `overwrite.cli.url` must have a trailing `/` – at least, if the domain has no path, cf. nextcloud/server#10678. Hence, a trailing `/` should be added in all apps where `overwrite.cli.url` is defined.

ignacionunez, 2018-09-28:
I just upgraded to NC14.0.1 and PHP7.2 and noticed that prettyURLs are not working. Will have to look into this later, maybe that's the reason.

They did work before the upgrade.

On a different note @TomTurnschuh you might want to join the telegram group to have a more direct communication channel. Don't need to, but you can ;)

TomTurnschuh, 2018-09-28:
> I just upgraded to NC14.0.1 and PHP7.2 and noticed that prettyURLs are not working. Will have to look into this later, maybe that's the reason.

Yes, this is probably the reason. It works if you add the trailing slash in `overwrite.cli.url`. I opened a pull request to catch this error as a first step.

nachoparker, 2018-10-01:
fixed in devel, thanks!

closed 2018-10-01 by nachoparker