nextcloud/nextcloudpi #664: Cron not working in docker

theCalcaholic, 2018-10-01:
On docker, cron is not being executed. I believe the issue is the same as described (and solved) in the following SO-question:

https://unix.stackexchange.com/questions/453006/getting-cron-to-work-on-docker

I'm currently testing if this fix works and if it needs to be executed after each container restart.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.60.8
distribution         Debian GNU/Linux 9 \n \l
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      126G/295G
rootfs usage         126G/295G
swapfile             /dev/sda3
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    14.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      down
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               4days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.178.115",
            "4": "cloud.abr-harburg.de"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/cloud.abr-harburg.de:553",
        "dbtype": "mysql",
        "version": "14.0.1.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
[... 90 lines cut ...]
```

{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:46:39+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No mounts updated","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:46:39+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair invalid paths in file cache","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:46:39+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add log rotate job","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:46:39+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Clear frontend caches","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:46:39+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Image cache cleared","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:46:56+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair pending cron jobs","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Repaired 1 pending cron job(s).","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:03+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:05+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"13.0.5.2"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:05+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"14.0.1.1"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:05+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceActive: Maintenance mode is kept active","userAgent":"--","version":"14.0.1.1"}
{"reqId":"oBKTz1WzIhy2Sw2k7fJV","level":1,"time":"2018-09-27T20:47:05+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"14.0.1.1"}
{"reqId":"W61CsawRAAIAAE7TUJYAAABQ","level":3,"time":"2018-09-27T20:51:07+00:00","remoteAddr":"172.17.0.1","user":"admin","app":"index","method":"GET","url":"\/index.php\/apps\/files\/","message":{"Exception":"Doctrine\\DBAL\\Exception\\UniqueConstraintViolationException","Message":"An exception occurred while executing 'INSERT INTO `oc_filecache` (`mimepart`,`mimetype`,`mtime`,`size`,`etag`,`storage_mtime`,`permissions`,`parent`,`checksum`,`path_hash`,`path`,`name`,`storage`) SELECT ?,?,?,?,?,?,?,?,?,?,?,?,? FROM `oc_filecache` WHERE `storage` = ? AND `path_hash` = ? HAVING COUNT(*) = 0' with params [\"7\", \"15\", 1538081467, 0, \"f9f7bd07132d358f41166ecc3fa0af1d\", 1538081467, 27, 82, \"\", \"222089b1af5cf1a176ab900221738166\", \"appdata_ocs86rkb5syt\\\/css\\\/core\\\/c064-1231-jquery-ui-fixes.css.gzip\", \"c064-1231-jquery-ui-fixes.css.gzip\", 2, 2, \"222089b1af5cf1a176ab900221738166\"]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1062 Duplicate entry '2-222089b1af5cf1a176ab900221738166' for key 'fs_storage_path_hash'","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":128,"function":"convertException","class":"Doctrine\\DBAL\\Driver\\AbstractMySQLDriver","type":"->","args":["An exception occurred while executing 'INSERT INTO `oc_filecache` (`mimepart`,`mimetype`,`mtime`,`size`,`etag`,`storage_mtime`,`permissions`,`parent`,`checksum`,`path_hash`,`path`,`name`,`storage`) SELECT ?,?,?,?,?,?,?,?,?,?,?,?,? FROM `oc_filecache` WHERE `storage` = ? AND `path_hash` = ? HAVING COUNT(*) = 0' with params [\"7\", \"15\", 1538081467, 0, \"f9f7bd07132d358f41166ecc3fa0af1d\", 1538081467, 27, 82, \"\", \"222089b1af5cf1a176ab900221738166\", \"appdata_ocs86rkb5syt\\\/css\\\/core\\\/c064-1231-jquery-ui-fixes.css.gzip\", \"c064-1231-jquery-ui-fixes.css.gzip\", 2, 2, \"222089b1af5cf1a176ab900221738166\"]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1062 Duplicate entry '2-222089b1af5cf1a176ab900221738166' for key 'fs_storage_path_hash'",{"errorInfo":["23000",1062,"Duplicate entry '2-222089b1af5cf1a176ab900221738166' for key 'fs_storage_path_hash'"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doct [... cut ...] lass":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"\/var\/www\/nextcloud\/cron.php","line":123,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"\/var\/www\/nextcloud\/apps\/survey_client\/lib\/BackgroundJobs\/MonthlyReport.php","Line":43,"CustomMessage":"--"},"userAgent":"--","version":"14.0.1.1"}

```
</details>

nachoparker, 2018-10-02:
hi Tobias,

cron works fine

```
root@5f85a9183673:/# ps -eHF 
UID        PID  PPID  C    SZ   RSS PSR STIME TTY          TIME CMD
root        12     0  0  1467  3288  14 01:02 pts/0    00:00:00 bash
root       404    12  0  5075  2756   9 01:02 pts/0    00:00:00   ps -eHF
root         1     0  0  1410  2908   6 01:02 ?        00:00:00 /bin/bash /run-parts.sh
root        82     1  0 95829 19244   7 01:02 ?        00:00:00   php-fpm: master process (/etc/php/7.2/fpm/php-fpm.conf)
www-data    83    82  0 95829 11880   5 01:02 ?        00:00:00     php-fpm: pool www
www-data    84    82  0 95829 11880   3 01:02 ?        00:00:00     php-fpm: pool www
root        90     1  0 21636  7776   6 01:02 ?        00:00:00   /usr/sbin/apache2 -k start
www-data   258    90  0 595307 6132   6 01:02 ?        00:00:00     /usr/sbin/apache2 -k start
www-data   259    90  0 595307 6132   1 01:02 ?        00:00:00     /usr/sbin/apache2 -k start
mysql       92     1 20 176449 112676 0 01:02 ?        00:00:02   mysqld
root       120     1  0  1445  2984   5 01:02 ?        00:00:00   /bin/bash /etc/services-enabled.d/020nextcloud start
root       229   120  0  1479  2256   4 01:02 ?        00:00:00     /bin/bash /etc/services-enabled.d/020nextcloud start
root       234   229  0  1482  2808   4 01:02 ?        00:00:00       /bin/bash /etc/services-enabled.d/020nextcloud start
root       248   234  0  6748  3048   7 01:02 ?        00:00:00         sudo -u www-data php occ maintenance:install --database mysql --database-name nextcloud --database-user ncadmin --database-pass default --admin-user ncp --admin-pass ownyourbits
www-data   249   248  9 118032 57824  6 01:02 ?        00:00:00           php occ maintenance:install --database mysql --database-name nextcloud --database-user ncadmin --database-pass default --admin-user ncp --admin-pass ownyourbits
root       235   229  0   570   772   6 01:02 ?        00:00:00       tee -a /var/log/ncp.log
redis      127     1  0  7939  2380   8 01:02 ?        00:00:00   redis-server 127.0.0.1:0
root       131     1  0  3413  2160  13 01:02 ?        00:00:00   cron
root       225     1  0 15358  4496   6 01:02 ?        00:00:00   /usr/lib/postfix/sbin/master -w
postfix    226   225  0 15870  7084   1 01:02 ?        00:00:00     pickup -l -t unix -u -c
postfix    227   225  0 15882  7148   3 01:02 ?        00:00:00     qmgr -l -t unix -u
```

thanks for the feedback ;)

closed 2018-10-02 by nachoparker

theCalcaholic, 2018-10-02:
Only because it's running doesn't mean it works. In the linked thread cron is running as well, but doesn't execute the cron jobs. On my ncp instance I have the same issue and the described fix solves it.

nachoparker, 2018-10-02:
I just tested this again, spawn a brand new container, and wait 5 minutes

[image]

nachoparker, 2018-10-02:
I am running docker on BTRFS, what are you running it on? maybe that's the issue