nextcloud/nextcloudpi #665: spdns error during updating the token

hurrikan77, 2018-10-01:
if I enter a new update token and start the procedure with run button i got the error message

[ DDNS_spDYN ]
[badauth] An invalid username and / or password was entered.

And it seams that the DDNS update functionality doesn't work correct.

TomTurnschuh, 2018-10-01:
There is an update yet to come. (#642)
If you have SSH access, please run the following code as root:
```
rm -r /usr/local/etc/spdnsupdater
rm /usr/local/etc/ncp-config.d/DDNS_spDYN.sh
ncp-update
```
This should give you the updated version of DDNS_spDYN.

hurrikan77, 2018-10-01:
removal and update hase worked as expected.

After entering the domain and  the token ones more and run I got the following error message

[ DDNS_spDYN ]
/usr/local/etc/spdnsupdater/spdnsUpdater.sh: line 63: syntax error near unexpected token `('
/usr/local/etc/spdnsupdater/spdnsUpdater.sh: line 63: `	registeredIP=\$(nslookup ${DOMAIN_}|tail -n2|grep A|sed s/[^0-9.]//g)'

TomTurnschuh, 2018-10-01:
OK, it seems that this line is not valid anymore.
We need to replace `registeredIP=\$(nslookup ${DOMAIN_}|tail -n2|grep A|sed s/[^0-9.]//g)` by `registeredIP=$(dig +short "$DOMAIN_"|tail -n1)` in the file `/usr/local/etc/spdnsupdater/spdnsUpdater.sh`.
Unfortunately, I didn't manage to find a working `SED` command yet.

traori, 2018-10-11:
Hi,
thats my solution

registeredIP=$(nslookup $1|sed s/[^0-9.]//g|tail -n1)

I can't read the Code but i tray something like change "$DOMAIN_" with $1
The rest i copied from here
https://groups.google.com/forum/#!msg/bifferboard/yJ5Znw6TsQQ/vmiCmuPAU6oJ
@TomTurnschuh maybe you can read this part of code and can say you can use it or not.
I don't know why is "$DOMAIN_", because at the beginning you write the Domain in $1.
Have a look - for me works in the moment :-)

cYu

TomTurnschuh, 2018-10-12:
Oops, thanks @traori. I didn't realize that. However, `$DIG` seems to be better suited anyway.

nachoparker, 2018-10-24:
I believe this is fixed now

closed 2018-10-24 by nachoparker

hurrikan77, 2018-10-24:
Yes workes like expected. Thanks both of you for fixing.