nextcloud/nextcloudpi #670: nc-autoupdate-ncp fails after updating to 0.62.2

Iolaum, 2018-10-03:
I have managed to update NCP to 0.62.2 successfully. Since I have `nc-autoupdate-nc` and `nc-autoupdate-ncp` turned on I left ncp to further keep updating itself. However next day when I logged in the overview administration panel showed me the following error:

> Some files have not passed the integrity check. Further information on how to resolve this issue can be found in the documentation. (List of invalid files… / Rescan…)

The missing files are:

```
	- EXCEPTION
		- OC\IntegrityCheck\Exceptions\InvalidSignatureException
		- Signature data not found.
```
Here are the relevant ncp logs:

<details>
<summary>`ncp.log`</summary>

```
[ nc-limits ]
Using 4 PHP threads

[ nc-autoupdate-nc ]
automatic Nextcloud updates enabled

[ nc-limits ]
Using 4 PHP threads

[ncp-update-nc]
Current   Nextcloud version 13.0.6.1
Available Nextcloud version 14.0.1
Download Nextcloud 14.0.1...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup base files...
compressing backup file...
backup /var/www//nextcloud-bkp_20181003_1538547986.tar.gz generated
Nextcloud is in maintenance mode - no apps have been loaded

Maintenance mode disabled
Install Nextcloud 14.0.1...
Restore apps...
 * admin_notifications
 * previewgenerator
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Set log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: admin_notifications
Updating <federatedfilesharing> ...
Updated <federatedfilesharing> to 1.4.0
Updating <files_pdfviewer> ...
Updated <files_pdfviewer> to 1.3.2
Updating <files_texteditor> ...
Updated <files_texteditor> to 2.6.0
Updating <files_videoplayer> ...
Updated <files_videoplayer> to 1.3.0
Updating <gallery> ...
Updated <gallery> to 18.1.0
Updating <password_policy> ...
Updated <password_policy> to 1.4.0
Updating <provisioning_api> ...
Updated <provisioning_api> to 1.4.0
Updating <serverinfo> ...
Updated <serverinfo> to 1.4.0
Updating <survey_client> ...
[... 248 lines cut ...]
```

{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove shares of a users root folder","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Move .step file of updater to backup location","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Fix potential broken mount points","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No mounts updated","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair invalid paths in file cache","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add log rotate job","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Clear frontend caches","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:49+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Image cache cleared","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:40:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair pending cron jobs","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Repaired 1 pending cron job(s).","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"14.0.1.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"14.0.1.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WOW1LNsJ07y2Ih4qxtAa","level":3,"time":"2018-10-03T06:42:41+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/apps\/news\/appinfo\/signature.json): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Helpers\/FileAccessHelper.php#41","userAgent":"--","version":"14.0.1.1"}

```
</details>

Iolaum, 2018-10-04:
The bug seems to have happened from the second line of the news app fix:

```
# workaround for updating news to NC14 - bug https://github.com/nextcloud/news/issues/327
F=/var/www/nextcloud/apps/news/vendor/ezyang/htmlpurifier/maintenance/.htaccess
[[ -f "$F" ]] || echo "Deny from all" > "$F" || true
sudo -u www-data php nextcloud/occ integrity:check-app --path=/var/www/nextcloud/apps/news news || true
```

I was surprised by this because I don't have the news app installed.

So I tried installing it from the NC UI. This fixed the issue with the integrity check. After disabling and removing the app the issue remained fixed.

I think the problem is that the integrity check shouldn't run if the app is not installed.

While looking at the problem I managed to screw my installation :(

```
# ncc integrity:check-core
  - INVALID_HASH:
    - .htaccess:
      - expected: 70183f641f39fc5f43065aeb15b86ac99e86b82310d5467e1c4d935df30acc6155c0905f84f59616885275c19ffacea59e354a87a5c97c94f50d50d4eff0370e
      - current: 0b3e83d629059781b820fa5157237d743f0faa345157433b55f245a72969c8d077a39568ab377f9c5067f09fbb553e44153f2d8a0026debcb128eabd58b6e44f
    - .user.ini:
      - expected: 467d4f028c447895716a2b7859ed6e569f8ee34b87b51d73dab2e6a9ca4fbe63172c7be3e365ae864a60408286afcce852dd38ee544b61685ed4ee5e021fecb0
      - current: a89e3effdee1adcfb0f40c033674c1f927e5568f0007c1c8c0d57df9cfa0c74d5ec8bef1698944b48469c05e7f9fb795d6e43bf0a64d1223d58db0fb295fa62b

```
Hopefully they 'll get fixed after next upgrade.

nachoparker, 2018-10-04:
Hi,

There was nothing wrong, it is just that you didn't have the news app installed, so the fix failed.

Still, the upgrade went through fine, but after that you "messed it up".

Just restore your files from the nextcloud/server repo and re-run the integrity check

nachoparker, 2018-10-04:
Edit: nevermind, you are right if the app is not installed it should not try the fix. I'll fix that too

thanks!

closed 2018-10-04 by nachoparker