nextcloud/nextcloudpi #671: Internal Server Error after Update

mcdogas, 2018-10-04:
Hi Gurus,

my nextcloudpi has no logon panel anymore after last update, only this:

>Internal Server Error
>
>The server encountered an internal error and was unable to complete your request.
>Please contact the server administrator if this error reappears multiple times, please include the >technical details below in your report.
>More details can be found in the server log.

Has someone a suggestion how i can fix this?!

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded

NextCloudPi version  v0.62.5
NextCloudPi image    NextCloudPi_06-27-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda sdb 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      1.3T/2.7T
rootfs usage         2.2G/111G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:52

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.58",
            "2": "192.168.1.59",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
[... 98 lines cut ...]
```

{"reqId":"W7ZMjH8AAQEAAA0@XQUAAFcT","level":3,"time":"2018-10-04T17:23:24+00:00","remoteAddr":"192.168.1.31","user":"--","app":"index","method":"GET","url":"\/","message":{"Exception":"OC\\HintException","Message":"Memcache \\OC\\Memcache\\Redis not available for local cache","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":502,"function":"__construct","class":"OC\\Memcache\\Factory","type":"->","args":["cdc396a5252edbf4f55856091e8c88b7",{"__class__":"OC\\Log"},"\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":171,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":1513,"function":"query","class":"OC\\ServerContainer","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":686,"function":"getMemCacheFactory","class":"OC\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":171,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":1703,"function":"query","class":"OC\\ServerContainer","type":"->","args":["AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/app.php","line":342,"function":"getAppManager","class":"OC\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/app.php","line":113,"function":"getEnabledApps","class":"OC_App","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":654,"function":"loadApps","class":"OC_App","type":"::","args":[["session"]]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/index.php","line":40,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Factory.php","Line":99,"Hint":"Is the matching PHP module installed and enabled?","CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:62.0) Gecko\/20100101 Firefox\/62.0","version":"14.0.1.1"}
{"reqId":"W7ZMjH8AAQEAAA0@XQUAAFcT","level":3,"time":"2018-10-04T17:23:24+00:00","remoteAddr":"192.168.1.31","user":"--","app":"index","method":"GET","url":"\/","message":{"Exception":"OC\\HintException","Message":"Memcache \\OC\\Memcache\\Redis not available for local cache","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":502,"function":"__construct","class":"OC\\Memcache\\Factory","type":"->","args":["cdc396a5252edbf4f55856091e8c88b7",{"__class__":"OC\\Log"},"\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":171,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utilit [... cut ...] mplate\\Base","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/index.php","line":61,"function":"printExceptionErrorPage","class":"OC_Template","type":"::","args":[{"__class__":"OC\\HintException"},500]}],"File":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Factory.php","Line":99,"Hint":"Is the matching PHP module installed and enabled?","CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:62.0) Gecko\/20100101 Firefox\/62.0","version":"14.0.1.1"}

```
</details>

nachoparker, 2018-10-04:
You seem to have a problem with redis. Can we see `/var/log/ncp.log`, and the output of `apt show php-redis`?

Henriktor, 2018-10-05:
@nachoparker You have to plan for upgrades better. People that previously used `php7.0-fpm` will not be able to update and get this error because you changed to `php7.2-fpm` resulting in an Server Internal error and will leave the NextcloudPi in an unusable state. 

Please fix backwards compability when doing upgrades like this!

Thanks for a great product otherwise.

#### LOGS
```
pi@KRONOS:~ $ tail /var/log/ncp.log 

[ nc-limits ]
Using 729172992B for PHP
grep: /etc/php/7.2/fpm/pool.d/www.conf: No such file or directory
Using 4 PHP threads
sed: can't read /etc/php/7.2/fpm/pool.d/www.conf: No such file or directory
sed: can't read /etc/php/7.2/fpm/pool.d/www.conf: No such file or directory
sed: can't read /etc/php/7.2/fpm/pool.d/www.conf: No such file or directory
Using 729172992B for Redis

pi@KRONOS:~ $ apt install php7.2-fpm  
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
pi@KRONOS:~ $ sudo apt install php7.2-fpm 
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
pi@KRONOS:~ $ sudo apt -f install  
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
pi@KRONOS:~ $ sudo apt -f install ^C
pi@KRONOS:~ $ sudo dpkg --configure -a
pi@KRONOS:~ $ sudo apt -f install 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_US.utf8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
Setting up php7.0-fpm (7.0.30-0+deb9u1) ...
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
NOTICE: Not enabling PHP 7.0 FPM by default.
NOTICE: To enable PHP 7.0 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php7.0-fpm
NOTICE: You are seeing this message because you have apache2 package installed.
W: APT had planned for dpkg to do more than it reported back (0 vs 4).
   Affected packages: php7.0-fpm:armhf
```

```
601:M 04 Oct 18:05:15.245 * Background saving terminated with success
601:M 04 Oct 18:10:16.017 * 10 changes in 300 seconds. Saving...
601:M 04 Oct 18:10:16.018 * Background saving started by pid 8894
8894:C 04 Oct 18:10:16.206 * DB saved on disk
8894:C 04 Oct 18:10:16.208 * RDB: 0 MB of memory used by copy-on-write
601:M 04 Oct 18:10:16.219 * Background saving terminated with success
601:M 04 Oct 18:15:17.097 * 10 changes in 300 seconds. Saving...
601:M 04 Oct 18:15:17.098 * Background saving started by pid 8908
8908:C 04 Oct 18:15:17.550 * DB saved on disk
8908:C 04 Oct 18:15:17.552 * RDB: 0 MB of memory used by copy-on-write
[... 189 lines cut ...]
                _._                                                  
           _.-``__ ''-._                                             
      _.-``    `.  `_.  ''-._           Redis 3.2.6 (00000000/0) 32 bit
  .-`` .-```.  ```\/    _.,_ ''-._                                   
 (    '      ,       .-`  | `,    )     Running in standalone mode
 |`-._`-...-` __...-.``-._|'` _.-'|     Port: 0
 |    `-._   `._    /     _.-'    |     PID: 474
  `-._    `-._  `-./  _.-'    _.-'                                   
 |`-._`-._    `-.__.-'    _.-'_.-'|                                  
 |    `-._`-._        _.-'_.-'    |           http://redis.io        
  `-._    `-._`-.__.-'_.-'    _.-'                                   
 |`-._`-._    `-.__.-'    _.-'_.-'|                                  
 |    `-._`-._        _.-'_.-'    |                                  
  `-._    `-._`-.__.-'_.-'    _.-'                                   
      `-._    `-.__.-'    _.-'                                       
          `-._        _.-'                                           
              `-.__.-'                                               

474:M 05 Oct 10:55:41.016 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
474:M 05 Oct 10:55:41.016 # Server started, Redis version 3.2.6
474:M 05 Oct 10:55:41.214 * DB loaded from disk: 0.198 seconds
474:M 05 Oct 10:55:41.214 * The server is now ready to accept connections at /var/run/redis/redis.sock
```

Updating from 14.0.0 --> 14.0.1

enoch85, 2018-10-05:
@nachoparker Do you need help with backwards compatibility? We changed from PHP to PHP-FPM 7.2 in the @nextcloud/vm a while a go. There are some things to consider.

Windyo, 2018-10-05:
Try moving the redis dump @mcdogas `sudo mv /var/lib/redis/dump.rdb /var/lib/redis/dump.rdbbak`

i've seen a few cases where redis dump gets borked after shutdown after update.
If that doesn't fix it we'll need the output of `sudo service redis-server start`

nachoparker, 2018-10-06:
I'll close as fixed, fixed here -> https://github.com/nextcloud/nextcloudpi/issues/669#issuecomment-427574912

closed 2018-10-06 by nachoparker

nachoparker, 2018-10-06:
@enoch85 thank you very much for offering.

I didn't touch any configuration, I am not aware of any configuration changes for PHP7.2, if you know of any please let me know.

My only struggle with this was that while most people take the Sury packages to bump to 7.2, I couldn't do this because there is no php7.2-redis for armhf.

So I had to take them from Debian Buster, which is not ideal at all because it is more tricky to mix packages from two releases. Then the other problem is that there is no Raspbian Buster, [it doesn't exist yet](http://archive.raspberrypi.org/debian/dists/), so for Raspbian I had to add the Debian keyring and get the sources from Debian Buster in order to be able to install php7.2-redis.

It would have been much easier if I only supported x86

Again, thanks!

enoch85, 2018-10-06:
@nachoparker OK, I understand. rPI is a bit tricky indeed.

Glad it's sorted out at least!