nextcloud/nextcloudpi #672: php7.2-fpm not started after reboot

TomTurnschuh, 2018-10-04:
Since the recent update, after every reboot, the service `php7.2-fpm` has to be started manually using `service php7.2-fpm start`.

I assume, this should be done automatically.

TomTurnschuh, 2018-10-04:
Closes #673

nachoparker, 2018-10-04:
`systemctl status php7.2-fpm`?

TomTurnschuh, 2018-10-05:
The following are the answers some minutes after reboot. The services seem to be stopped again some seconds after starting.

<details>
<summary>systemctl status php7.2-fpm</summary>

```
● php7.2-fpm.service - The PHP 7.2 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.2-fpm.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2018-10-05 09:31:57 CEST; 5min ago
     Docs: man:php-fpm7.2(8)
  Process: 1102 ExecStart=/usr/sbin/php-fpm7.2 --nodaemonize --fpm-config /etc/php/7.2/fpm/php-fpm.conf (code=exited, status=0/SUCCESS)
 Main PID: 1102 (code=exited, status=0/SUCCESS)
   Status: "Processes active: 1, idle: 7, Requests: 8, slow: 0, Traffic: 0.2req/sec"

Oct 05 09:31:23 nextcloudpi systemd[1]: Starting The PHP 7.2 FastCGI Process Manager...
Oct 05 09:31:26 nextcloudpi systemd[1]: Started The PHP 7.2 FastCGI Process Manager.
Oct 05 09:31:57 nextcloudpi systemd[1]: Stopping The PHP 7.2 FastCGI Process Manager...
Oct 05 09:31:57 nextcloudpi systemd[1]: Stopped The PHP 7.2 FastCGI Process Manager.
```
</details>

<details>
<summary>systemctl status mysqld</summary>

```
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2018-10-05 09:32:00 CEST; 6min ago
  Process: 2303 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 2301 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 2258 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 1383 ExecStartPre=/bin/sleep 20 (code=exited, status=0/SUCCESS)
  Process: 1164 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION
  Process: 1145 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 1114 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 2258 (code=exited, status=0/SUCCESS)
   Status: "MariaDB server is down"

Oct 05 09:31:23 nextcloudpi systemd[1]: Starting MariaDB database server...
Oct 05 09:31:44 nextcloudpi mysqld[2258]: 2018-10-05  9:31:44 3070102864 [Note] /usr/sbin/mysqld (mysqld 10.1.26-MariaDB-0+deb9u1) starting as process 2258 ...
Oct 05 09:31:47 nextcloudpi systemd[1]: Started MariaDB database server.
Oct 05 09:31:57 nextcloudpi systemd[1]: Stopping MariaDB database server...
Oct 05 09:32:00 nextcloudpi systemd[1]: Stopped MariaDB database server.
```
</details>
<br>
I additionally sent you access to the `mail.*` logs on Telegram. They might be related.

Henriktor, 2018-10-05:
My update failed as well from 14.0.0 --> 14.0.01.

```pi@KRONOS:/media/USBdrive/ncdata $ tail nextcloud.log 
{"reqId":"W7clQH8AAAEAAA26wiAAAABF","level":3,"time":"2018-10-05T08:48:00+00:00","remoteAddr":"37.247.30.174","user":"--","app":"core","method":"GET","url":"\/index.php\/204","message":{"Exception":"OC\\HintException","Message":"Memcache \\OC\\Memcache\\Redis not available for local cache","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":502,"function":"__construct","class":"OC\\Memcache\\Factory","type":"->","args":["1ccb1185a3bc16e1f22fee9b8f93d0bb",{"__class__":"OC\\Log"},"\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":974,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Template\\JSCombiner"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Template\\JSCombiner"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/TemplateLayout.php","line":316,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Template\\JSCombiner"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/TemplateLayout.php","line":157,"function":"findJavascriptFiles","class":"OC\\TemplateLayout","type":"::","args":[["vendor\/core","js\/merged-template-prepend","search\/js\/search","core\/l10n\/en","search\/l10n\/en","search\/js\/searchprovider","js\/jquery-ui-fixes","js\/files\/fileinfo","js\/files\/client","js\/contactsmenu"]]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/template.php","line":210,"function":"__construct","class":"OC\\TemplateLayout","type":"->","args":["error",""]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Template\/Base.php","line":132,"function":"fetchPage","class":"OC_Template","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/legacy\/template.php","line":359,"function":"printPage","class":"OC\\Template\\Base","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/index.php","line":61,"function":"printExceptionErrorPage","class":"OC_Template","type":"::","args":[{"__class__":"OC\\HintException"},500]}],"File":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Factory.php","Line":99,"Hint":"Is the matching PHP module installed and enabled?","CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.3.0","version":"14.0.0.18"}
{"reqId":"W7clQH8AAAEAAA26wiIAAABE","level":3,"time":"2018-10-05T08:48:00+00:00","remoteAddr":"37.247.30.174","user":"--","app":"core","method":"GET","url":"\/index.php\/204","message":{"Exception":"OC\\HintException","Message":"Memcache \\OC\\Memcache\\Redis not available for local cache","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":502,"function":"__construct","class":"OC\\Memcache\\Factory","type":"->","args":["1ccb1185a3bc16e1f22fee9b8f93d0bb",{"__class__":"OC\\Log"},"\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":974,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":123,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Template\\JSCombiner"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Template\\JSCombiner"]},{"file":"\/var\/www\/nextcloud\/lib\/p [... cut ...] /537.36 (KHTML, like Gecko) Chrome\/69.0.3497.100 Safari\/537.36","version":"14.0.0.18"}
pi@KRONOS:/media/USBdrive/ncdata $                    
```
Please advise! My Nextcloud is down giving this errors:

```Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.
```

TomTurnschuh, 2018-10-05:
@Henriktor: Please refer to #671 for this issue. It seems that you have the same Redis problem.

ArtJames, 2018-10-05:
Similar problem here.
After rebooting I get the same internal error message.
Restarting php-fpm, redis, mariadb, apache manually, and NC is running fine again.
Nothing suspicious to find in the usual logs beside in daemon.log:

_Oct  5 05:11:57 nextcloudpi bash[1184]: An unhandled exception has been thrown:
Oct  5 05:11:57 nextcloudpi bash[1184]: RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Oct  5 05:11:57 nextcloudpi bash[1184]: Stack trace:
Oct  5 05:11:57 nextcloudpi bash[1184]: #0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('611f78162d06097...')
Oct  5 05:11:57 nextcloudpi bash[1184]: #1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
Oct  5 05:11:57 nextcloudpi bash[1184]: #2 /var/www/nextcloud/lib/private/App/AppManager.php(380): OC\App\InfoParser->parse('/var/www/nextcl...')
Oct  5 05:11:57 nextcloudpi bash[1184]: #3 /var/www/nextcloud/lib/private/legacy/app.php(551): OC\App\AppManager->getAppInfo('files', false, NULL)
Oct  5 05:11:57 nextcloudpi bash[1184]: #4 /var/www/nextcloud/lib/private/AppFramework/App.php(65): OC_App::getAppInfo('files')
Oct  5 05:11:57 nextcloudpi bash[1184]: #5 /var/www/nextcloud/lib/private/legacy/app.php(236): OC\AppFramework\App::buildAppNamespace('files')
Oct  5 05:11:57 nextcloudpi bash[1184]: #6 /var/www/nextcloud/lib/private/legacy/app.php(119): OC_App::registerAutoloading('files', '/var/www/nextcl...')
Oct  5 05:11:57 nextcloudpi bash[1184]: #7 /var/www/nextcloud/lib/base.php(654): OC_App::loadApps(Array)
Oct  5 05:11:57 nextcloudpi bash[1184]: #8 /var/www/nextcloud/lib/base.php(1068): OC::init()
Oct  5 05:11:57 nextcloudpi bash[1184]: #9 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
Oct  5 05:11:57 nextcloudpi bash[1184]: #10 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')_

Redis Version 4.1.1-1

nachoparker, 2018-10-06:
@ArtJames let's deal with the redis issue [here](https://github.com/nextcloud/nextcloudpi/issues/671)

nachoparker, 2018-10-06:
@TomTurnschuh I wonder who's stopping your services? Can you try

```
sudo bash /usr/local/bin/ncp-provisioning.sh
```

and see what happens? try to force it to restart your services by changing some values in `limits.sh`

TomTurnschuh, 2018-10-06:
@nachoparker: Thanks for the hint.
Seems to be related to the log files:
<details>
<summary>sudo bash /usr/local/bin/ncp-provisioning.sh</summary>

```
Updating NextCloud config with Redis password
Updating NextCloud config with MariaDB password
Activating nc-limits
/usr/local/etc/library.sh: line 117: echo: write error: No space left on device
Using 1570464768B for PHP
tee: /var/log/ncp.log: No space left on device
Using 8 PHP threads
Using 1570464768B for Redis
/usr/local/etc/library.sh: line 120: echo: write error: No space left on device
```
</details>
<br>

It does boot including start of PHP and MySQL after deleting the mail logs in `/var/log` – even though `df -h` still shows me that the ZRAM space is filled 100 % and the mail logs are suddenly created again including the original entries. Still, booting does work correctly for some time until there is "no space left" again.

I'm wondering, why I get that many log entries...

nachoparker, 2018-10-06:
I fixed this in `v0.62.6`. Please update to the latest version and confirm.

@TomTurnschuh many thanks, as always ;)

closed 2018-10-06 by nachoparker

TomTurnschuh, 2018-10-06:
I (intentionally) still have `no space left on device`, however, the boot process works now. Thanks!

BTW: If my device runs out of log space, one of the first things it does is emptying `ncp.log`. :smile: