nextcloud/nextcloudpi #673: After update: Failed to connect to the database.

TomTurnschuh, 2018-10-04:
This issue relates to #672.

After the recent update, I get an error every now and then:
```
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:64
```
<details>
<summary>Stack trace</summary>

```
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql                                     \Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(D                                     octrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(630): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(166): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(O                                     C\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1532): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(352): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('OC\\Authenticati...')
#19 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#20 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(69): OC\ServerContainer->query('OC\\Authenticati...')
#21 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(99): OC\AppFramework\Utility\SimpleContainer->buildClass(Object(ReflectionClass))
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(120): OC\AppFramework\Utility\SimpleContainer->resolve('OC\\Authenticati...')
#23 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#24 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(69): OC\ServerContainer->query('OC\\Authenticati...')
#25 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(99): OC\AppFramework\Utility\SimpleContainer->buildClass(Object(ReflectionClass))
#26 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(120): OC\AppFramework\Utility\SimpleContainer->resolve('OC\\Authenticati...')
#27 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#28 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(166): OC\ServerContainer->query('OC\\Authenticati...')
#29 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(O                                     C\Server))
#30 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('OC\\Authenticati...')
#31 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#32 /var/www/nextcloud/lib/private/Server.php(364): OC\ServerContainer->query('OC\\Authenticati...')
#33 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#34 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('OCP\\IUserSessio...')
#35 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IUserSessio...')
#36 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(166): OC\ServerContainer->query('OCP\\IUserSessio...')
#37 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(O                                     C\Server))
#38 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('UserSession')
#39 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('UserSession')
#40 /var/www/nextcloud/lib/private/Server.php(1408): OC\ServerContainer->query('UserSession')
#41 /var/www/nextcloud/lib/private/Server.php(683): OC\Server->getUserSession()
#42 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#43 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('OC\\App\\AppManag...')
#44 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\App\\AppManag...')
#45 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(166): OC\ServerContainer->query('OC\\App\\AppManag...')
#46 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(O                                     C\Server))
#47 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(118): Pimple\Container->offsetGet('AppManager')
#48 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('AppManager')
#49 /var/www/nextcloud/lib/private/Server.php(1703): OC\ServerContainer->query('AppManager')
#50 /var/www/nextcloud/lib/private/legacy/app.php(342): OC\Server->getAppManager()
#51 /var/www/nextcloud/lib/private/legacy/app.php(113): OC_App::getEnabledApps()
#52 /var/www/nextcloud/lib/base.php(654): OC_App::loadApps(Array)
#53 /var/www/nextcloud/lib/base.php(1068): OC::init()
#54 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#55 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#56 {main}
```

</details>

<br>

It occurs for example when trying to use the OCC command.
(Re)starting MYSQL helps in this case.

nachoparker, 2018-10-04:
`systemctl status mysqld`?

TomTurnschuh, 2018-10-05:
MYSQL is not active:

<details>
<summary>systemctl status mysqld</summary>

```
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2018-10-05 09:32:00 CEST; 6min ago
  Process: 2303 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 2301 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 2258 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 1383 ExecStartPre=/bin/sleep 20 (code=exited, status=0/SUCCESS)
  Process: 1164 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION
  Process: 1145 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 1114 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 2258 (code=exited, status=0/SUCCESS)
   Status: "MariaDB server is down"

Oct 05 09:31:23 nextcloudpi systemd[1]: Starting MariaDB database server...
Oct 05 09:31:44 nextcloudpi mysqld[2258]: 2018-10-05  9:31:44 3070102864 [Note] /usr/sbin/mysqld (mysqld 10.1.26-MariaDB-0+deb9u1) starting as process 2258 ...
Oct 05 09:31:47 nextcloudpi systemd[1]: Started MariaDB database server.
Oct 05 09:31:57 nextcloudpi systemd[1]: Stopping MariaDB database server...
Oct 05 09:32:00 nextcloudpi systemd[1]: Stopped MariaDB database server.
```
</details>

TomTurnschuh, 2018-10-05:
This issue is probably directly related to #672.
Maybe we just continue there.

I would still leave this issue open, it will be closed automatically as soon as the other gets solved.

nachoparker, 2018-10-06:
#672 was fixed, so I think this should be fixed too. Please confirm

Thanks!

closed 2018-10-06 by nachoparker