nextcloud/nextcloudpi #674: Automount not working with clean install of RPi_10-01-18 image

langfingaz, 2018-10-04:
Hey! Thanks for the new RPi image with NC version 14!

I burned the new RPi_10-01-18 image to a 32GB sd card and powered up my RPi. As I started the configuration wizard and tried to move the data dir to an connected 3TB harddrive this error was shown:

"[ nc-automount ]
automount enabled

[ nc-datadir ]
/media/USBdrive does not exist"

But as the output of fdisk shows, the drive is detected (/dev/sda1). It was formated with the previous nextcloudPi version with btrfs.

Hope this helps you out! Any idears, how I could fix the problem?

Greetings, Daniel

<details>
<summary>sudo fdisk -l</summary>

pi@nextcloudpi:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

[... 182 lines cut ...]
2018-10-04 19:44:51 1995706352 [Note] InnoDB: Using generic crc32 instructions
2018-10-04 19:44:51 1995706352 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2018-10-04 19:44:51 1995706352 [Note] InnoDB: Completed initialization of buffer pool
2018-10-04 19:44:51 1995706352 [Note] InnoDB: Highest supported file format is Barracuda.
2018-10-04 19:45:00 1995706352 [Note] InnoDB: 128 rollback segment(s) are active.
2018-10-04 19:45:00 1995706352 [Note] InnoDB: Waiting for purge to start
2018-10-04 19:45:00 1995706352 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3697216
2018-10-04 19:45:00 1430238000 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-10-04 19:45:00 1995706352 [Note] Plugin 'FEEDBACK' is disabled.
2018-10-04 19:45:00 1995706352 [Note] Server socket created on IP: '127.0.0.1'.
2018-10-04 19:45:00 1995706352 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"6MBqeXepbvdKlFetHA20","level":3,"time":"2018-10-02T01:30:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":""}

```
</details>

chiyi4488, 2018-10-04:
I think you need to change a clerk adapter.

nachoparker, 2018-10-05:
Hi,

This is a kernel problem. Let's deal with this here https://github.com/nextcloud/nextcloudpi/issues/668

Thanks for reporting

closed 2018-10-05 by nachoparker

nachoparker, 2018-10-05:
Fixed in the latest image