nextcloud/nextcloudpi #675: After upgrade to PHP7.2: Two Redis versions installed in parallel

TomTurnschuh, 2018-10-08:
Just after last night I also encountered a Redis problem, as first described in #671.
It seems that there is still an old version installed as a leftover from PHP7.0, as the following indicates:

<details>
<summary>apt show php-redis -a</summary>

```
Package: php-redis
Version: 4.1.1-1
Priority: optional
Section: php
Maintainer: Debian PHP PECL Maintainers <team+php-pecl@tracker.debian.org>
Installed-Size: 299 kB
Provides: php7.2-redis
Depends: php-igbinary, php-common (>= 1:7.0+33~), phpapi-20170718, libc6 (>= 2.4)
Suggests: redis-server
Homepage: https://pecl.php.net/package/redis
Download-Size: 101 kB
APT-Manual-Installed: yes
APT-Sources: http://deb.debian.org/debian buster/main armhf Packages
Description: PHP extension for interfacing with Redis

Package: php-redis
Version: 3.1.1-1
Priority: optional
Section: php
Maintainer: Debian PHP PECL Maintainers <pkg-php-pecl@lists.alioth.debian.org>
Installed-Size: 299 kB
Provides: php7.0-redis
Depends: php-igbinary, php-common (>= 1:7.0+33~), phpapi-20151012, libc6 (>= 2.4)
Suggests: redis-server
Homepage: http://pecl.php.net/package/redis
Download-Size: 101 kB
APT-Sources: http://httpredir.debian.org/debian stretch/main armhf Packages
Description: PHP extension for interfacing with Redis
```
</details>

#### Installation
NCP 0.62.6 after update from earlier versions.

TomTurnschuh, 2018-10-08:
I further have the following comment in NCP log:
`Job for redis-server.service failed because the control process exited with error code.`

Further information:
<details>
<summary>systemctl status redis-server.service</summary>

```
redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-10-08 10:24:08 CEST; 7min ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)
  Process: 2429 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=1/FAILURE)
  Process: 2425 ExecStartPre=/bin/run-parts --verbose /etc/redis/redis-server.pre-up.d (code=exited, status=0/SUCCESS)

Oct 08 10:24:08 nextcloudpi systemd[1]: redis-server.service: Unit entered failed state.
Oct 08 10:24:08 nextcloudpi systemd[1]: redis-server.service: Failed with result 'exit-code'.
Oct 08 10:24:08 nextcloudpi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Oct 08 10:24:08 nextcloudpi systemd[1]: Stopped Advanced key-value store.
Oct 08 10:24:08 nextcloudpi systemd[1]: redis-server.service: Start request repeated too quickly.
Oct 08 10:24:08 nextcloudpi systemd[1]: Failed to start Advanced key-value store.
Oct 08 10:24:08 nextcloudpi systemd[1]: redis-server.service: Unit entered failed state.
Oct 08 10:24:08 nextcloudpi systemd[1]: redis-server.service: Failed with result 'exit-code'.
```
</details>

<details>
<summary>Excerpts from systemctl status redis-server.service</summary>

```
Oct 08 10:42:08 nextcloudpi sudo[6165]:     root : TTY=unknown ; PWD=/var/www/nextcloud ; USER=www-data ; COMMAND=/usr/bin/php occ config:system:set trusted_domains 1 --value=192.168.1.30
Oct 08 10:42:08 nextcloudpi sudo[6165]: pam_unix(sudo:session): session opened for user www-data by (uid=0)
Oct 08 10:42:08 nextcloudpi bash[6154]: An unhandled exception has been thrown:
Oct 08 10:42:08 nextcloudpi bash[6154]: RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Oct 08 10:42:08 nextcloudpi bash[6154]: Stack trace:
Oct 08 10:42:08 nextcloudpi bash[6154]: #0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('1905bc9b8c9ca98...')
Oct 08 10:42:08 nextcloudpi bash[6154]: #1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
Oct 08 10:42:08 nextcloudpi bash[6154]: #2 /var/www/nextcloud/lib/private/App/AppManager.php(380): OC\App\InfoParser->parse('/var/www/nextcl...')
Oct 08 10:42:08 nextcloudpi bash[6154]: #3 /var/www/nextcloud/lib/private/legacy/app.php(551): OC\App\AppManager->getAppInfo('files', false, NULL)
Oct 08 10:42:08 nextcloudpi bash[6154]: #4 /var/www/nextcloud/lib/private/AppFramework/App.php(65): OC_App::getAppInfo('files')
Oct 08 10:42:08 nextcloudpi bash[6154]: #5 /var/www/nextcloud/lib/private/legacy/app.php(236): OC\AppFramework\App::buildAppNamespace('files')
Oct 08 10:42:08 nextcloudpi bash[6154]: #6 /var/www/nextcloud/lib/private/legacy/app.php(119): OC_App::registerAutoloading('files', '/var/www/nextcl...')
Oct 08 10:42:08 nextcloudpi bash[6154]: #7 /var/www/nextcloud/lib/base.php(654): OC_App::loadApps(Array)
Oct 08 10:42:08 nextcloudpi bash[6154]: #8 /var/www/nextcloud/lib/base.php(1068): OC::init()
Oct 08 10:42:08 nextcloudpi bash[6154]: #9 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
Oct 08 10:42:08 nextcloudpi bash[6154]: #10 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
Oct 08 10:42:08 nextcloudpi sudo[6165]: pam_unix(sudo:session): session closed for user www-data
Oct 08 10:42:08 nextcloudpi bash[6154]: #11 {main}
Oct 08 10:42:08 nextcloudpi systemd[1]: nextcloud-domain.service: Main process exited, code=exited, status=1/FAILURE
Oct 08 10:42:08 nextcloudpi systemd[1]: nextcloud-domain.service: Unit entered failed state.
Oct 08 10:42:08 nextcloudpi systemd[1]: nextcloud-domain.service: Failed with result 'exit-code'.

Oct 08 10:44:07 nextcloudpi fail2ban-client[6540]: WARNING 'backend' not defined in 'ssh'. Using default one: 'auto'
Oct 08 10:44:07 nextcloudpi fail2ban-client[6540]: WARNING 'backend' not defined in 'nextcloud'. Using default one: 'auto'
Oct 08 10:44:07 nextcloudpi fail2ban-client[6540]: ERROR  No file(s) found for glob /var/www/nextcloud/data/nextcloud.log
Oct 08 10:44:07 nextcloudpi fail2ban-client[6540]: ERROR  Failed during configuration: Have not found any log file for nextcloud jail
Oct 08 10:44:07 nextcloudpi systemd[1]: fail2ban.service: Control process exited, code=exited status=255
Oct 08 10:44:07 nextcloudpi systemd[1]: Failed to start Fail2Ban Service.
```
</details>

TomTurnschuh, 2018-10-08:
After reinstalling and setting up Redis, at least the internal server error disappeared.

Could it be that Redis is not configured properly on update? I went with the following guidance:
[Redis configuration for Nextcloud](https://help.nextcloud.com/t/install-nextcloud-into-root-directory-of-my-domain/2513/26)

Could anybody check the configuration after updating from NCP <v0.61 to current version?
I don't have a test installation a the moment.

nachoparker, 2018-10-10:
`apt show` lists available versions. We have the Buster version and the Stretch version available to install, but only one installed

```
dpkg -l  | grep php-redis
ii  php-redis                         4.1.1-1                           armhf        PHP extension for interfacing with Redis
```

, which is `4.1.1-1`

nachoparker, 2018-10-10:
We would need `/var/log/redis/redis-server.log` to see why it failed to start in this case. Please, reopen if someone else runs into this

closed 2018-10-10 by nachoparker