nextcloud/nextcloudpi #678: Nextcloud 14 user creation errors

stu-d, 2018-10-10:
Installed using Curl on Debian Stretch, User creation screen shows errors and log has following errors in ti.

Error | core | Doctrine\DBAL\Exception\DriverException: An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid` COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname` COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue` COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0' with params ["%%", "%%", "%%"]: SQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DBALException.php - line 128:Doctrine\DBAL\Driver\AbstractMySQLDriver->convertException("An exceptio ... '", Doctrine\DBA ... ]})/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php - line 855:Doctrine\DBAL\DBALException::driverExceptionDuringQuery(Doctrine\DBA ... {}, Doctrine\DBA ... ]}, "SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"})/var/www/nextcloud/lib/private/DB/Connection.php - line 195:Doctrine\DBAL\Connection->executeQuery("SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"}, { 1: 2,2: 2,3: 2}, null)/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Query/QueryBuilder.php - line 206:OC\DB\Connection->executeQuery("SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"}, { 1: 2,2: 2,3: 2})/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php - line 214:Doctrine\DBAL\Query\QueryBuilder->execute()/var/www/nextcloud/lib/private/User/Database.php - line 278:OC\DB\QueryBuilder\QueryBuilder->execute()/var/www/nextcloud/lib/private/User/Database.php - line 379:OC\User\Database->getDisplayNames("", 500, 0)/var/www/nextcloud/lib/private/User/Manager.php - line 229:OC\User\Database->getUsers("", 500, 0)/var/www/nextcloud/apps/files/lib/BackgroundJob/ScanFiles.php - line 101:OC\User\Manager->search("", 500, 0)/var/www/nextcloud/lib/private/BackgroundJob/Job.php - line 61:OCA\Files\BackgroundJob\ScanFiles->run(null)/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php - line 55:OC\BackgroundJob\Job->execute(OC\BackgroundJob\JobList {}, OC\Log {})/var/www/nextcloud/cron.php - line 123:OC\BackgroundJob\TimedJob->execute(OC\BackgroundJob\JobList {}, OC\Log {})Cased by Doctrine\DBAL\Driver\PDOException: SQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php - line 847:Doctrine\DBAL\Driver\PDOStatement->execute()/var/www/nextcloud/lib/private/DB/Connection.php - line 195:Doctrine\DBAL\Connection->executeQuery("SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"}, { 1: 2,2: 2,3: 2}, null)/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Query/QueryBuilder.php - line 206:OC\DB\Connection->executeQuery("SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"}, { 1: 2,2: 2,3: 2})/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php - line 214:Doctrine\DBAL\Query\QueryBuilder->execute()/var/www/nextcloud/lib/private/User/Database.php - line 278:OC\DB\QueryBuilder\QueryBuilder->execute()/var/www/nextcloud/lib/private/User/Database.php - line 379:OC\User\Database->getDisplayNames("", 500, 0)/var/www/nextcloud/lib/private/User/Manager.php - line 229:OC\User\Database->getUsers("", 500, 0)/var/www/nextcloud/apps/files/lib/BackgroundJob/ScanFiles.php - line 101:OC\User\Manager->search("", 500, 0)/var/www/nextcloud/lib/private/BackgroundJob/Job.php - line 61:OCA\Files\BackgroundJob\ScanFiles->run(null)/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php - line 55:OC\BackgroundJob\Job->execute(OC\BackgroundJob\JobList {}, OC\Log {})/var/www/nextcloud/cron.php - line 123:OC\BackgroundJob\TimedJob->execute(OC\BackgroundJob\JobList {}, OC\Log {})Cased by PDOException: SQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Driver/PDOStatement.php - line 105:PDOStatement->execute(null)/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php - line 847:Doctrine\DBAL\Driver\PDOStatement->execute()/var/www/nextcloud/lib/private/DB/Connection.php - line 195:Doctrine\DBAL\Connection->executeQuery("SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"}, { 1: 2,2: 2,3: 2}, null)/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Query/QueryBuilder.php - line 206:OC\DB\Connection->executeQuery("SELECT `uid ... 0", { 1: "%%",2: "%%",3: "%%"}, { 1: 2,2: 2,3: 2})/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php - line 214:Doctrine\DBAL\Query\QueryBuilder->execute()/var/www/nextcloud/lib/private/User/Database.php - line 278:OC\DB\QueryBuilder\QueryBuilder->execute()/var/www/nextcloud/lib/private/User/Database.php - line 379:OC\User\Database->getDisplayNames("", 500, 0)/var/www/nextcloud/lib/private/User/Manager.php - line 229:OC\User\Database->getUsers("", 500, 0)/var/www/nextcloud/apps/files/lib/BackgroundJob/ScanFiles.php - line 101:OC\User\Manager->search("", 500, 0)/var/www/nextcloud/lib/private/BackgroundJob/Job.php - line 61:OCA\Files\BackgroundJob\ScanFiles->run(null)/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php - line 55:OC\BackgroundJob\Job->execute(OC\BackgroundJob\JobList {}, OC\Log {})/var/www/nextcloud/cron.php - line 123:OC\BackgroundJob\TimedJob->execute(OC\BackgroundJob\JobList {}, OC\Log {})
-- | -- | --

nachoparker, 2018-10-11:
This is a Nextcloud issue in any case, please report it there

Thanks

closed 2018-10-11 by nachoparker

GuyoST, 2018-11-16:
Hello, I have the same problem on the SQL query when I want to see "every one" in the users panel. And also when I want to share a folder or document with someone, the autocompletion is not working with a similar error, so It's not possible to share anything.

Nextcloud 14.0.3.
Example when I try to share something with the autocompletion box : 

`Doctrine\DBAL\Exception\DriverException: An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid` COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname` COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue` COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 200 OFFSET 0' with params ["%max%", "%max%", "%max%"]: SQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'`

I saw this bug is closed, but not solved for me, did I miss something ? or is there a workaround to fix it manually ?

Thank you very much for your help.
I installed it on a VM with debian 9 and followed the instruction with curl on the NextCloudPi website.

Thank you very much.

GuyoST, 2018-11-16:
Ok, just for information and to help other people who can have the same issue, I just made it work.
I followed the instructions on this page :

nachoparker, 2018-11-17:
> https://docs.nextcloud.com/server/13/admin_manual/configuration_database/mysql_4byte_support.html

NCP already supports this