nextcloud/nextcloudpi #679: Internal Server Error After Update

lasportsman13, 2018-10-11:
I was running NextCloud 12 until about a week ago when I decided to update. I used the GUI to upgrade to NextCloud 13 and everything was working well. At that point, I didn't realize that v14 was available so I decided to run nc-update. Since then, I have been receiving an Internal Server Error. I've attached my logs since they are too long to paste in. Sorry in advance for the PDFs - it wouldn't let me upload the RTF files. I'd appreciate any help!

**System information**

NextCloudPi Version v0.62.6

cat /usr/local/etc/ncp-baseimage
> cat: /usr/local/etc/ncp-baseimage: No such file or directory

[Apache2.pdf](https://github.com/nextcloud/nextcloudpi/files/2469179/Apache2.pdf)
[MariaDB.pdf](https://github.com/nextcloud/nextcloudpi/files/2469180/MariaDB.pdf)
[NextCloud.pdf](https://github.com/nextcloud/nextcloudpi/files/2469181/NextCloud.pdf)

sudo ncp-report

> <--! Paste this in GitHub report -->
> 
> <details>
> <summary>NextCloudPi diagnostics</summary>
> 
> ```
> 
> NextCloudPi version  v0.62.6
> distribution         Raspbian GNU/Linux 9 \n \l
> automount            no
> USB devices          sda 
> datadir              /media/USBdrive/ncdata
> data in SD           no
> data filesystem      ext2/ext3
> data disk usage      98G/1.8T
> rootfs usage         2.9G/7.3G
> swapfile             /var/swap
> dbdir                /var/lib/mysql
> Nextcloud check      ok
> Nextcloud version    13.0.6.1
> HTTPD service        up
> PHP service          up
> MariaDB service      up
> Redis service        up
> Postfix service      up
> internet check       ok
> port check 80        open
> port check 443       open
> IP                   ***REMOVED SENSITIVE VALUE***
> gateway              ***REMOVED SENSITIVE VALUE***
> interface            eth0
> certificates         ***REMOVED SENSITIVE VALUE***
> NAT loopback         yes
> uptime               4days
> 
> ```
> </details>
> 
> 
> <details>
> <summary>Nextcloud configuration</summary>
> 
> ```
> 
> {
>     "system": {
>         "passwordsalt": "***REMOVED SENSITIVE VALUE***",
>         "secret": "***REMOVED SENSITIVE VALUE***",
>         "trusted_domains": {
>             "0": "localhost",
>             "1": "169.254.138.126",
>             "3": "kermanian5.no-ip.biz",
>             "4": "kermanian5.no-ip.biz"
>         },
>         "datadirectory": "***REMOVED SENSITIVE VALUE***",
>         "overwrite.cli.url": "http:\/\/localhost",
>         "dbtype": "mysql",
>         "version": "13.0.6.1",
>         "dbname": "***REMOVED SENSITIVE VALUE***",
>         "dbhost": "***REMOVED SENSITIVE VALUE***",
>         "dbport": "",
>         "dbtableprefix": "oc_",
>         "mysql.utf8mb4": true,
>         "dbuser": "***REMOVED SENSITIVE VALUE***",
>         "dbpassword": "***REMOVED SENSITIVE VALUE***",
>         "installed": true,
>         "instanceid": "***REMOVED SENSITIVE VALUE***",
>         "memcache.local": "\\OC\\Memcache\\APCu",
>         "maintenance": false,
>         "loglevel": "2",
>         "log_type": "file",
>         "logfile": "\/var\/www\/nextcloud\/data\/nextcloud.log",
>         "mail_from_address": "***REMOVED SENSITIVE VALUE***",
>         "mail_smtpmode": "smtp",
>         "mail_smtpauthtype": "LOGIN",
>         "mail_domain": "***REMOVED SENSITIVE VALUE***",
>         "mail_smtpsecure": "ssl",
>         "mail_smtpauth": 1,
>         "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
>         "mail_smtpport": "465",
>         "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
>         "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
>         "theme": ""
>     }
> }
> 
> ```
> </details>
> 
> 
> <details>
> <summary>HTTPd logs</summary>
> 
> ```
> 
> [Thu Oct 11 06:25:04.678762 2018] [ssl:warn] [pid 19687:tid 1996054896] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
> [Thu Oct 11 06:25:05.000766 2018] [mpm_event:notice] [pid 19687:tid 1996054896] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
> [Thu Oct 11 06:25:05.000936 2018] [core:notice] [pid 19687:tid 1996054896] AH00094: Command line: '/usr/sbin/apache2'
> 
> ```
> </details>
> 
> 
> <details>
> <summary>Database logs</summary>
> 
> ```
> 
> 2018-10-11  6:47:05 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:06 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:07 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:08 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:09 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:10 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:11 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:12 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:13 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:14 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:15 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:16 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:17 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:18 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:19 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:20 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:21 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:22 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:23 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 2018-10-11  6:47:24 1995952128 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
> 
> ```
> </details>
> 
> 
> <details>
> <summary>Nextcloud logs</summary>
> 
> ```
> 
> 
> ```
> </details>

nachoparker, 2018-10-11:
seems like your database is in trouble. Unclean poweroff maybe?

if you have backups, I would try this

https://serverfault.com/questions/477448/mysql-keeps-crashing-innodb-unable-to-lock-ibdata1-error-11/748778#748778

lasportsman13, 2018-10-12:
Thank you for your response and for all your hard work in producing and maintaining an incredible product. I don't think I have backups but I'm not sure - do you know where they might be hiding?

nachoparker, 2018-10-13:
Well, if you don't have backups I still would try that solution above

lasportsman13, 2018-10-13:
Thank you for your help. After following the instructions and running service mysql start, I get the following:

> Job for mariadb.service failed because the control process exited with error code.
> See "systemctl status mariadb.service" and "journalctl -xe" for details.

Running systemctl status mariadb.service gives:

> Unknown operation staus.

journalctl -xe gives:

> Oct 13 12:25:37 raspberrypi systemd[1]: Starting MariaDB database server...
> -- Subject: Unit mariadb.service has begun start-up
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- Unit mariadb.service has begun starting up.
> Oct 13 12:25:38 raspberrypi sshd[10320]: Failed password for root from 116.31.11
> Oct 13 12:25:38 raspberrypi sshd[10320]: Received disconnect from 116.31.116.42 
> Oct 13 12:25:38 raspberrypi sshd[10320]: Disconnected from 116.31.116.42 port 36
> Oct 13 12:25:38 raspberrypi sshd[10320]: PAM 2 more authentication failures; log
> Oct 13 12:25:39 raspberrypi sshd[10311]: Received disconnect from 125.65.42.183 
> Oct 13 12:25:39 raspberrypi sshd[10311]: Disconnected from 125.65.42.183 port 35
> Oct 13 12:25:59 raspberrypi mysqld[10524]: 2018-10-13 12:25:59 1996120064 [Note]

I think the ssh attempts are just Chinese hackers so I don't think it's relevant. Not sure what to do about the mysql error though.

nachoparker, 2018-10-14:
Do `status` again after a while, you pasted here the output while it was being initialized.

More interesting probably to look at `/var/log/mysql/error.log`

lasportsman13, 2018-10-15:
Just ran service mysql status again:

> ● mariadb.service - MariaDB database server
>    Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: 
>    Active: activating (start) since Mon 2018-10-15 07:15:07 PDT; 22s ago
>   Process: 31195 ExecStartPre=/bin/sleep 20 (code=exited, status=0/SUCCESS)
>   Process: 31022 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VA
>   Process: 31016 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_STAR
>   Process: 31014 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/r
>  Main PID: 31251 (mysqld)
>    CGroup: /system.slice/mariadb.service
>            └─31251 /usr/sbin/mysqld
> 
> Oct 15 07:15:07 raspberrypi systemd[1]: Starting MariaDB database server...
> Oct 15 07:15:29 raspberrypi mysqld[31251]: 2018-10-15  7:15:29 1996406784 [Note]

Here is the mysql error log:

> 2018-10-15  6:25:57 1996230656 [ERROR] mysqld: Can't lock aria control file '/var/lib/mysql/aria_log_control' for exclusive use, error: 11. Will retry for 30 seconds
> 2018-10-15  6:26:28 1996230656 [ERROR] mysqld: Got error 'Could not get an exclusive lock; file is probably in use by another process' when trying to use aria control file '/var/lib/mysql/aria_log_control'
> 2018-10-15  6:26:28 1996230656 [ERROR] Plugin 'Aria' init function returned error.
> 2018-10-15  6:26:28 1996230656 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Using mutexes to ref count buffer pool pages
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: The InnoDB memory heap is disabled
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Compressed tables use zlib 1.2.8
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Using Linux native AIO
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Using generic crc32 instructions
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Initializing buffer pool, size = 128.0M
> 2018-10-15  6:26:28 1996230656 [Note] InnoDB: Completed initialization of buffer pool
> 2018-10-15  6:26:29 1996230656 [Note] InnoDB: Highest supported file format is Barracuda.
> 2018-10-15  6:26:29 1996230656 [Note] InnoDB: 128 rollback segment(s) are active.
> 2018-10-15  6:26:29 1996230656 [Note] InnoDB: Waiting for purge to start
> 2018-10-15  6:26:29 1996230656 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 1625953
> 2018-10-15  6:26:29 1585427248 [Note] InnoDB: Dumping buffer pool(s) not yet started
> 2018-10-15  6:26:29 1996230656 [Note] Plugin 'FEEDBACK' is disabled.
> 2018-10-15  6:26:29 1996230656 [ERROR] Aria engine is not enabled or did not start. The Aria engine must be enabled to continue as mysqld was configured with --with-aria-tmp-tables
> 2018-10-15  6:26:29 1996230656 [ERROR] Aborting
> 
> 2018-10-15  6:26:59 1996124160 [ERROR] mysqld: Can't lock aria control file '/var/lib/mysql/aria_log_control' for exclusive use, error: 11. Will retry for 30 seconds
> 2018-10-15  6:27:30 1996124160 [ERROR] mysqld: Got error 'Could not get an exclusive lock; file is probably in use by another process' when trying to use aria control file '/var/lib/mysql/aria_log_control'
> 2018-10-15  6:27:30 1996124160 [ERROR] Plugin 'Aria' init function returned error.
> 2018-10-15  6:27:30 1996124160 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Using mutexes to ref count buffer pool pages
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: The InnoDB memory heap is disabled
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Compressed tables use zlib 1.2.8
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Using Linux native AIO
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Using generic crc32 instructions
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Initializing buffer pool, size = 128.0M
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Completed initialization of buffer pool
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Highest supported file format is Barracuda.
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: 128 rollback segment(s) are active.
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB: Waiting for purge to start
> 2018-10-15  6:27:30 1996124160 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 1625963
> 2018-10-15  6:27:30 1585427248 [Note] InnoDB: Dumping buffer pool(s) not yet started
> 2018-10-15  6:27:30 1996124160 [Note] Plugin 'FEEDBACK' is disabled.
> 2018-10-15  6:27:30 1996124160 [ERROR] Aria engine is not enabled or did not start. The Aria engine must be enabled to continue as mysqld was configured with --with-aria-tmp-tables
> 2018-10-15  6:27:30 1996124160 [ERROR] Aborting
> 
> 2018-10-15  6:28:01 1996029952 [ERROR] mysqld: Can't lock aria control file '/var/lib/mysql/aria_log_control' for exclusive use, error: 11. Will retry for 30 seconds
> 2018-10-15  6:28:32 1996029952 [ERROR] mysqld: Got error 'Could not get an exclusive lock; file is probably in use by another process' when trying to use aria control file '/var/lib/mysql/aria_log_co [... cut ...]
> 2018-10-15  7:16:01 1585427248 [Note] InnoDB: Dumping buffer pool(s) not yet started
> 2018-10-15  7:16:01 1996406784 [Note] Plugin 'FEEDBACK' is disabled.
> 2018-10-15  7:16:01 1996406784 [ERROR] Aria engine is not enabled or did not start. The Aria engine must be enabled to continue as mysqld was configured with --with-aria-tmp-tables
> 2018-10-15  7:16:01 1996406784 [ERROR] Aborting
> 
> 2018-10-15  7:16:31 1995894784 [ERROR] mysqld: Can't lock aria control file '/var/lib/mysql/aria_log_control' for exclusive use, error: 11. Will retry for 30 seconds

nachoparker, 2018-10-15:
maybe try 

```
systemctl stop mysqld
mv /var/lib/mysql/aria_log_control /var/lib/mysql/aria_log_control_bak
systemctl start mysqld
```

lasportsman13, 2018-10-16:
systemctl start mysqld returned:

> Job for mariadb.service failed because the control process exited with error code.
> See "systemctl status mariadb.service" and "journalctl -xe" for details.

systemctl status mariadb.service returned:

> ● mariadb.service - MariaDB database server
>    Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: 
>    Active: activating (start-pre) since Tue 2018-10-16 14:34:43 PDT; 14s ago
>   Process: 16200 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_W
>   Process: 16246 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VA
>   Process: 16242 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_STAR
>   Process: 16236 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/r
>  Main PID: 16200 (code=exited, status=1/FAILURE); Control PID: 16420 (sleep)
>    CGroup: /system.slice/mariadb.service
>            └─control
>              └─16420 /bin/sleep 20
> 
> Oct 16 14:34:43 raspberrypi systemd[1]: Starting MariaDB database server...

journalctl -xe returned:

> Oct 16 14:35:40 raspberrypi systemd[1]: Failed to start MariaDB database server.
> -- Subject: Unit mariadb.service has failed
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- Unit mariadb.service has failed.
> -- 
> -- The result is failed.
> Oct 16 14:35:40 raspberrypi systemd[1]: mariadb.service: Unit entered failed sta
> Oct 16 14:35:40 raspberrypi systemd[1]: mariadb.service: Failed with result 'exi
> Oct 16 14:35:45 raspberrypi systemd[1]: mariadb.service: Service hold-off time o
> Oct 16 14:35:45 raspberrypi systemd[1]: Stopped MariaDB database server.
> -- Subject: Unit mariadb.service has finished shutting down
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- Unit mariadb.service has finished shutting down.
> Oct 16 14:35:45 raspberrypi systemd[1]: Starting MariaDB database server...
> -- Subject: Unit mariadb.service has begun start-up
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- Unit mariadb.service has begun starting up.

ttoopp, 2018-10-24:
test

ttoopp, 2018-10-24:
Hi, I have the same error in my browser, after the upgrade to v0.64.7.

> Internal Server Error
> 
> The server encountered an internal error and was unable to complete your request.
> Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
> More details can be found in the server log.

nc-error.log

> [Wed Oct 24 14:04:46.982926 2018] [proxy_fcgi:error] [pid 6954:tid 1683338272] [client 11.22.68.227:62772] AH01071: Got error 'PHP message: PHP Fatal error: Uncaught Doctrine\\DBAL\\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:64\nStack trace:\n#0 /var/www/nextcloud

Database and apache2 are running. 

I try a update with occ, but it didn't work. After I swich php from 7.3 to 7.2 it was possible, to update the system with occ command, but the error is still there.

ttoopp, 2018-10-24:
If I switch installed to false in the config.php, the login screen is loaded and I can take a new configuration.

ttoopp, 2018-10-24:
Okay, after comment out MEMCACHE in config.php my instalaltion working again 💃

nachoparker, 2018-10-27:
There's something corrupted with your database. Sorry, I  can't reproduce here, maybe look for help in a mysql forum

closed 2018-10-27 by nachoparker

imk13, 2022-06-23:
@lasportsman13 
Restarting RPi fixed the issue for me but before that I rebuild the nextcloudpi container