nextcloud/nextcloudpi #681: Testing update to 14.0.3 - v0.64

Iolaum, 2018-10-12:
While trying to update from master (v0.62.6) I had this problem:

```
$ sudo ncp-update devel
INFO: updating to development branch 'devel'
Downloading updates
Performing updates
./update.sh: line 291: syntax error near unexpected token `||'
./update.sh: line 291: `      [[ -f /etc/rsyslog.d/20-ufw.conf ]] && { grep -q "^\& stop" /etc/rsyslog.d/20-ufw.conf &> ||  echo "& stop" >> /etc/rsyslog.d/20-ufw.conf; }'
```

### NCP report (after failed update):

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.62.6
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      388G/917G
rootfs usage         1.9G/3.6G
swapfile             /media/wd1/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    14.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               4min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "$WEBSITE:443",
            "1": "192.168.1.20",
            "4": "$WEBSITE",
            "2": "$WEBSITE",
            "3": "$WEBSITE",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/$WEBSITE\/",
[... 91 lines cut ...]
```

{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:00+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"13.0.6.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"14.0.1.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"14.0.1.1"}
{"reqId":"bduQDTVVASfo32m79Mc5","level":1,"time":"2018-10-03T06:41:37+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"14.0.1.1"}
{"reqId":"WOW1LNsJ07y2Ih4qxtAa","level":3,"time":"2018-10-03T06:42:41+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/apps\/news\/appinfo\/signature.json): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Helpers\/FileAccessHelper.php#41","userAgent":"--","version":"14.0.1.1"}
{"reqId":"hJrvca5t2pJ6CoodXt9l","level":3,"time":"2018-10-03T23:35:24+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/apps\/news\/appinfo\/signature.json): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Helpers\/FileAccessHelper.php#41","userAgent":"--","version":"14.0.1.1"}
{"reqId":"NkXxocHTVdYD3Jt5FgMb","level":3,"time":"2018-10-03T23:46:12+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/apps\/news\/appinfo\/signature.json): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Helpers\/FileAccessHelper.php#41","userAgent":"--","version":"14.0.1.1"}
{"reqId":"gytpcsMVrBF5o4yiownd","level":3,"time":"2018-10-04T00:12:34+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"hash_file(\/var\/www\/nextcloud\/.user.ini-bk): failed to open stream: Permission denied at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#229","userAgent":"--","version":"14.0.1.1"}
{"reqId":"LS0uOjPKxgorktbrn3zh","level":3,"time":"2018-10-04T00:23:54+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"copy(\/var\/www\/nextcloud\/.user.ini): failed to open stream: Permission denied at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#194","userAgent":"--","version":"14.0.1.1"}
{"reqId":"LS0uOjPKxgorktbrn3zh","level":2,"time":"2018-10-04T00:23:54+00:00","remoteAddr":"","user":"--","app":"files","method":"","url":"--","message":"Can't write upload limit to \/media\/wd1\/ncdata\/tmp\/oc_tmp_ovcjx0-folder\/.user.ini. Please check the file permissions","userAgent":"--","version":"14.0.1.1"}
{"reqId":"LS0uOjPKxgorktbrn3zh","level":3,"time":"2018-10-04T00:24:03+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"copy(\/var\/www\/nextcloud\/.user.ini): failed to open stream: Permission denied at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#194","userAgent":"--","version":"14.0.1.1"}
{"reqId":"LS0uOjPKxgorktbrn3zh","level":2,"time":"2018-10-04T00:24:03+00:00","remoteAddr":"","user":"--","app":"files","method":"","url":"--","message":"Can't write upload limit to \/media\/wd1\/ncdata\/tmp\/oc_tmp_FTWdqM-folder\/.user.ini. Please check the file permissions","userAgent":"--","version":"14.0.1.1"}
{"reqId":"LS0uOjPKxgorktbrn3zh","level":3,"time":"2018-10-04T00:24:03+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/media\/wd1\/ncdata\/tmp\/oc_tmp_FTWdqM-folder\/.user.ini): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#208","userAgent":"--","version":"14.0.1.1"}
{"reqId":"r4wSYq9UKdce03xmVejy","level":3,"time":"2018-10-04T00:24:43+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"copy(\/var\/www\/nextcloud\/.user.ini): failed to open stream: Permission denied at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#194","userAgent":"--","version":"14.0.1.1"}
{"reqId":"r4wSYq9UKdce03xmVejy","level":2,"time":"2018-10-04T00:24:43+00:00","remoteAddr":"","user":"--","app":"files","method":"","url":"--","message":"Can't write upload limit to \/media\/wd1\/ncdata\/tmp\/oc_tmp_1oPSh2-folder\/.user.ini. Please check the file permissions","userAgent":"--","version":"14.0.1.1"}
{"reqId":"r4wSYq9UKdce03xmVejy","level":3,"time":"2018-10-04T00:24:50+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"copy(\/var\/www\/nextcloud\/.user.ini): failed to open stream: Permission denied at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#194","userAgent":"--","version":"14.0.1.1"}
{"reqId":"r4wSYq9UKdce03xmVejy","level":2,"time":"2018-10-04T00:24:50+00:00","remoteAddr":"","user":"--","app":"files","method":"","url":"--","message":"Can't write upload limit to \/media\/wd1\/ncdata\/tmp\/oc_tmp_9dSbWJ-folder\/.user.ini. Please check the file permissions","userAgent":"--","version":"14.0.1.1"}
{"reqId":"r4wSYq9UKdce03xmVejy","level":3,"time":"2018-10-04T00:24:50+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/media\/wd1\/ncdata\/tmp\/oc_tmp_9dSbWJ-folder\/.user.ini): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/IntegrityCheck\/Checker.php#208","userAgent":"--","version":"14.0.1.1"}
{"reqId":"W7fK1X8AAQEAAAPVDswAAAAA","level":3,"time":"2018-10-05T20:34:36+00:00","remoteAddr":"192.168.1.7","user":"--","app":"remote","method":"GET","url":"\/status.php","message":{"Exception":"RedisExce [... cut ...] loadApps","class":"OC_App","type":"::","args":[["session"]]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/status.php","line":36,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Redis.php","Line":54,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.3 (Nextcloud)","version":"14.0.1.1"}

```
</details>

ovpc, 2018-10-12:
rpi3B+ running berryboot latest ncp image:
sudo ncp-update devel
INFO: updating to development branch 'devel'
Downloading updates
Performing updates
Installing DDNS_spDYN
Installing nc-backup
Installing nc-restore
An unhandled exception has been thrown:
ParseError: syntax error, unexpected '5' (T_LNUMBER), expecting ')' in /var/www/nextcloud/config/config.php:10
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#1 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#3 /var/www/nextcloud/lib/base.php(1068): OC::init()
#4 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#5 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#6 {main}An unhandled exception has been thrown:
ParseError: syntax error, unexpected '5' (T_LNUMBER), expecting ')' in /var/www/nextcloud/config/config.php:10
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#1 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#3 /var/www/nextcloud/lib/base.php(1068): OC::init()
#4 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#5 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#6 {main}Activating nc-autoupdate-nc
automatic Nextcloud updates enabled
An unhandled exception has been thrown:
ParseError: syntax error, unexpected '5' (T_LNUMBER), expecting ')' in /var/www/nextcloud/config/config.php:10
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#1 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#3 /var/www/nextcloud/lib/base.php(1068): OC::init()
#4 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#5 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#6 {main}An unhandled exception has been thrown:
ParseError: syntax error, unexpected '5' (T_LNUMBER), expecting ')' in /var/www/nextcloud/config/config.php:10
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#1 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#3 /var/www/nextcloud/lib/base.php(1068): OC::init()
#4 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#5 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#6 {main}An unhandled exception has been thrown:
ParseError: syntax error, unexpected '5' (T_LNUMBER), expecting ')' in /var/www/nextcloud/config/config.php:10
Stack trace:
#0 /var/www/nextcloud/lib/private/Config.php(63): OC\Config->readData()
#1 /var/www/nextcloud/lib/base.php(140): OC\Config->__construct('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/base.php(591): OC::initPaths()
#3 /var/www/nextcloud/lib/base.php(1068): OC::init()
#4 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#5 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#6 {main}NextCloudPi updated to version v0.64.0

Iolaum, 2018-10-12:
Update fixed the issue:

```
$ sudo ncp-update devel
INFO: updating to development branch 'devel'
Downloading updates
Performing updates
Installing DDNS_spDYN
Installing nc-backup
Installing nc-restore
previewgenerator already installed
previewgenerator enabled
Activating nc-autoupdate-nc
automatic Nextcloud updates enabled
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
NextCloudPi updated to version v0.64.0
```

closed 2018-10-12 by Iolaum

nachoparker, 2018-10-12:
thanks for the help!

ovpc, 2018-10-13:
[image]
This morning found that the test rpi3b+ (NCP updated to devel 64.0 yesterday) was in process of updating NC. (as expected)
It had created a backup, moved nextcloud to nextcloud-old and created a new nextcloud folder. 
I am asuming it will run ./occ upgrade, and I do not need to take any action, except grab coffee and wait ;-)

ovpc, 2018-10-13:
[image]
That seemed to have worked fine. Nice job, thank you.

nachoparker, 2018-10-13:
Awesome, thanks guys!