nextcloud/nextcloudpi #685: Issue with nc-backup

GarySheldon, 2018-10-20:
Hi, I have the data and the database in a sdcard and I want to make a backup of everything to a usbdrive, the database backup works but when I include the data it returns this:

[ nc-backup ]
check free space...
clean up old backup /media/minube/nextcloud-bkp_20181020_1540133202.tar
Maintenance mode enabled
backup database...
backup base files...
backup data files...
tar: /data/app/data/..: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
error generating backup
Maintenance mode disabled

I've checked that the /data/app/data directory exists.

nachoparker, 2018-10-21:
I just checked this, it works for me

```
[ nc-backup ]
check free space...
Maintenance mode enabled
backup database...
backup base files...
backup /data/backups/ncp-backups/nextcloud-bkp_20181021_1540083421.tar generated
Maintenance mode disabled
```

you must have some permission issues on the mounted volume

closed 2018-10-21 by nachoparker

paschaef, 2018-12-19:
Hi there, I think this issue is still relevant. I'm observing the same issue as @GarySheldon . (I am running the nextcloudpi-x86 docker image)
In fact, there might not be a permission issue but a path issue causing the `tar: /data/app/data/..: Cannot open: No such file or directory` error. There is a path/directory inconsistency. 

In the file [nc-backup.sh](https://github.com/nextcloud/nextcloudpi/blob/f80ee23e064a74b19ce8f68ec7fed00007133574/etc/ncp-config.d/nc-backup.sh) at f80ee23 at line 64 the directory `/data/app` is renamed to `/data/nextcloud`.

However, at line 91 `$DATADIR` is still pointing to `/data/app/data` but the directory does not exist because of the actions at line 64 and hence `tar`command reports 'No such file or directory'. 

Quick fix would be to replace line 91 as follows:

from:
```bash
-rf "$DESTFILE" -C "$DATADIR"/.. "$( basename "$DATADIR" )" \
```
to:
```bash
-rf "$DESTFILE" -C "$BASEDIR"/nextcloud "$( basename "$DATADIR" )" \
```

However, this is a hardcoded solution, but because the `mv /data/app /data/nextcloud` is done in a hardcoded fashion this would be appropriate. 

@nachoparker is this worth reopening the issue?

nachoparker, 2018-12-20:
@paschaef we are terribly lacking manpower, could you send a PR with the fix?

paschaef, 2018-12-20:
@nachoparker here it is #746