nextcloud/nextcloudpi #691: Getting brute-force-detection when accessing newest NextCloudPi from inside the house

Steffinuss, 2018-10-24:
Hello,

I have an up-to-date NextCloudPi system (NextCloudPi 0.64.7 , NextCloud 14.0.3, completely updated via SSH apt-get upgrade) installed on a Raspberry Pi 3 B+. When accecing inside, I get an error, telling me I have to wait 30 seconds to access because of multiple logins. It seems to be a brute force protecsion, but it always appears and it's terribly slow. When accesing from outside (via 4G oder Tor Browser) I don't get that error and it's running smooth. I appeared since updating to Nextcloud 14 and PHP 7.2.9. I found the following link, but I technically do not know if it is related and how to solve: [NextCloud_Help](https://help.nextcloud.com/t/nextcloud-14-0-1-1-probleme-mit-php7-2/37950/2)

[image]

ovpc, 2018-10-24:
If it works fine from outside, no reason for it to be  PHP related. And no reason to create an issue at github, So closing this here for now, anyone can re-open if it turns out to be NCP bug.
Will gladly assist further, if you post your request at [help.nextcloud.com](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)
That  said:
- Assuming UFW and Fail2ban are enabled!
- Check the logs! They should provide clues to whats going on! 
- Do you have another pc or app inside local network trying to log in?
- Try to think of what changed since it last worked!? Did it work correctly at first

ttoopp, 2018-10-24:
Hi, I have the same problem. After update to v0.64.7, I cannot connect to my cloud.
My brwoser shows just this:

> Internal Server Error
> 
> The server encountered an internal error and was unable to complete your request.
> Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
> More details can be found in the server log.

nc-error.log shows

> $Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:64\nStack trace:\n#0 /var/www/nextcloud

But DB and apache are running.

Steffinuss, 2018-10-27:
I completey reinstalled my system and it is now working. So it seems something was messed up. Couldn't really figure out what the problem was, but now it is working.
Greetings and Thanks for your support. 
@ttoopp hope you get your system running as well. I am now on 6.47.10 and everything it smooth, "indoor and outdoor".
Stefan

closed 2018-10-27 by Steffinuss

misterlindsey, 2019-02-27:
I was getting this as well until I setup new DDNS today. Seems to be working using custom domain now. Was getting a brute error and wasn't able to see any files listed when it finally logged in. Was using ddns.net address. Now am using my own domain and reinstalled certificate.