nextcloud/nextcloudpi #692: Redis just died and the machine was literally happy for a few days.

markedphillips, 2018-10-24:
Thanks for reporting issues back to NextCloudPi! 

Here you can file bugs and feature requests. **Do NOT ask questions**, this is not a support forum.

ncp-scan returned:
An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('9ac8046ce2701bd...')
#1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/private/App/AppManager.php(380): OC\App\InfoParser->parse('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/private/legacy/app.php(551): OC\App\AppManager->getAppInfo('files', false, NULL)
#4 /var/www/nextcloud/lib/private/AppFramework/App.php(65): OC_App::getAppInfo('files')
#5 /var/www/nextcloud/lib/private/legacy/app.php(236): OC\AppFramework\App::buildAppNamespace('files')
#6 /var/www/nextcloud/lib/private/legacy/app.php(119): OC_App::registerAutoloading('files', '/var/www/nextcl...')
#7 /var/www/nextcloud/lib/base.php(654): OC_App::loadApps(Array)
#8 /var/www/nextcloud/lib/base.php(1068): OC::init()
#9 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#10 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')

nc-info returns:
Gathering information...
NextCloudPi version  v0.64.7
NextCloudPi image    NextCloudPi_09-29-18
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/Nextcloud/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      148G/2.8T
rootfs usage         5.7G/30G
swapfile             /var/swap
dbdir                /ncpdata/ncdatabase
Nextcloud check      error
HTTPD service        down
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   192.168.1.44
gateway              192.168.1.1
interface            enx001e0632c07b
certificates         sadperson.ddns.net
NAT loopback         no
uptime               1:16

You should open your ports for Lets Encrypt and external access
Done. Press any key...

nachoparker, 2018-10-24:
output from `systemctl status redis` and `cat /var/log/redis/redis-server.log`

klundry, 2018-10-27:
I am having this same issue. 
`● redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-10-27 03:03:49 UTC; 5min ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)
  Process: 10608 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=1/FAILURE)
  Process: 10605 ExecStartPre=/bin/run-parts --verbose /etc/redis/redis-server.pre-up.d (code=exited, status=0/SUCCESS)

Oct 27 03:03:49 nextcloudplus systemd[1]: redis-server.service: Unit entered failed state.
Oct 27 03:03:49 nextcloudplus systemd[1]: redis-server.service: Failed with result 'exit-code'.
Oct 27 03:03:49 nextcloudplus systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Oct 27 03:03:49 nextcloudplus systemd[1]: Stopped Advanced key-value store.
Oct 27 03:03:49 nextcloudplus systemd[1]: redis-server.service: Start request repeated too quickly.
Oct 27 03:03:49 nextcloudplus systemd[1]: Failed to start Advanced key-value store.
Oct 27 03:03:49 nextcloudplus systemd[1]: redis-server.service: Unit entered failed state.
Oct 27 03:03:49 nextcloudplus systemd[1]: redis-server.service: Failed with result 'exit-code'.`

/var/log/redis/redis-server.log is empty

klundry, 2018-10-27:
Ok, this may be the issue. 

`root@nextcloudplus:/var/log/redis# sudo -u redis redis-server /etc/redis/redis.conf

*** FATAL CONFIG FILE ERROR ***
Reading the configuration file, at line 163
>>> 'logfile /var/log/redis/redis-server.log'
Can't open the log file: Permission denied`

I tried doing "chown redis:redis /var/log/redis/redis-server.log" but still get the same error. I then just commented out line 163 in /etc/redis/redis.conf like so "#logfile /var/log/redis/redis-server.log" and Nextcloud is working again after a reboot.

klundry, 2018-10-27:
It looks like a permissions issue with the log file but I'm not sure why it complains even after making sure the redis user owns it.

nachoparker, 2018-10-27:
@klundry check the permissions, that is weird

`ls -l /var/log/redis -d`, `ls -l /var/log/redis`

markedphillips, 2018-12-16:
1. drwxr-x--- 2 root root 4096 Nov 12 06:25 /var/log/redis
2. -rw-rw---- 1 root root 0 Dec 16 19:00 redis-server.log

nachoparker, 2018-12-17:
`systemctl status redis` and `cat /var/log/redis/redis-server.log`?

zedocrob, 2018-12-26:
Same problem with redis here since yesterday (after update 0.67.8 ?)

```
 redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Wed 2018-12-26 21:24:59 UTC; 3min 14s ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)
  Process: 1782 ExecStopPost=/bin/run-parts --verbose /etc/redis/redis-server.post-down.d (code=exited, status=0/SUCCES
  Process: 1724 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=1/FAILURE)
  Process: 1779 ExecStartPre=/bin/run-parts --verbose /etc/redis/redis-server.pre-up.d (code=killed, signal=TERM)

Dec 26 21:24:58 nextcloudpi systemd[1]: Starting Advanced key-value store...
Dec 26 21:24:59 nextcloudpi run-parts[1782]: run-parts: executing /etc/redis/redis-server.post-down.d/00_example
Dec 26 21:24:59 nextcloudpi systemd[1]: Stopped Advanced key-value store.
Dec 26 21:24:59 nextcloudpi systemd[1]: redis-server.service: Start request repeated too quickly.
Dec 26 21:24:59 nextcloudpi systemd[1]: Failed to start Advanced key-value store.
Dec 26 21:24:59 nextcloudpi systemd[1]: redis-server.service: Unit entered failed state.
Dec 26 21:24:59 nextcloudpi systemd[1]: redis-server.service: Failed with result 'start-limit-hit'.

```
@klundry  i did the same and working after reboot it's working.

> Ok, this may be the issue.
>
> `root@nextcloudplus:/var/log/redis# sudo -u redis redis-server /etc/redis/redis.conf
>
> *** FATAL CONFIG FILE ERROR ***
> Reading the configuration file, at line 163
>
>
> I [...]

nachoparker, 2018-12-28:
@zedocrob any idea about how to reproduce this? I can't find what it is in a normal installation

wayneoutthere, 2019-05-14:
looks like i have this after I (late to the party) just upgraded to 0.67.8 
Before that, my mariadb was down, so I did an upgrade (which worked it seemed) and when it came back redis was down too with these 'exit-code' messages.  

You were asking above for 
systemctl status redis and cat /var/log/redis/redis-server.log

systemctl status redis gives this:

` redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-05-14 04:16:08 BST; 43min ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)
  Process: 876 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=1/FAILURE)
  Process: 835 ExecStartPre=/bin/run-parts --verbose /etc/redis/redis-server.pre-up.d (code=exited, status=0/SUCCESS)

May 14 04:16:08 nextcloudpi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 14 04:16:08 nextcloudpi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
May 14 04:16:08 nextcloudpi systemd[1]: Stopped Advanced key-value store.
May 14 04:16:08 nextcloudpi systemd[1]: redis-server.service: Start request repeated too quickly.
May 14 04:16:08 nextcloudpi systemd[1]: Failed to start Advanced key-value store.
May 14 04:16:08 nextcloudpi systemd[1]: redis-server.service: Unit entered failed state.
May 14 04:16:08 nextcloudpi systemd[1]: redis-server.service: Failed with result 'exit-code'.
May 14 04:16:10 nextcloudpi systemd[1]: redis-server.service: Start request repeated too quickly.
May 14 04:16:10 nextcloudpi systemd[1]: Failed to start Advanced key-value store.
May 14 04:16:10 nextcloudpi systemd[1]: redis-server.service: Failed with result 'exit-code'.
`
while 

creates this: https://pastebin.com/9Fzr4Vtb

plus this: https://pastebin.com/23LXUDcB

I don't know much but it looks like some permission problem maybe?   No idea..

Byggvir, 2019-05-30:
Same problem. 
Log file does not exists or is empty.

en3r0, 2019-06-02:
This also fixed my problem after I unplugged the server abruptly:

> chown -R redis:redis /var/log/redis

sserkez, 2020-07-27:
rock64 hanged (was not SSHable), had to manually disconnect from power
Then same problem
Same solution

debu99, 2020-08-18:
chmod 770 /var/lib/redis

wzorroman, 2020-10-14:
mkdir /var/log/redis
    $ chown -R redis:redis /var/log/redis
    $ chmod -R u+rwX,g+rwX,u+rx /var/log/redis
    $ chmod +r /etc/redis/redis.conf