nextcloud/nextcloudpi #693: Problem with default password

MJB-G, 2018-10-27:
Hi

I can't login to nextcloud. Default passwords work for panel, not for webui. I tried to reset password but no luck.

Thank for your help.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.64.10
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      6,7G/16G
rootfs usage         6,7G/16G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    14.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            ens33
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:05

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.95.133",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "14.0.3.0",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
[... 79 lines cut ...]
<summary>Nextcloud logs</summary>

```

{"reqId":"8NMUcrhkcsyLYGFyxkzw","level":3,"time":"2018-10-27T06:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/var\/www\/nextcloud\/cron.php","line":41,"args":["\/var\/www\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":""}
{"reqId":"5YJKKMwtraZW6lt6KZ7i","level":3,"time":"2018-10-27T06:05:02+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":{"Exception":"Doctrine\\DBAL\\Exception\\DriverException","Message":"An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0' with params [\"%%\", \"%%\", \"%%\"]:\n\nSQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":128,"function":"convertException","class":"Doctrine\\DBAL\\Driver\\AbstractMySQLDriver","type":"->","args":["An exception occurred while executing 'SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0' with params [\"%%\", \"%%\", \"%%\"]:\n\nSQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'",{"errorInfo":["42000",1253,"COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Connection.php","line":855,"function":"driverExceptionDuringQuery","class":"Doctrine\\DBAL\\DBALException","type":"::","args":[{"__class__":"Doctrine\\DBAL\\Driver\\PDOMySql\\Driver"},{"errorInfo":["42000",1253,"COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"},"SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0",{"1":"%%","2":"%%","3":"%%"}]},{"file":"\/var\/www\/nextcloud\/lib\/private\/DB\/Connection.php","line":195,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->","args":["SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0",{"1":"%%","2":"%%","3":"%%"},{"1":2,"2":2,"3":2},null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Query\/QueryBuilder.php","line":206,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->","args":["SELECT `uid`, `displayname` FROM `oc_users` `u` LEFT JOIN `oc_preferences` `p` ON (`userid` = `uid`) AND (`appid` = 'settings') AND (`configkey` = 'email') WHERE (`uid`  COLLATE utf8mb4_general_ci LIKE ?) OR (`displayname`  COLLATE utf8mb4_general_ci LIKE ?) OR (`configvalue`  COLLATE utf8mb4_general_ci LIKE ?) ORDER BY `uid_lower` ASC LIMIT 500 OFFSET 0",{"1":"%%","2":"%%","3":"%%"},{"1":2,"2":2,"3":2}]},{"file":"\/var\/www\/nextcloud\/lib\/private\/DB\/QueryBuilder\/QueryBuilder.php","line":214,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/User\/Database.php","line":282,"function":"execute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/User\/Database.php","line":383,"function":"getDisplayNames","class":"OC\\User\\Database","type":"->","args":["",500,0]},{"file":"\/var\/www\/nextcloud\/lib\/private\/User\/Manager.php","line":229,"function":"getUsers","class":"OC\\User\\Database","type":"->","args":["",500,0]},{"file":"\/var\/www\/nextcloud\/apps\/files\/lib\/BackgroundJob\/ScanFiles.php","line":101,"function":"search","class":"OC\\User\\Manager","type":"->","args":["",500,0]},{"file":"\/var\/www\/nextcloud\/lib\/private\/BackgroundJob\/Job.php","line":61,"function":"run","class":"OCA\\Files\\BackgroundJob\\ScanFiles","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/lib\/private\/BackgroundJob\/TimedJob.php","line":55,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"\/var\/www\/nextcloud\/cron.php","line":123,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Driver\/AbstractMySQLDriver.php","Line":115,"Previous":{"Exception":"Doctrine\\DBAL\\Driver\\PDOException","Message":"SQLSTATE[42000]: Syntax error or access violation: 1253 COLLATION 'utf8mb4_general_ci' is not valid for CHARACTER SET 'utf8'","Code":"42000","Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Connection.php","line":847,"function":"execute","class":"Doctrine\\DBAL\\Driver\\PDOStatement","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/lib\/private\/DB\/Connection.php","line":195,"function [... cut ...]
{"reqId":"W9QJDn8AAQEAANMqdsQAAAAA","level":2,"time":"2018-10-27T06:43:51+00:00","remoteAddr":"::1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=ncp","message":"Login failed: 'ncp' (Remote IP: '::1')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"14.0.3.0"}
{"reqId":"W9QJ6H8AAQEAANM@O2cAAAAJ","level":2,"time":"2018-10-27T06:47:29+00:00","remoteAddr":"::1","user":"--","app":"core","method":"POST","url":"\/index.php\/login?user=ncp","message":"Login failed: 'ncp' (Remote IP: '::1')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:60.0) Gecko\/20100101 Firefox\/60.0","version":"14.0.3.0"}

```
</details>

ovpc, 2018-10-27:
There no longer is a default password, they are generated at first run.
You can reset it from ncp-web or ncp-config by running nc-admin.

MJB-G, 2018-10-27:
This generated password doesn't work and I have already try to reset it.

MJB-G, 2018-10-27:
I have installed Debian 9 in virtualbox, apache, php, maridb : 

# apt install apache2 php libapache2-mod-php mariadb-server php-mysql
# apt install php-curl php-gd php-intl php-json php-mbstring php-xml php-zip
# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash

Then I go to localhost and click on activate. The generated password for the panel works, but not the other. It says the password is incorrect.

ovpc, 2018-10-27:
There is a ready to go ncp VM check it [here](https://ownyourbits.com/2018/10/25/introducing-the-nextcloudpi-vm/)

But any case no need for all the apt install (it's probably whats messing it up)
Just run the curl command and should be fin.

nachoparker, 2018-10-27:
hello @MJB-G 

please use the forums for support, until we discover that there is really something to fix

thanks!

closed 2018-10-27 by nachoparker