nextcloud/nextcloudpi #695: Be able to set the domain when not using dynamic dns options.

rvanlaar, 2018-10-27:
**Description**  
I would like to be able to set the domain name of my NextcloudPi to something of my own choosing.
Preferably via the web interface or the ncp-config utility.
Or at the minimum information about setting the config via the wiki.

**Example**  
My domain is configured to point to my raspberry Pi. When I go to that domain I get the message: 
"Access through untrusted domain". 
It mentions I need to configure my domain via editing the config.php.

nachoparker, 2018-12-01:
@BETEP-ok  I accept PRs but I won't develop or maintain things that I don't use personally. I only use no-ip and it's the only thing I personally maintain.

If somebody is up to provide that **and maintain it** I will accept it ;)

rvanlaar, 2018-12-03:
It seems for developing the following is needed:

in the likes of: https://github.com/nextcloud/nextcloudpi/blob/308a1a2ec67580d44591dbbe21c6373e60309cf2/etc/ncp-config.d/DDNS_duckDNS.sh
Some javascript: https://github.com/nextcloud/nextcloudpi/blob/e26a834b63da30904735636b0920ecfb7d0418b0/ncp-web/wizard/JS/wizard.js
And some php/html: https://github.com/nextcloud/nextcloudpi/blob/e26a834b63da30904735636b0920ecfb7d0418b0/ncp-web/wizard/index.php

Is there more needed for developing? 
I didn't find automated tests, is that correct?

What do you need for support and the process around it?

nachoparker, 2018-12-04:
@rvanlaar 

hi!

the info you need is here

https://github.com/nextcloud/nextcloudpi/blob/master/CONTRIBUTING.md

https://github.com/nextcloud/nextcloudpi/wiki/ncp-app-developer-guide

there is some basic automated testing in the `tests` folder, but it's not very comprehensive (help is welcome)

See https://github.com/nextcloud/nextcloudpi/wiki/Testing-NextCloudPi

Feel free to ask here and in the Telegram group ;)

nachoparker, 2021-10-15:
you can set trusted domains from ncp-web as of today

closed 2021-10-15 by nachoparker