nextcloud/nextcloudpi #699: QueryException: Could not resolve defaultTokenProvider! Class defaultTokenProvider does not exist in /data/app/lib/private/AppFramework/Utility/SimpleContainer.php:110

de-code, 2018-10-29:
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.62.2
NextCloudPi image    NextCloudPi_docker_10-02-18
distribution         Debian GNU/Linux 9 \n \l
USB devices          none
datadir              /data/app/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      8.8G/30G
rootfs usage         8.8G/30G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               13:08

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
OCP\AppFramework\QueryException: Could not resolve defaultTokenProvider! Class defaultTokenProvider does not exist in /data/app/lib/private/AppFramework/Utility/SimpleContainer.php:110
Stack trace:
#0 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(125): OC\AppFramework\Utility\SimpleContainer->resolve('defaultTokenPro...')
#1 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('defaultTokenPro...')
#2 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(81): OC\ServerContainer->query('defaultTokenPro...')
#3 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(104): OC\AppFramework\Utility\SimpleContainer->buildClass(Object(ReflectionClass))
#4 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(125): OC\AppFramework\Utility\SimpleContainer->resolve('OC\\Authenticati...')
#5 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#6 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(171): OC\ServerContainer->query('OC\\Authenticati...')
#7 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#8 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(123): Pimple\Container->offsetGet('OC\\Authenticati...')
#9 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#10 /data/app/lib/private/Server.php(364): OC\ServerContainer->query('OC\\Authenticati...')
#11 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#12 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(123): Pimple\Container->offsetGet('OCP\\IUserSessio...')
#13 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IUserSessio...')
#14 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(171): OC\ServerContainer->query('OCP\\IUserSessio...')
#15 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#16 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(123): Pimple\Container->offsetGet('UserSession')
#17 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('UserSession')
#18 /data/app/lib/private/Server.php(1408): OC\ServerContainer->query('UserSession')
#19 /data/app/lib/private/Server.php(683): OC\Server->getUserSession()
#20 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#21 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(123): Pimple\Container->offsetGet('OC\\App\\AppManag...')
#22 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\App\\AppManag...')
#23 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(171): OC\ServerContainer->query('OC\\App\\AppManag...')
#24 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
[... 55 lines cut ...]
2018-10-28 21:50:43 1995808096 [Note] Server socket created on IP: '127.0.0.1'.
2018-10-28 21:50:43 1995808096 [Note] mysqld: ready for connections.
Version: '10.1.26-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.1
2018-10-28 21:50:49 1988838192 [Note] mysqld: Normal shutdown

2018-10-28 21:50:49 1988838192 [Note] Event Scheduler: Purging the queue. 0 events
2018-10-28 21:50:49 1434436400 [Note] InnoDB: FTS optimize thread exiting.
2018-10-28 21:50:49 1988838192 [Note] InnoDB: Starting shutdown...
2018-10-28 21:50:50 1988838192 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2018-10-28 21:50:52 1988838192 [Note] InnoDB: Shutdown completed; log sequence number 5453547
2018-10-28 21:50:52 1988838192 [Note] mysqld: Shutdown complete

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"4Cu5V6ZcAJOQJzTsKQMD","level":3,"time":"2018-10-28T21:30:19+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/data\/app\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/data\/app\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/data\/app\/cron.php","line":41,"args":["\/data\/app\/lib\/base.php"],"function":"require_once"}],"File":"\/data\/app\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":"14.0.1.1"}
{"reqId":"W9YtQKwSAAIAAAb8Eq0AAAMU","level":2,"time":"2018-10-28T21:42:24+00:00","remoteAddr":"192.168.0.22","user":"--","app":"core","method":"POST","url":"\/index.php\/login","message":"Login failed: 'ncp' (Remote IP: '192.168.0.22')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/70.0.3538.77 Safari\/537.36","version":"14.0.1.1"}

```
</details>

nachoparker, 2018-10-29:
Somebody else reported this with the latest NC version. Try to remove container and volume, and then run again. Wait until you see `NC Init done` in the docker logs and try again.

de-code, 2018-10-30:
I've reset it but running into the same issue when it's reloading the server. I'm never seeing `NC Init done`. I'm only seeing `Initializing empty volume..` but after a while the page with the generated ncp passwords appear.

nachoparker, 2018-10-31:
you might have to wait for a while, that message has to appear

de-code, 2018-10-31:
Just run it again and this time `NC Init done` appeared (maybe the console didn't update correctly before). I can also login (as was the case the first time). But as soon as the server is restarted, it's broken.

BTW I am using `docker-compose` to encapsulate the config:

```yaml
version: '3'

volumes:
  ncdata:
    driver: local

services:
  nextcloud:
    image: ownyourbits/nextcloudpi-armhf:latest
    restart: always
    ports:
      - ${HTTP_PORT:-80}:80
      - ${HTTPS_PORT:-443}:443
    volumes:
      - ncdata:/data
```

Does restarting the server work for you with the latest version?

Looks like a bug reported here: https://github.com/nextcloud/server/issues/11225
(with #698 release tags I could try a previous version)

nachoparker, 2018-11-01:
how do you restart? how is it broken?

de-code, 2018-11-01:
> how do you restart?

Stop and start the docker container. Also changing the config using `nc-config` will restart the server.

> how is it broken?

HTTP 500 trying to access the website, with the reported error message in the log (`QueryException: Could not resolve defaultTokenProvider`)

nachoparker, 2018-11-02:
I was able to reproduce

nachoparker, 2018-11-02:
I think I have found the fix