nextcloud/nextcloudpi #702: No Sync with davDroid

Ricardosgeral, 2018-10-30:
Hi.

Does anyone have calendar and contacts synced with nextcloudpi?
This used to work but now i cant sync anything.

Did anything changed?
thanks

**System information**

hurrikan77, 2018-10-30:
I have no problem with calendar, task and contact sync.

Best regards

nachoparker, 2018-10-31:
works well here, nothing changed

Ricardosgeral, 2018-10-31:
ok. Something must be messing with my configuration in my android phone.

Just to clear, is this what i need to put in davdroid?
URL: https://ricarcloud.duckdns.org/remote.php/dav

or should i put
URL: https://ricarcloud.duckdns.org/*nextcloud*/remote.php/dav

I did not understood the instructions from
https://www.davdroid.com/tested-with/nextcloud/

Thanks

rvanlaar, 2018-11-07:
I had similar problems while configuring them.
I couldn't put my finger on it. In the end it seemed that after a reboot it magically started working.
You should use the complete URL that the 'share' button gives.

hurrikan77, 2018-11-08:
In my opinion the easiest way ist to install the nextcloud app, login into your account. Then enter the go to the settings menu and use the function "calender and contact sync". This funtion opens davdroid and enters all nessesary data. The only thing what you have to do is to enter your password ones more.

closed 2018-11-09 by nachoparker