nextcloud/nextcloudpi #703: Time delay when logging in

Spielmops, 2018-10-31:
I tried Nextcloud on my Raspberry Pi 3B with the image for Raspies. And I have this horrible time-delay, when gaining access: about 75 seconds with nothing plus 30 seconds if a failure log-in happened before. 

I looked at several logs and found something interesting in /var/log/apache2/nc-access.log If you want to look at it, do as root with ssh in a terminal:

tail -f /var/log/apache2/nc-access.log 

Go to the login-page with your browser and wait, until tail stops, do several "Enter" to separate the following lines. Type your username and password into the webpage and click "log in". It took nearly 75 seconds for the next line to appear in tail (with a wrong timestamp), so I assume that it is JavaScript on my client-machine, that makes this delay.

I read about this delay and found no solution, but several remarks, that this happens on some machines and sometimes not. And one post with "chromium works" which made me look further to JavaScript.

Spielmops

ArtJames, 2018-11-16:
Check /etc/php/7.x/fpm/conf.d/ and /etc/php/7.x/cli/conf.d/ for double opcache links. If there are @10-opcache.ini and @20-opcache.ini, remove the @20's and reboot the Pi. Check for them after each ncp-update, 'cause they shop up again.  This happens since NCP is using PHP7.2. My login times went down from ~30 seconds to acceptable 5 seconds.

Spielmops, 2018-11-16:
That's not the case with my Raspberry Pi, but I found those double-links in my Odroid-installation.

nachoparker, 2018-11-27:
the first time it takes longer, but then it is way faster. You may be having hardware issues.

the OPcache warnings are unrelated (and harmless)

closed 2018-11-27 by nachoparker

Spielmops, 2018-11-28:
> the first time it takes longer, but then it is way faster

I don't write about first times

> You may be having hardware issues

The Pi3 is running quite fast. And: my old cloud (Owncloud) runs on a Pi2 since 2 years so I know, how fast things can go ...