nextcloud/nextcloudpi #706: MySQL and Apache don't run correctly on nextcloudpi-armhf

matrob, 2018-11-04:
I'm using the command:
`docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-armhf 192.168.178.12`

`docker logs -f nextcloudpi`

> Initializing empty volume..
> Making /usr/local/etc/ncp-config.d persistent ...
> Making /etc/services-enabled.d persistent ...
> Making /etc/letsencrypt persistent ...
> Making /etc/shadow persistent ...
> Making /etc/cron.d persistent ...
> Making /etc/cron.daily persistent ...
> Making /etc/cron.hourly persistent ...
> Making /etc/cron.weekly persistent ...
> Making /etc/apache2 persistent ...
> Starting PHP-fpm
> Starting Apache
> Action 'start' failed.
> The Apache error log may have more information.
> Starting Redis
> Starting Cron
> Starting Postfix
> postsuper: warning: inet_protocols: disabling IPv6 name/address support: Address family not supported by protocol
> Uninitialized instance, running nc-init...
> Activating nc-init
> Setting up a clean Nextcloud instance... wait until message 'NC init done'
> Setting up database...
> ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "No such file or directory")
> 2018-11-04 11:26:19 3069619552 [Note] mysqld (mysqld 10.1.26-MariaDB-0+deb9u1) starting as process 200 ...
> Init done

What can I do to solve the problem?

nachoparker, 2018-11-27:
please, reopen if somebody else is seeing this

closed 2018-11-27 by nachoparker

paschaef, 2019-07-12:
Hi, 
I encountered the same issue. I started a new instance of nextcloudpi-armhf docker with docker-compose and erverything is fine, the nextcloudpi version after the fresh install is 1.10.0
docker-compose down and start again works fine as well after the installation.
However, after upgrading nextcloudpi to version 1.14.3 and then doing a docker-compose down followed by a docker-compse up the same error with MySQL occurs:

```
nextcloudpi_1  | Starting PHP-fpm
nextcloudpi_1  | /etc/services-enabled.d/010lamp: line 22: php-fpm: command not found
nextcloudpi_1  | Starting Redis
nextcloudpi_1  | Starting Cron
nextcloudpi_1  | Starting Postfix
nextcloudpi_1  | ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "No such file or directory")
nextcloudpi_1  | Init done
````

Any Idea how to solve this issue?

(I've run a nextcloud pi docker instance for month with this setting, but stopped the container for the first time for maintanance reasons)

nachoparker, 2019-07-13:
@paschaef it should go away by running the upgrade again. The problem will stop happening when we release the new container.

paschaef, 2019-07-13:
@nachoparker thanks. Unfortunately performing the nextcloud pi upgrade twice did not help. After stoping the container and then restarting it the same error message gets printed out. 
After that neither nextcloud pi pannel nor nextcloud web interface is accessible.
Think I have to wait for the release of the new container....
Have you any idea when you will release the new container?

nachoparker, 2019-07-19:
Hopefully within the next few days.

paschaef, 2019-08-07:
Issue is fixed with new docker container. Thanks a lot  :smiley: