nextcloud/nextcloudpi #707: Better Clarity in NCP Admin settings related to database and datadir

wayneoutthere, 2018-11-04:
**Description**  
After losing many hours on two separate occasions I realized that I can file a bug here and maybe help improve clarity to save other people time.

**What we see now in the admin settings for data dir option** 
[image]

**What we see now for the admin settings for database option**
[image]

## The problem
It seems like these default lines in the field are representing the **current** location of the db and datadir, but, it's not true.  After I ran 'sudo ncp-report' I discovered that's not where my db or datadir was pointing.

After wasting much time I realized that the intention of these two pages is if you plan to **change from the defaults** you enter new information here.

I surveyed two other people and they agreed this is not clear.

### The Solution

#### Option 1 - Show the current actual location of the two in the field
You could just put the path where they are now with the real names and locations and that would be more clear.  Then say 'if you wish to change the location, etc, just enter in the field and click 'run'"

#### Option 2 - give a long clear description in the page
Something like this:

>The current path in the field is **just a template**. 
>if you leave it blank and do not click 'run' your paths will remain as they were during your NCP >installation.  If you wish to change the location of the {datadir} just type in the new path.  Note >'USBdevice' is the NCP default label, blah blah blah"

I hope this is clear and will help the next version! :)

nachoparker, 2018-11-18:
implemented by @ovpc 

thanks both!

closed 2018-11-18 by nachoparker

wayneoutthere, 2018-11-19:
coooool thx