nextcloud/nextcloudpi #708: Preview Generator not working on freshly setup device

Hofer-Julian, 2018-11-04:
I set up my device (odroid htc2) three days ago.

- format to BTRFS
- activate automount
- move datadir to HDD

When I tried to start *nc-preview* I got the error:
```
There are no commands defined in the "preview" namespace.
```

It now works  after running:

```bash
sudo -u www-data php /var/www/nextcloud/occ app:install previewgenerator
```

Is there a bug or should I have run a special *nc-* command beforehand?

nachoparker, 2018-11-06:
How did you install NCP? SD-card? curl installer?

Hofer-Julian, 2018-11-06:
I flashed NextCloudPi_OdroidHC2_09-29-18 on my SD-card, formatted to BTRFS, activated automount, moved datadir to HDD and then updated to the newest nextcloudpi and nextcloud version.

Afterwards I tried nc-preview and got the error mentioned above.

Thank you for all the work you put into this project!

nachoparker, 2018-11-07:
That's because you didn't flash the latest version. We don't support updating a system that is older than a month.

Thanks again!

Hofer-Julian, 2018-11-07:
Okay! :) 

I can't 100% rule out that I downloaded an outdated version. But please let me ask you: Why is the outdated version still offered in the downloads?

nachoparker, 2018-11-08:
normally there is only the latest version. In this case, we also have the previous one because I didn't have the time to test that image. If you tell me that the last one works, I'll get rid of the older one

Hofer-Julian, 2018-11-08:
Thank you for the information! :) 

I started like we discussed with the older image and I don't really know how to thoroughly test it without endangering the data I have currently on my SD card/HDD.

How do you test your images?

nachoparker, 2018-11-09:
Here https://github.com/nextcloud/nextcloudpi/wiki/Testing-NextCloudPi.

You can always backup and restore on the new image if you want to

closed 2018-11-09 by nachoparker